rstrouse/ESPSomfy-RTS #1: Problem with long press

philder, 2023-02-23:
Hi there!

I've read all your project and made a ESPSomfy-RTS with your tutorial. Thank you for the share!

Now I have two problems: I can't open or close the blinds because they only opens or closes when I do a long press on the remote (up or down). If I do a short press in the buttons the blinds only jogs. Is there any way to change this? I don't care if I change the behavior in the remote also.

Attached a photo of my remote:
[image]

https://user-images.githubusercontent.com/49433063/221030398-27fa3983-0f02-4772-bb53-5f952ef81840.mp4

My second problem is the remote pairing to ESPSomfy-RTS. I put the pairing mode, click the buttons on the remote but nothing happens in the UI.

Thank you!

rstrouse, 2023-02-23:
Interesting.  This looks like you are still in programming mode to set the endpoints.  Press the prog button on the back of the remote again.

rstrouse, 2023-02-23:
The linking problem (not pairing) you are describing above seems to indicate that your transceiver is not receiving.  Can I get you to open the Somfy tab, then press configure transceiver button.   On this screen press the buttons on your remote.  You should see the number next to the RSSI label increment on each button press.  If it is not double check your wiring to ensure the radio pins are connected correctly for your transceiver module.

[image]

philder, 2023-02-24:
> Interesting. This looks like you are still in programming mode to set the endpoints. Press the prog button on the back of the remote again.

This is the normal mode. I don't know if there is an option to program this without long press.

philder, 2023-02-24:
> The linking problem (not pairing) you are describing above seems to indicate that your transceiver is not receiving. Can I get you to open the Somfy tab, then press configure transceiver button. On this screen press the buttons on your remote. You should see the number next to the RSSI label increment on each button press. If it is not double check your wiring to ensure the radio pins are connected correctly for your transceiver module.
> 
> [image]
> 
> EDIT: Ignore the pin settings in the screenshot above. These are from another board that I am working with.

I realized that I made a mistake with RX and TX gpios but I can't save the new values on the GUI. I set the RX/TX new values but it doesn't save.

rstrouse, 2023-02-24:
Thanks I have a release that I am putting together to alleviate that.  It appears that the setting and getting are out of order.  

As for the long press.  See this video. https://www.youtube.com/watch?v=Wq7cArPeoFA.  A few years ago one of my shades had this behavior and it turned out that the shade was stuck in the endpoint setting mode.

rstrouse, 2023-02-24:
Ok so I published 1.2.3 that should take care of your issue with the pins not saving.

rstrouse, 2023-02-24:
Just so you know you should be able to update your device by navigating to the Updates tab.  Then take the binaries from the release and first update the firmware.  Then update the application.

Firmware = SomfyController.ino.esp32.bin
Application  = SomfyController.littlefs.bin

philder, 2023-02-24:
Thank you so much! With the update I can save the GPIOs now and after a while changing the configs I can pair the remote now. For the other problem I have found the receiver module manual and there is two modes of operation: US mode and EU mode. I have to factory reset all the blinds and program everything from scratch in order to get to US mode. See part of the manual below:

[image]

Thank you for your kind and support!

I'll let you know when I reprogram everything. Wish me luck!

rstrouse, 2023-02-24:
Interesting.  My shades do not have tilt control so I have some questions on their operation.

1. The entire shade opens and closes when pressing the up or down buttons for less than .5 seconds according to how I read above.  However, it you press the button for .5 to 1.99999 seconds, does it just do nothing or does it look at that as an up or down command?
2. I am assuming here so just to confirm.  Once the shade is in motion up or down do you still use the my button to stop it?
3. If tilting up or down does it require the my button to stop the tilt or does it stop when you release the button?
4. Europe mode simply seems to be an inverse of the US mode.  Does the my button stop the slat opening if you press < .5  seconds up or down as well?

Sorry for the stupid questions but I have watched enough videos that tell me how to program them but not how to actually use them in real life.  Oh and good luck.

rstrouse, 2023-03-04:
@philder I took a stab at controlling the tilt position on your blinds.  Pull the latest release and update your controller.  You do not need to reflash using the flash download tool.  Simply navigate to the gears menu --> Updates tab then update the firmware with the file `SomfyController.ino.esp32.bin` file attached to the release.  Then update the application using `SomfyController.littlefs.bin` file attached to the release.

After the page reloads, change the shade type to Blind and check the Has Tilt checkbox.  This will then allow you to change and position the tilt for the shades.  I do not have a tilt enabled blind so I used the timing described in the sparse documentation on how to change the tilt position of a blind.  If you are a Home Assistant user, I will be pushing a version update there as well.  It will enable the tilt functions and provide the services to setting the tilt in Home Assistant.

rstrouse, 2023-03-11:
Just curious if the tilt functions work as you expect.  I just posted another release v1.4.0 that cleans up some of the tilt functions.

jfparis, 2023-03-25:
The tilt function not working for me. 

I will try to record the messages that the remote send in a datadump

rstrouse, 2023-03-26:
Yeah I do not have a tilt blind so I guessed at the command info.  Please open the logs screen and tilt the blinds with the remote.  Then copy the commands from your clipboard and paste them here.

rstrouse, 2023-03-28:
I take it this is just a tilt up.  Is that correct?

jfparis, 2023-03-28:
correct

rstrouse, 2023-03-28:
Thanks.  This is helpful.

rstrouse, 2023-03-28:
Is there more than one tilt involved in your output?  I am seeing 4 different addresses in your output.

jfparis, 2023-03-29:
This is a 5-channel remote. Only one of them is active at any point in time. As I had to press & hold the button to make it toltI certainly did not change the channel

rstrouse, 2023-03-29:
Thanks, that is good to know.  I created a couple of more tools to pinpoint the repeats and got a cleaner picture of what was communicating on the frequency.  It looks like there is another device somewhere nearby that is transmitting on address 14164537.  ESPSomfyRTS is ignoring frames from this device,  but there were a couple of times that this device scrambled the communications.  I am pretty sure the address of your remote is 2105913.  

That being said, after I filtered out the other stuff that I didn't need to look at, it appears I am not repeating the command long enough.  To be sure you tilt the blinds up by holding the up button until the shade moves.  I assume you let go of the button as soon as the tilt action started.  Are my assumptions correct?

Again I don't have a shade that does this function.

rstrouse, 2023-03-29:
Go ahead and update your firmware to v1.4.4.  I added additional pulses to the tilt commands.

jfparis, 2023-03-30:
I will test thoroughly this weekend. Thank you

jfparis, 2023-04-05:
Brief update: I have tested and it works better. There is still one "logic" issue which I need to try to report and document properly. I will try to do that after the easter weekend

rstrouse, 2023-04-05:
Thanks for the update.  Let me know how it goes.  I do not have a tilt blind and have been guessing so you are my eyes on this.

Strangehadron, 2023-04-06:
Hi,
I've tilt one too,
Short press = control the tilt
Long press = Control the opening/closing

With the web interface, short click control the tilt, long click on the button (stay click for like 2sec) seem to control the opening/closing.

But with the HA integration, this is unusable. Up and down arrow move the blind for like couple of mm, and tilt command rise a little the opening or closing.

I've tried to untick tilt function in the parameters, but same behavior in ha.

Tell me if you need some info, log or whatever to help !

rstrouse, 2023-04-06:
There are two modes that the motor can operate in.  If you switch it the nudge thingy goes away and you can set a position by percentage. The instructions on how to do it are somewhere online.  I think the mode you are in is called euro mode.

Strangehadron, 2023-04-08:
> There are two modes that the motor can operate in. If you switch it the nudge thingy goes away and you can set a position by percentage. The instructions on how to do it are somewhere online. I think the mode you are in is called euro mode.

Yes, i've read that years ago when i've built my rflink but never achieve to change the mode on my blind. :/

rstrouse, 2023-04-08:
Which remote do you have?

Strangehadron, 2023-04-08:
> Which remote do you have?

I think it’s a situo pure.

[image]

rstrouse, 2023-04-08:
You should be able to disconnect power from the blind for 30 seconds then power it back on.  Within 30 seconds you should press the my button until the blind jogs.  This is supposed to change the mode.

Strangehadron, 2023-04-12:
> You should be able to disconnect power from the blind for 30 seconds then power it back on. Within 30 seconds you should press the my button until the blind jogs. This is supposed to change the mode.

Thanks, i will try that tomorow ! 👍

rstrouse, 2023-05-09:
@Strangehadron & @philder have your issues been resolved?  Please update to the current release as there have been many update related to tilt motors.