rstrouse/ESPSomfy-RTS #2: AsyncTCP.h missing

MadsALarsen, 2023-02-23:
i cant compile the bin

error message:

In file included from /Users/Downloads/Somfy_Remote-master/Somfy_Remote/Somfy_Remote.ino:28:
/Users/Documents/Arduino/libraries/AsyncElegantOTA/src/AsyncElegantOTA.h:11:14: fatal error: ESPAsyncTCP.h: No such file or directory
   11 |     #include "ESPAsyncTCP.h"
      |              ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

rstrouse, 2023-02-23:
It does not use AysncElegantOTA.  That include does not exist in the repository so I am guessing you are trying to compile a completely different repository than this.  

It looks like potentially this one.  https://github.com/Nickduino/Somfy_Remote

MadsALarsen, 2023-02-25:
Sorry my bad, thanks for for the headsup

now I got the right one but I can't compile the file??

"Arduino: 1.8.19 (Mac OS X), Board: "WEMOS LOLIN32, 80MHz, Default, 240MHz (WiFi/BT), 921600"

ConfigSettings.cpp:2:104: fatal error: LittleFS.h: No such file or directory
compilation terminated.
exit status 1
LittleFS.h: No such file or directory
"

see screenshot
https://pasteboard.co/Z8jhDkmbtMVm.png

MadsALarsen, 2023-02-25:
I think I found that problem now its a bigger one :-( 

"Arduino: 1.8.19 (Mac OS X), Board: "WEMOS LOLIN32, 80MHz, Default, 240MHz (WiFi/BT), 921600"

In file included from /Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.cpp:3:0:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h: In function 'std::__cxx11::string util::to_string(const T&)':
Utils.h:21:21: error: 'to_string' is not a member of 'std'
     std::string str{std::to_string (t)};
                     ^
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:21:21: note: suggested alternative:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:20:37: note:   'util::to_string'
   template <typename T> std::string to_string(const T& t) {
                                     ^
In file included from /Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/SSDP.cpp:2:0:
/Users/madsalarsen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/AsyncUDP/src/AsyncUDP.h:47:1: error: expected class-name before '{' token
 {
 ^
In file included from /Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/SSDP.cpp:3:0:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h: In function 'std::__cxx11::string util::to_string(const T&)':
Utils.h:21:21: error: 'to_string' is not a member of 'std'
     std::string str{std::to_string (t)};
                     ^
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:21:21: note: suggested alternative:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:20:37: note:   'util::to_string'
   template <typename T> std::string to_string(const T& t) {
                                     ^
In file included from /Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Somfy.cpp:5:0:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h: In function 'std::__cxx11::string util::to_string(const T&)':
Utils.h:21:21: error: 'to_string' is not a member of 'std'
     std::string str{std::to_string (t)};
                     ^
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:21:21: note: suggested alternative:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:20:37: note:   'util::to_string'
   template <typename T> std::string to_string(const T& t) {
                                     ^
/Users/madsalarsen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/variants/lolin32/pins_arduino.h: At global scope:
Utils.h:5:23: error: expected unqualified-id before numeric constant
   #define LED_BUILTIN 2
                       ^
/Users/madsalarsen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/variants/lolin32/pins_arduino.h:14:22: note: in expansion of macro 'LED_BUILTIN'
 static const uint8_t LED_BUILTIN = 5;
                      ^
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/ConfigSettings.cpp: In member function 'bool BaseSettings::loadFile(const char*)':
ConfigSettings.cpp:14:6: error: 'LittleFS' was not declared in this scope
   if(LittleFS.exists(filename)) {
      ^
ConfigSettings.cpp:15:5: error: 'File' was not declared in this scope
     File file = LittleFS.open(filename, "r");
     ^
ConfigSettings.cpp:16:17: error: 'file' was not declared in this scope
     filesize += file.size();
                 ^
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/ConfigSettings.cpp: In member function 'bool BaseSettings::saveFile(const char*)':
ConfigSettings.cpp:30:3: error: 'File' was not declared in this scope
   File file = LittleFS.open(filename, "w");
   ^
ConfigSettings.cpp:34:22: error: 'file' was not declared in this scope
   serializeJson(doc, file);
                      ^
In file included from /Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Network.cpp:9:0:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h: In function 'std::__cxx11::string util::to_string(const T&)':
Utils.h:21:21: error: 'to_string' is not a member of 'std'
     std::string str{std::to_string (t)};
                     ^
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:21:21: note: suggested alternative:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:20:37: note:   'util::to_string'
   template <typename T> std::string to_string(const T& t) {
                                     ^
In file included from /Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Web.cpp:8:0:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h: In function 'std::__cxx11::string util::to_string(const T&)':
Utils.h:21:21: error: 'to_string' is not a member of 'std'
     std::string str{std::to_string (t)};
                     ^
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:21:21: note: suggested alternative:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:20:37: note:   'util::to_string'
   template <typename T> std::string to_string(const T& t) {
                                     ^
In file included from /Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/SomfyController.ino:7:0:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h: In function 'std::__cxx11::string util::to_string(const T&)':
Utils.h:21:21: error: 'to_string' is not a member of 'std'
     std::string str{std::to_string (t)};
                     ^
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:21:21: note: suggested alternative:
/Users/madsalarsen/Documents/ESPSomfy-RTS/SomfyController/Utils.h:20:37: note:   'util::to_string'
   template <typename T> std::string to_string(const T& t) {
                                     ^
exit status 1
'to_string' is not a member of 'std'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

rstrouse, 2023-02-25:
The LittleFS files should not exist in the project directly, that will cause the compiler to spit out vile errors that mask the real problem.  To include a library use the `Sketch->Include` Library menu.  

Did you follow this wiki https://github.com/rstrouse/ESPSomfy-RTS/wiki/Compiling-the-Firmware to set up the Arduino IDE?  If you did lets start with the board that you are trying to compile the firmware to.   Which one is it?

rstrouse, 2023-02-26:
@MadsALarsen just checking did you get your board working?

MadsALarsen, 2023-02-27:
> @MadsALarsen just checking did you get your board working?

I did not have the time this weekend so I will look at it tonight or tomorrow 👍🏻
I will give you a comment on the progress 😀

rstrouse, 2023-02-27:
Well good news!  I managed to sit down and figure out how to install the initial firmware using the ESP32 flash download tool.  You will find complete instructions in the following wiki to get this done.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware

rstrouse, 2023-03-11:
@MadsALarsen did you get your firmware loaded?

MadsALarsen, 2023-03-19:
No not yet, I was having a little trouble with it but I think I will try again tomorrow or so

rstrouse, 2023-03-19:
Have a look at the installing firmware wiki.  The process there makes it much easier to get started.  Let me know if you need anything.

rstrouse, 2023-03-29:
@MadsALarsen I assume you got the firmware loaded onto your device.  If you look at the wiki https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware there are quite a few different methods to onboard your device that do not require re-compiling the firmware.

I am closing this for now and if you need further assistance let me know.

closed 2023-03-29 by rstrouse

MadsALarsen, 2023-03-29:
Thanks for your help👍🏻