rstrouse/ESPSomfy-RTS #3: "My" button not captured

schumijo, 2023-03-07:
Hi @rstrouse,

First, thank you for your work !

I managed to pair my Somfy roller shutter and Smoove Origin RTS remote successfully.
But "My" button is not captured. "Up" and "Down" are working correctly.

Any idea ?

rstrouse, 2023-03-07:
Are you referring to the long press my button when the shade is setting its favorite, the my button from idle, or my button to stop it?  These are 3 different things.  If it is one of the former two, you need to set the My Position from the app first.  The remotes have no idea what these setting are and the only time they are transmitted is when they are being set.

Also, under the gears menu check to make sure you are running the most recent release.  Click on gears then Updates it will show you two version numbers.  These both should be 1.3.2.

schumijo, 2023-03-08:
Oh sorry it seems to be a reception problem. When I move my ESP near to the remote it works.
Really strange I have no problem with Up and Down.

rstrouse, 2023-03-08:
That is weird.  How close is near?  You should be getting at least as good of signal that the motor gets (same distance) and in my case at least 30% further than the remotes.  I ordered another batch of radios from another source just to make sure I just didn't get lucky with the E07-M1101Ds I have.  Did you go into the Configure Transceiver menu and tune the radio?

So far I have 6 of them and the reception is much greater than the motors.  But, the performance from the D-SUN radios are very spotty and a couple do not receive unless I am on top of them.  Looking closer at them I realized the CC1101 chip has no label.  The E07-M1101Ds all clearly have the Texas Instruments label.

rstrouse, 2023-03-11:
Just checking in did you get your reception issues hammered out?

schumijo, 2023-03-13:
Yes I started to do some tweaks on transceiver and it is already better.

rstrouse, 2023-03-13:
Which transceiver do you have and where did you land on the receiver settings?  I am starting to compile information on settings for various transceivers.  The Somfy Telis remotes have terrible range.  I wish I knew what the transmit power is but I suspect that it is a 2dBm antenna based on my reading and would explain a lot.  However, the sensitivity on the E07-M1101Ds is very good and I can hear the remotes well outside of their normal range.

Let me know if there is anything I can do to help.

schumijo, 2023-03-15:
I am also using E07-M1101D and I confirm settings described in documentation works fine with Smoove Origin RTS remote and Somfy solar roller shutter
Next step, add my gate with Somfy Slidymoove motor

closed 2023-03-15 by schumijo