rstrouse/ESPSomfy-RTS #4: Almost working

Povilas92, 2023-03-08:
Hello, 

Thanks for making this great tutorial. I've got most of this working - the HA integration and pairing the remote - but i can't get the awning to pair with the ESP. Is there some way to check if the antenna is actually transmitting something? 
I do receive remote commands and can see the awning simulation going up and down in the UI.

I'm using the E07-M1101D RF module with the default pins to ESP. 

[image]

Povilas92, 2023-03-08:
Tried a couple of different pins and found PIN14 on my NodeMCU to be working for TX. It paired instantly!

rstrouse, 2023-03-08:
Glad you got it working!

Thanks for reporting back I will make a note of that for NodeMCU.  The pinouts are hard to nail down.   Just because the ESP32 docs say a pin is ok to use, it doesn't mean that it is ok to use because when they put the chip on the board they steal some pins for other supporting items.

rstrouse, 2023-03-11:
Closing this.  Open a new issue if you need further assistance.

closed 2023-03-11 by rstrouse