rstrouse/ESPSomfy-RTS #7: Can't wait to give this a whirl!

FlaMike, 2023-03-20:
I found your project when looking around the Share Your Projects forum on the Home Assistant site.  I am using the nodejs Pool Controller and its associated HA integration & have been very happy with the results.  Hoping for more of the same with this  project.

To date, I've been using Somfy's ZRTSI RTS to Zwave controller for my roller shades but have been less than delighted with the experience.  In addition to being extremely over-priced, they're Zwave & not even Zwave Plus devices.  And, unlike your project, you cannot prescribe the percentage you would like to open/close the shade--it's either opened or closed, unless you hit the stop button.  Furthermore, there is  no confirmation a command was received/executed.  Finally, if someone uses a Somfy remote to open or close a shade, HA and the shade go out of synch.

As you indicated in the wiki, obtaining a CC1101 transceiver may be challenging--at least in the US.  The most likely source I've located is aliexpress and it could be mid-May before it shows up at my house.  Have you found any better sources for the transciever of choice (an E07-M1101D-SMA with a duck antenna) in the US?

Thank you for sharing this project!  Like the pool controller, it should be a fine and welcome addition to my house.

FlaMike, 2023-03-20:
I just ordered 2 of the transceivers (so I have a spare) from aliexpress & will sit on my hands in anticipation for a month or so.

rstrouse, 2023-03-20:
I recognized your handle right away!

I toyed around getting a ZRTSI but read so many bad reviews on it.  I have to say that I am glad I decided to go this route.  It really bugged me that I couldn't just tell the shades to close half way and walk away after I spent over $20k for them.  You won't be disappointed with this.

Initially, I had a bit of an issue getting ahold of a CC1101.  However, I did manage to nab some from Bezos though.
https://www.amazon.com/dp/B01DS1WUEQ?psc=1&ref=ppx_yo2ov_dt_b_product_details

FlaMike, 2023-03-21:
Thanks so much for getting back to me so quickly!  I guess I'm a bit infamous if you recognized me so quickly.  I can only hope it's because I so appreciate your amazing projects and generous help & not because I'm a difficult consumer of your projects and advice!!

As of this morning, the slow boat from overseas claims my transceivers will arrive before mid April as opposed to late May and they've already been "shipped."  I can hang in there with the current ZRTSI for a few more weeks, but if the transceivers turn out to be delayed, I will certainly do the Bezos "quick ship" alternative.  Thanks for the link.

In the interim, I am awaiting a project box so I may install a "liquid level sensor" on the pool's acid tank.  It goes on the outside of the tank & is actually quite accurate in reporting when the acid level drops below the sensor.  My tank is translucent plastic container, but the sensor may also work for non-translucent vessels.  Super easy project and HA integration.  Here's a link to the project write-up:  https://peyanski.com/contactless-liquid-sensor-home-assistant-and-xkc-y25/

I am curious to see if the Somfy project can also work for ceiling fan controllers.  I have a Bond Bridge that works for the fans about as well as the ZRTSI does for the roller shades i.e., not all that well.  I believe Bond advertises they can control Somfy shades, so I'm hoping your project can turn into a double win for me.  One step at a time, but I look forward to finding out & sharing success on that front as well.

I look forward to reporting my success and saying goodbye to the ZRTSI.

Thanks again for your generosity!

rstrouse, 2023-03-21:
Are you running REM Chem or do you just have a Stenner?  If the latter, did you know that there is a Chem Doser that will schedule your pump and manage your tank levels?  There were some other folks including tagyoureit who had those external tank sensors.  Weird but I suspect there is a small amount of acid fumes that destroy them every few months.

FlaMike, 2023-03-21:
The pump is just a Stenner pump, nothing smart about it.  Similarly, the salt cell just uses a run time percentage, no ORP installed.

In our previous house, we had a Hayward Omnilogic installed with ORP and pH control via the Omnilogic.  The installers never got the ORP to work properly, but the pH feeder was certainly nice to have automated.

I'm confident REM Chem with the recommended Chem Doser would be vastly superior to anything Hayward, Pentair, or others offer.  And I am certainly open to having a more self-sufficient pool sporting more stable chemistry.

If you could kindly send me references for "upping" my pool's game, I would appreciate it greatly.

Thanks!

FlaMike, 2023-03-21:
Never mind, I found DIY Rem Chem at https://github.com/tagyoureit/nodejs-poolController/wiki/DIY-REM-Chem-Controller

FlaMike, 2023-03-21:
Wow!  That's fabulous--I was just reading about the probe installation & wondering if I could pull it off without messing up the plumbing or probes.  This is a much better solution for me.

I'm running the pool controller & the dashboard in a docker container.  Is it possible to run REM Chem in a container as well?  If not, could I run it on a Linux box as a service?  I'm trying to keep all of my home automation stuff on a single box, but I could add a Pi to the mix (or an ESP32!) if that makes life easier.

You are truly a wealth of helpful & much appreciated information!

FlaMike, 2023-03-21:
Is the relay you refer to just for the acid pump?  I can control the pump via the pool controller currently--I had the pool installer add a relay to shut off the acid pump when the pool pump was not running (the circuit breaker kept failing for months after the pool went "live" & I didn't want to create an acid bath in the pool).

rstrouse, 2023-03-21:
Oh it is smarter than that and it already exists in your njsPC/dashPanel installation so the docker answer is yes.  The relay is for control of the acid pump and you can even have it use an optional flow switch for added peace of mind.  It can ensure no dosing takes place when in spa mode as well.  

It can be controlled via an OCP relay but the most failsafe method is using a REM relay because of the latching functions.  This ensures contact is broken and remains broken should any part of the system stop working.

rstrouse, 2023-03-21:
Oh and you won't need that tank level gadget.

FlaMike, 2023-03-21:
Would you like a liquid level sensor?  :-)  Just let me know your address & it will be on its  way.

The pool is super-simple.  No spa, just a main pump, Polaris pump, heater, Intellichlor,  acid pump, drain valve control (for overflow due to heavy rain), & lights.  It's a lap pool, so it's long, skinny, & wonderful :-)

Holy smokes--I had no idea REM was already in my system--I need to dig in a bit. Not sure how/where they added the relay for the acid feeder as it appears to be mis-labeled.  I believe it's the chemical feeder which also shows as the master cleaner.  Perhaps to keep the acid pump from running when the main pump is off??

Do I add the acid pump under the chemistry tab to get to the dosing page?

Sorry if I'm asking naive questions, but this is all new & fabulous to me.

[image]

rstrouse, 2023-03-21:
Yeah I remember now they put a shared body panel on your single body pool.  I think somebody doesn't really understand what the Master Cleaner circuit type really does but that is likely where your relay is connected.

FlaMike, 2023-03-21:
Found the goodies under the chemistry tab.

Yet another newbie question, can I use the water flow rate from the Intelliflow pump as  my flow sensor?  I can certainly use Home Assistant to lock out the acid pump when the flow rate is zero, but I haven't seen how to do that within the Dashpanel.

I'm using the HACS Scheduler Component &  Card to program & run the pool equipment, not the EasyTouch.

FlaMike, 2023-03-21:
Yep, the joy of I have it on my truck, so that's what I'm installing, I'm guessing.

FlaMike, 2023-03-21:
BTW, just did a docker pull to get pool controller & dashpanel 8.0.  I didn't realize they were updated until I looked a second time at the wiki.  Drinking from a fire hose with the chemical doser, but I'll get there!

rstrouse, 2023-03-21:
The water flow rate is not a great way to indicate flow.  Unfortunately, it is tied to electronics in the motor controller.  You are better off not using it.   The built in mode selector should be more reliable.  Typical flow sensors are mechanical and the ones found on a flow cell are the most reliable since they require a column of water in order to activate.   Flapper based flow sensors are the next best, the mode selector is after that.  The way the comms work on the output of the pump are not a failsafe.  The energizing of the filter relay is a better option.

Essentially, that Chem Doser is IntelliPh on roids.

rstrouse, 2023-03-29:
I am going to close this for now.  Open another issue once you get your hardware set up.

closed 2023-03-29 by rstrouse