rstrouse/ESPSomfy-RTS #8: Motors pick up sent signal unreliably

nymanjens, 2023-03-23:
Hi!

Sorry it's me again: My setup is working now, but the Somfy motors are only picking up a fraction of the messages I send its way, even though they are less than 2 meters apart:

## My hardware
- 2 x ESP32 board: az-delivery-devkit-v4
- 2 x CC1101 board: E07-M1101D-SMA
- 2 x Somfy motor for draperies

## My findings:
- **The Somfy remote:**
  - The motors respond to 100% of the remote's commands
  - The transceivers pick up the remotes' signal at ~20 dBm 100% of the time. Haven't optimized the RX settings yet
- **The two CC1101 transceivers**
  - They pick up each others' signal very reliably (99% or so) with better than -30dBm
  - The motors respond to ~50% of the commands from the CC1101
    - Especially the My command seems to be extremely flaky. Only about 10% of commands are picked up
  - The behavior is consistent for both CC1101 transceivers
  - TX power: 10 dBm

## My thoughts

- Could this be the cables between de ESP and CC1101 being too long? My speakers pick up a signal when there is activity on TX, which after fixing https://github.com/rstrouse/ESPSomfy-RTS/issues/5 almost must mean that the wire to the CC1101 acts as an antenna.

Is this something that you have ever noticed in your tinkering?

## Screenshot of example logs

Addresses:
- 270713 and 270714: CC1101
- 10173223: Somfy remote
- Logs file JSON: https://pastebin.com/raw/CprWs11n

[image]

## Photo of setup
[image]

rstrouse, 2023-03-24:
You might try reducing the tx power.  It could be reflecting the signal and that can saturate the receiver.  Your receive signal at -20dBm is spectacular but ESPSomfy can narrow that down so it isn’t overwhelming the receiver.

nymanjens, 2023-03-25:
I varied the TX power from +5 dBm to -30 dBm, but the success rate is roughly consistent.

I used this script to test it:

```
echo Up
curl "http://somfyB.local/shadeCommand?shadeId=1&command=Up&repeat=2"
sleep 2
echo My
curl "http://somfyB.local/shadeCommand?shadeId=1&command=My&repeat=2"
```

which resulted in these logs on the other receiver: https://pastebin.com/raw/5nSR16K2

The success rate for both "up" and "my=stop" together was about 40% for all TX power rates. I'm no longer noticing much difference between "up" and "my=stop" though. They both seem equally flaky.

prbtt, 2023-03-26:
After conducting some testing, I realized that the TX Power has a great impact on the reliability of the signal. Initially, I tested my ESP device right next to the motor using a TX Power of +10-12 dB. The motor was responding to 80-90% of the commands.

I then moved the ESP device about 5 meters away, behind a wooden door. I noticed that the motor did not respond to any of the commands sent at +10-12 dB. However, when I switched the TX Power to -6, the motor responded to all of the commands.

Next, I decided to move the ESP device around 10 meters away from the motor and tested it again with -6 dB. Unfortunately, the motor didn't respond to any commands. However, when I switched the TX Power to -15 dB, the motor responded to 100% of the commands sent.

Furthermore, I also realised that after boot, the esp doesn't save the TX Power that was previously set. It seems to always default to +10 dB. Is it possible to keep the previous settings?

nymanjens, 2023-03-26:
@prbtt : To test the hypothesis that this is about the length of the wires between the ESP and the CC1101: What does your setup look like?

prbtt, 2023-03-26:
> @prbtt : To test the hypothesis that this is about the length of the wires between the ESP and the CC1101: What does your setup look like?

Right now I'm testing with the PCB I designed for this purpose. But I also have the same components as you do with some dupont wires. I'll perform the same tests with it.

[image]

prbtt, 2023-03-26:
After testing with an esp32+c1101 SMA using Dupont wires, I noticed that the motor responded to 100% of the commands sent from -30 to +12.
[image]
[image]

rstrouse, 2023-03-29:
I modified the code so it will now allow you to use a single GDO0 pin for RX and TX.  While you do not have to set it up this way perhaps the multiplexing of the pin settings may reduce any noise from the wire.  Another thought is dupont connectors to not love to be plugged in to many times.  They get loose after a while.  I used 10inch dupont connectors for my test subjects.

v1.4.4 also fixes the issue where the TX power was not being written to NVS.  It should now persist this value between reboots.

nymanjens, 2023-03-30:
> I modified the code so it will now allow you to use a single GDO0 pin for RX and TX

How do I do that? If I configure it as follows:

[image]

I get an error saying that all pins must be unique.

rstrouse, 2023-03-30:
Update the firmware and software to 1.4.4 then hard refresh your browser.

nymanjens, 2023-03-31:
> Update the firmware and software to 1.4.4 then hard refresh your browser.

Ah, I may have forgotten to refresh my browser.

> I modified the code so it will now allow you to use a single GDO0 pin for RX and TX

This didn't make any difference.

Other things I tried
- Different power source / USB cable
- A different TX pin (GPIO14 instead of GPIO12)
- Different board (OLIMEX ESP32-EVB)
- Brand new dupont wires
- Increasing length of TX dupont wire
- Playing a lot with the TX power

But none of it seems to make any difference.

My theories:
- Somehow, the wrong wiring in https://github.com/rstrouse/ESPSomfy-RTS/issues/5 maybe broke the CC1101 boards in a very subtle way so that they still pick up each others' signals, but the signal is too distorted for the motors
- My motor is maybe different in those that you and others have in such a way that the standard remote works (I even bought the remote separately) but the CC1101 doesn't
- The wires are too long

rstrouse, 2023-03-31:
I am more likely to suspect a couple of those things together.  Although, it is impossible to tell the condition of the CC1101 boards I doubt there is damage to the CC1101 boards based on switching GDO0/2.  

It is remarkable that you are getting RSSI values in the -20 to -50dBm range.  That is an extremely strong signal.  If you were dipping into the -90dBm range I would say the signal isn't making it.  Any signal stronger than what you have, might start getting into the saturation point where a receiver can see the signal as out of range.

With v1.4.4 I also added an indication for failed frames where failed frames will show up in the log.  I do not believe you have any of those either since the receiving side of the CC1101 is very strong.  The TX side is as well, given the communications between the two devices.

If the motor wasn't picking up any signal I might be inclined to think there is a difference in the protocol.  But since it is receiving some of them, that can be ruled out.   I wonder if a receiver is near the location of the antenna on the shade if it too would unreliably receive signal.

The hum in your speakers are an indication that there is a receiver nearby that picks up only part of the signal and is translating it to electrical pulses on the speakers.  Do those speakers hum from any other conditions.  With the split phase power in the US this is typically an indication of ground loop.  Do you by chance have a 5v battery like the ones used to charge a cell phone?

nymanjens, 2023-04-01:
> Do those speakers hum from any other conditions.
No, usually not. Perhaps very occasionally from a cellphone.

> Do you by chance have a 5v battery like the ones used to charge a cell phone?

Yes, but it doesn't make any difference

rstrouse, 2023-04-01:
What is the model number for your motor?

nymanjens, 2023-04-01:
> What is the model number for your motors?

Movelite 35 RTS W

1240239A

rstrouse, 2023-04-01:
So the Movelite line does look like the Glydea drapery motors.  Do you know if there is potentially a difference in the frequency where you are located.  Right now the software is hardcoded to 433.42Mhz but I see references for these motors at 426 to 447Mhz.  Clearly it is speaking RTS but perhaps there is some tuning of the carrier frequency is required.  Can you compile the firmware?

nymanjens, 2023-04-02:
>  Can you compile the firmware?

Yes. Happy to try any changes.

rstrouse, 2023-04-02:
Keep in mind I am just fishing at this point to see if there so this may just be a goose chase.

On line 2091 of Somfy.cpp.  The line currently reads.
```
ELECHOUSE_cc1101.setMHZ(this->frequency);                 // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.
```

Add two slashes to comment out this line and add another one underneath it.
```
// ELECHOUSE_cc1101.setMHZ(this->frequency);                 // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.

ELECHOUSE_cc1101.setMHZ(426);
```
If the motor still responds, then increase the frequency and a little bit at a time to see if there are changes.  If it does not respond try changing it to 447 and work backwards.  I don't know whether this will have any impact or not but it is worth a try.  I haven't had seen an issue where motor drops a command sent by the CC1101.

nymanjens, 2023-04-02:
I paired up one of my ESPs to two motors simultaneously (so a single shade in ESPSomfy and two motors paired to it) and used this script (https://pastebin.com/wW8jmNeM) to test them. I manually counted how often they responded vs didn't.

Generally, it was pretty random which motor would respond. There was no one motor better than the other.

Results:

|  Frequency (MHz)  |  Successes  |  Failures  |  Failure rate  | 
| --- | --- | --- | --- |
| 426  |  0  |   all  |  100%  | 
| 433  |  0  |   all  |  100%  | 
| 433.32  |  9  |   8  |  50%  | 
| 433.42  |  17  |   9  |  35%  | 
| 433.46  |  13  |  13  |  50%  | 
| 433.52  |  20  |  15  |  45%  | 
| 433.56  |  2  |  12  |  85%  | 
| 433.62  |  0  |   all  |  100%  | 
| 433.92  |  0  |   all  |  100%  | 
| 435  |  0  |   all  |  100%  | 
| 447  |  0  |   all  |  100%  | 

My conclusion: The frequency seems to be the 433.42 MHz that is documented. Which is consistent with the Somfy remote working so well (it literally always works).

rstrouse, 2023-04-02:
Great work on that.  I agree our carrier appears to be 433.42.   I am shocked that you have any failures on TX.  I say that because my device has a 0% failure rate and have installed it in two different very large homes also without any dropped frames.  The difference may be the motor.

Do you have any ideas?

nymanjens, 2023-04-03:
> The device also sends every command with one repeat.

Increasing the repeat makes it less flaky I think, but only a little bit, as there does seem to be a bit of a correlation in time.

> Which remote do you have?

Situo 1 RTS Pure II

> Do you have any ideas?

From my initial theories:

> My theories:
> 
> * Somehow, the wrong wiring in [Somfy motor doesn't react to transmissions #5](https://github.com/rstrouse/ESPSomfy-RTS/issues/5) maybe broke the CC1101 boards in a very subtle way so that they still pick up each others' signals, but the signal is too distorted for the motors
> * My motor is maybe different in those that you and others have in such a way that the standard remote works (I even bought the remote separately) but the CC1101 doesn't
> * The wires are too long

they all seem quite unlikely, but I have it on my TODO to solder a board where I make the wires as short as possible. Also, buying a different CC1101 board (maybe a different type this time) could at least falsify the first option, so I'm thinking about doing that.

prbtt, 2023-04-03:
> Increasing the repeat makes it less flaky I think, but only a little bit, as there does seem to be a bit of a correlation in time.
>
>
> Situo 1 RTS Pure II
>
>
> From my initial theories:
>
>
> they all seem [...]

I've had success with this one: https://a.aliexpress.com/_mOKChpk

rstrouse, 2023-04-03:
This is definitely a mystery.  Given that the two transceivers send and receive reliably, this has me wondering where the goose might fly off to next.   One might assume that because we have a transceiver pair that are communicating reliably that this might be not ***how*** we are sending the data, but ***what*** we are sending.  I wonder if there is some pulse expected by the motor that we are not giving it.

Somfy, does not describe anything about the differences between the Pure and Pure II protocols so I don't know if there are any real differences.   That being said, one of my friends have the same remote as you for their venetian blinds.  The only difference is the motor.  

Interestingly, the cut sheet for the motor has the protocol listed as Somfy RTS U80 which seems to indicate 80 bits.  What is odd though is the hardware and software sync from the remote clearly indicate 56-bits of transmission from the remote.  When a remote is sending the initial frame, there are 4 beats indicating 2 hardware sync cycles.  Then the repeats have 14 indicating 7 sync cycles.  This differs from the 24 and 12 beat output from an 80 bit remote.

Given the Situo 1 does not have any extended functions like sun or wind flags, tilt wheels, or any other functions than the standard up, down, my, and prog commands I would assume this to be only sending 56 bits as well and seems to be what we are seeing from the pulses we are receiving.

Can I get you to capture a log where you are only using the remote to control the motor?  The log output you posted above, I believe is from one of the ESPSomfy RTS devices.  I want to see if there are any variances for the hardware sync transitions.

nymanjens, 2023-04-04:
> Can I get you to capture a log where you are only using the remote to control the motor?

https://pastebin.com/raw/dCv4JLyL

> Let's see if my theory is correct that the interwebs let me down.

I pushed your changes via the Arduino IDE, but it didn't make any difference. I then uploaded the SomfyController.ino.esp32.bin and SomfyController.littlefs.bin via the update page, but now the pairing is somehow broken, and I'm not getting any response./

rstrouse, 2023-04-04:
Can I now get a capture of the short press pair command from your remote.  I looked at the pulse outputs from your remote and mine and they are now identical.

rstrouse, 2023-04-04:
Just to be clear when you say it made no difference, do you mean that it is no better or worse when it comes to dropped frames?

nymanjens, 2023-04-04:
> the short press pair command 

What is that? 

> Just to be clear when you say it made no difference, do you mean that it is no better or worse when it comes to dropped frames?

Correct

nymanjens, 2023-04-04:
> but now the pairing is somehow broken, and I'm not getting any response.

Hmm, the pairing seems to work again after a reboot, or I was just unlucky with the errors giving the impression that there was no connection.

Anyway, with the latest firmware on the `main` branch, I get the following success rate:

|Successes  |  Failures  |  Failure rate  | 
| --- | --- | --- |
|  11  |  17  |  60%  |

rstrouse, 2023-04-04:
Interesting.  I only need the pulses that are being put out by your remote.  I removed all the stuff that I have verified as noise on the pulses output a few minutes ago, so if I can get you to pull the latest from the `main` branch and press the up button, down button and my button when capturing the logs.  Then press the copy button on the logs screen and paste it here.  You can paste it using the code formatting using the three single quotes thing if it is easier.

nymanjens, 2023-04-06:
>  if I can get you to pull the latest from the main branch and press the up button, down button and my button when capturing the logs.

Done:
```
[{"encKey": 168,
  "address": 10173223,
  "rcode": 313,
  "command": "Up",
  "rssi": -43,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [100635,102375,104230,101027,1327,1790,100903,101321,9815,99278,101438,98172,66452,10277,17164,2517,2541,2496,2518,4852,1260,1316,1271,1313,1271,1310,621,667,609,661,1292,1295,640,642,646,643,1283,666,619,1320,612,664,1274,1310,618,678,613,685,1270,654,639,1283,1295,663,626,1292,1285,664,620,1306,1276,1299,642,650,618,678,610,684,1270,1308,1272,649,635,661,624,668,629,1312,620,668,610,688,1267,1293,1268,1322,638,637,640,672,619,657,1289,646,623,1309,646,643,1283,656,643],
  "time": "2023-04-06T13:45:08.389+0200"
},
{"encKey": 169,
  "address": 10173223,
  "rcode": 314,
  "command": "Down",
  "rssi": -43,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [621,96098,27352,101932,102416,102392,105477,100273,81935,471,820,578,4832,524,12361,81192,10304,17140,2547,2494,2527,2503,4860,1266,1303,1277,1306,1272,1311,618,674,1265,648,645,654,640,657,620,1322,1264,1294,640,657,1275,674,613,659,647,643,634,1308,1270,1307,624,668,630,660,1267,668,632,1308,1270,1306,625,671,1275,1289,624,684,1265,1294,636,660,1274,1300,641,648,1296,1287,1294,659,614,654,642,1299,631,664,1273,1311,618,674,1266,1298,1284,1305,1274,1311,1269],
  "time": "2023-04-06T13:45:09.242+0200"
},
{"encKey": 169,
  "address": 10173223,
  "rcode": 314,
  "command": "Down",
  "rssi": -44,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [637,26936,2542,2514,2506,2524,2513,2520,2512,2519,2515,2528,2492,2537,2495,2535,4825,1282,1290,1302,1284,1295,1287,643,646,1289,649,618,676,636,658,619,1320,1260,1317,614,660,1272,676,611,660,645,644,632,1312,1270,1308,623,668,629,660,1281,668,633,1294,1268,1310,625,671,1277,1284,644,654,1265,1322,635,661,1265,1294,640,656,1270,1309,1274,650,25,637,666,624,1320,614,665,1275,1301,641,639,25,1294,1286,1296,1302,1267,1298,1279],
  "time": "2023-04-06T13:45:09.413+0200"
},
{"encKey": 170,
  "address": 10173223,
  "rcode": 315,
  "command": "My",
  "rssi": -46,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [644,73442,102241,103877,100920,102405,102523,109843,37254,10308,17165,2497,2530,2507,2524,4838,1266,1321,1260,1307,1272,1309,1294,1287,1270,1310,1272,675,614,659,647,642,633,1312,1269,656,641,1307,1274,661,614,670,623,656,634,1315,629,659,1283,1293,639,657,618,670,635,662,1277,646,636,661,624,1316,614,659,647,645,1281,657,642,646,637,1311,619,673,614,660,647,643,1283,1306,621,656,1292,644,649,659,608,1314,630,660,642,649,619,677,613,684,623,666,609,662,626,665],
  "time": "2023-04-06T13:45:10.227+0200"
}]
```

rstrouse, 2023-04-06:
Just to be sure these are pulses from your remote.  Is that correct?

nymanjens, 2023-04-06:
> Just to be sure these are pulses from your remote. Is that correct?

Yes that is correct

rstrouse, 2023-04-06:
Can you replicate these same commands with ESPSomfy RTS.  I think that is the exact pattern being transmitted just at a different address.

nymanjens, 2023-04-06:
These commands sent from another ESP with ESPSomfy RTS 1.4.3:
```
[{"encKey": 162,
  "address": 892541,
  "rcode": 130,
  "command": "Up",
  "rssi": -19,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [8306,2586,2575,2572,2524,4475,1275,1301,1263,1294,614,659,626,668,1256,1288,1268,1283,641,647,643,653,616,667,601,671,1247,1317,1270,1309,603,650,643,655,613,668,616,671,1247,1314,620,644,629,636,642,680,590,669,637,641,642,648,618,676,615,666,591,691,592,669,636,644,640,649,1269,638,638,1288,1274,662,616,1293,625,657,1268,1301,627,637,1293,642,619,658,615,685,594,693,581,1337,1244,652,634,662,598,1315],
  "time": "2023-04-06T19:53:03.087+0200"
},
{"encKey": 162,
  "address": 892541,
  "rcode": 130,
  "command": "Up",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [31041,2587,2531,2597,2523,2572,2558,2574,2521,2613,2520,2571,2571,2577,2519,4485,1240,1293,1294,1289,621,665,634,640,1289,1295,1240,1289,651,638,636,656,617,669,610,660,1256,1317,1267,1299,628,636,635,659,617,669,612,660,1257,1295,636,635,654,644,618,683,592,667,623,667,627,636,640,655,613,670,611,684,590,672,622,661,615,667,1266,644,631,1289,1274,648,613,1342,592,671,1262,1283,643,655,1266,647,644,626,636,667,623,669,600,1307,1273,647,633,636,616,1304],
  "time": "2023-04-06T19:53:03.158+0200"
},
{"encKey": 163,
  "address": 892541,
  "rcode": 131,
  "command": "Down",
  "rssi": -19,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [102313,109263,69906,8945,89540,2626,2523,2582,2568,4461,1266,1277,1272,1312,616,671,589,688,1243,667,606,682,617,653,622,643,631,1318,620,643,1281,1307,1245,661,616,680,592,670,637,1288,618,668,1261,1293,1268,1312,1246,675,615,1267,647,668,1261,641,638,1295,636,638,1290,634,637,1287,1247,1316,1244,677,636,639,619,645,632,691,602,674,594,1326,1251,1293,1267,649,622,1323,599,663,623,657,1266,1299,625,662,616,655],
  "time": "2023-04-06T19:53:04.522+0200"
},
{"encKey": 163,
  "address": 892541,
  "rcode": 131,
  "command": "Down",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [31043,2592,2542,2565,2559,2564,2548,2593,2519,2589,2538,2577,2546,2596,2520,4477,1262,1282,1268,1313,614,668,610,662,1267,669,602,681,619,653,620,646,627,1314,620,643,1254,1337,1245,661,618,656,617,669,635,1289,616,668,1260,1284,1292,1287,1269,653,611,1289,641,673,1240,652,643,1297,616,643,1299,628,636,1294,1269,1280,1268,669,628,637,640,629,639,674,609,660,612,1318,1257,1289,1278,659,598,1296,631,687,602,673,1245,1313,621,643,628,664],
  "time": "2023-04-06T19:53:04.656+0200"
},
{"encKey": 164,
  "address": 892541,
  "rcode": 132,
  "command": "My",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [102425,63502,8951,89591,2616,2522,2599,2542,4475,1242,1292,1290,1298,604,655,1261,1315,632,661,1252,1296,1264,674,594,1324,1264,645,632,661,613,664,607,1305,1264,650,635,1288,1268,654,641,628,639,1296,641,631,1273,672,614,1311,591,670,1264,661,643,1282,616,679,1243,684,591,686,595,667,630,662,617,1306,1246,1290,1299,1280,618,668,618,645,633,639,641,669,1269,630,636,1287,1267,682,610,664,617,1302,1258],
  "time": "2023-04-06T19:53:05.854+0200"
},
{"encKey": 164,
  "address": 892541,
  "rcode": 132,
  "command": "My",
  "rssi": -19,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [664,30379,2595,2544,2598,2512,2580,2568,2566,2542,2592,2538,2569,2540,2590,2512,4481,1260,1315,1261,1290,648,635,1291,1295,610,660,1268,1321,1237,663,618,1301,1259,664,628,636,642,656,619,1294,1268,649,622,1322,1251,644,629,662,616,1293,643,653,1268,648,617,1303,641,625,1268,658,642,1306,595,674,1264,650,619,674,639,642,616,645,644,1281,1269,1296,1286,1276,622,654,640,656,620,642,629,663,1279,644,624,1287,1273,670,613,659,614,1317,1240],
  "time": "2023-04-06T19:53:06.058+0200"
}]
```

rstrouse, 2023-04-06:
Unfortunately, v1.4.3 didn't capture the timings outside of the standard frame and it dropped the carrier as soon as it completed the frame.  v1.4.6 holds the carrier through the inter-frame silence and detects the pulses in between the frames.  You should do your testing with the firmware from the repo.  These timing changes only seem to affect certain motors.

nymanjens, 2023-04-06:
For clarity: The ESP that is capturing the logs is at 1.4.6 from the `main` branch. It's only the ESP that is sending the message that is the old version (that ESP is a bit tricky to connect via USB for some reason). But if both sender and receiver must be at 1.4.6, I'll check it tomorrow

rstrouse, 2023-04-06:
The 1.4.3 code will not capture the space in between the frames so they both need to be 1.4.6.  You can download the code from the repo and do not need to be connected to USB.  Simply upload it from the updates tab.

nymanjens, 2023-04-07:
> they both need to be 1.4.6.

Done. Here is a ESPSomfy 1.4.6 talking to another ESPSomfy 1.4.6:

```
[{"encKey": 165,
  "address": 892541,
  "rcode": 133,
  "command": "Up",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [103209,101066,107668,100292,32263,67443,1222,100730,102405,102363,76114,10463,7377,2570,2525,2614,2513,4470,1281,1283,1294,1292,594,668,1254,1314,1265,637,641,1303,620,658,640,634,641,644,1275,1291,1269,654,614,1314,617,647,613,662,641,646,1267,1312,1269,1279,642,647,616,650,636,661,624,643,624,656,641,642,649,616,649,661,618,665,644,620,632,665,1268,642,647,1288,1250,669,611,1314,618,671,1257,1293,642,630,1260,655,642,650,634,636,632,646,641,1280,1268,667,628,635,636,1288],
  "time": "2023-04-07T08:29:13.905+0200"
},
{"encKey": 165,
  "address": 892541,
  "rcode": 133,
  "command": "Up",
  "rssi": -19,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [28120,2581,2562,2578,2545,2589,2528,2562,2579,2573,2520,2590,2535,2572,2574,4452,1270,1286,1276,1271,641,667,1269,1281,1267,668,601,1308,621,665,604,683,636,624,1270,1306,1272,648,634,1289,620,644,630,664,617,678,1243,1312,1240,1319,610,685,596,666,630,640,636,652,646,643,612,661,653,640,620,659,637,663,615,646,613,662,1281,667,607,1317,1241,652,636,1287,643,648,1285,1274,623,654,1267,671,595,676,636,639,643,622,657,1287,1245,677,636,640,637,1303],
  "time": "2023-04-07T08:29:13.906+0200"
},
{"encKey": 166,
  "address": 892541,
  "rcode": 134,
  "command": "Down",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [109237,95516,102451,102469,12488,483,1529,3775,90702,102580,38432,10494,7329,2581,2569,2569,2527,4465,1291,1295,1269,1272,649,636,1289,642,645,1288,1241,667,631,660,636,1273,649,636,636,654,620,669,611,661,1269,667,602,658,616,1312,618,676,613,666,628,641,620,661,639,659,1247,652,635,1287,643,647,1284,625,643,1301,615,653,1245,678,640,1283,1270,1285,1265,649,637,659,627,667,603,683,611,663,592,1321,1256,1292,1261,663,644,1283,615,679,641,628,1262,1279,644,657,616,661],
  "time": "2023-04-07T08:29:17.515+0200"
},
{"encKey": 166,
  "address": 892541,
  "rcode": 134,
  "command": "Down",
  "rssi": -19,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [28126,2584,2573,2545,2549,2590,2540,2571,2569,2566,2521,2602,2525,2567,2567,4460,1263,1296,1255,1310,616,678,1264,648,617,1301,1258,664,632,636,641,1304,620,658,614,659,627,667,602,684,1243,670,622,654,615,1312,618,672,588,688,617,643,643,657,615,661,1274,622,658,1291,620,644,1281,653,620,1295,640,629,1272,674,639,1287,1245,1289,1271,674,637,639,620,644,629,688,591,676,617,1297,1268,1293,1262,654,617,1320,617,655,618,668,1260,1281,643,658,615,661],
  "time": "2023-04-07T08:29:17.705+0200"
},
{"encKey": 167,
  "address": 892541,
  "rcode": 135,
  "command": "My",
  "rssi": -19,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [114300,5545,96883,5500,103946,100859,104988,99822,102404,102405,102419,102365,102441,71676,10471,7353,2561,2540,2597,2543,4452,1268,1312,1269,1280,642,647,1267,659,643,625,643,622,657,1293,1246,675,640,1284,1267,1286,625,641,1298,1288,1271,663,590,1304,1294,1290,594,676,640,643,626,643,1295,640,619,1296,641,630,1273,647,642,1307,593,668,1264,677,592,669,637,635,642,670,620,1281,1282,1296,1262,1298,628,635,636,639,642,674,612,662,1251,668,610,1312,1268,630,635,686,596,1293,1288],
  "time": "2023-04-07T08:29:18.968+0200"
},
{"encKey": 167,
  "address": 892541,
  "rcode": 135,
  "command": "My",
  "rssi": -19,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [647,27448,2610,2554,2547,2581,2566,2554,2571,2524,2611,2518,2573,2573,2555,2544,4477,1261,1314,1259,1289,595,678,1267,650,643,632,637,660,615,1300,1253,669,620,1318,1262,1289,627,636,1261,1314,1263,652,645,1270,1268,1316,604,684,591,685,621,644,1255,661,647,1296,639,633,1251,670,636,1286,618,669,1261,655,618,670,608,662,641,647,615,1304,1268,1294,1285,1291,620,631,640,660,615,670,623,655,1266,672,594,1302,1293,643,625,637,640,1280,1269],
  "time": "2023-04-07T08:29:19.139+0200"
}]
```

nymanjens, 2023-04-08:
FYI: I soldered a board with the shortest possible wires, but it didn't make any difference alas

rstrouse, 2023-04-08:
Yesterday, I made a little tweak and published 1.4.6 based on the timings you provided.  Are you using the most recent firmware?

rstrouse, 2023-04-08:
Maybe getting another set of eyes on it to see if you can see something I cannot might help.  Below is the typical output from your remote.  I will go through it so you know what you are looking at.  You will find the code for what is transmitted starting on Transceiver::sendFrame(...) line 1637 of Somfy.cpp.
```
  "pulses": [100635,102375,104230,101027,1327,1790,100903,101321,9815,99278,101438,98172,66452,10277,17164,2517,2541,2496,2518,4852,1260,1316,1271,1313,1271,1310,621,667,609,661,1292,1295,640,642,646,643,1283,666,619,1320,612,664,1274,1310,618,678,613,685,1270,654,639,1283,1295,663,626,1292,1285,664,620,1306,1276,1299,642,650,618,678,610,684,1270,1308,1272,649,635,661,624,668,629,1312,620,668,610,688,1267,1293,1268,1322,638,637,640,672,619,657,1289,646,623,1309,646,643,1283,656,643]
```
The pulses are duration in microseconds.  The first series of 13 pulses are simply interrupts that are triggered from noise or other environmental factors.  I don't have as many as you but I still have them.  The next two pulses are the wakeup and silence pulse.  I have tried a variety of pulses here but the consensus from most remotes is that the values closely match `10920` and `7375`.   You sometimes get a longer silence from your remote but I do not think that is important.  The wakeup pulse appears to be the important one.

Next there are 4 pulse for an initial frame and 7 pulses for a repeat (this differs with an 80-bit protocol to 24 and 14 respectively).   Theses pulses are 4 * 640 in length or 2,560us in length.  These are the hardware sync where the hardware is pulsed so that the receiver and set up to listen for the payload.

The next pulse should be in the neighborhood of 4450us and is the software sync.  The software sync tells the motor that we are starting at a bit value 0.

The rest of the pulses are the payload in OOK Manchester encoding where each transition represents an alternating bit either high or low starting with the most significant bit.  The one thing that I am the most sure about is this encoding.  If it were incorrect it would fail the checksum and the frame would be invalid.  A length of ~1280us represents a flip of a bit from the previous bit and a length of ~640us represents a repeat of the previous bit.

A total of 56-bits are written until the end of the frame, then we get into an inter-frame silence or cooling off period until the next repeat or the next initial frame.   You can see these timings show up on the next frame when you either press a command very quickly or hold the button down so it makes a repeat.

This is a bit of a weird thing.  If the last payload ended in a 0 then the remotes appear to send an extra pulse.  ESPSomfy RTS replicates this and you will see it as a ~640us on the next frame.  If it does not then it simply goes quiet for ~27434us.  If there is a repeat then the hardware sync will start directly after this silence.

I have stared at these numbers until I go cross-eyed and cannot see a discrepancy.  Perhaps you can see one.  If we are in agreement with it maybe there is some setting on the driver that I can look at.  I think we are solid on RX.

nymanjens, 2023-04-08:
> Yesterday, I made a little tweak and published 1.4.6 based on the timings you provided. Are you using the most recent firmware?

I am now. It may be a little bit less flaky, but the difference falls within the margin of error.

nymanjens, 2023-04-08:
> I have stared at these numbers until I go cross-eyed and cannot see a discrepancy.

Thanks for the explanation. I started looking into the numbers and will continue when I have some time.

nymanjens, 2023-04-08:
FYI: My very preliminary conclusion. Based on the snippets from above, the wakeup pulses are:

* remote
  - 10277
  - 10304
  - 10308
* esp
  - 10516
  - 10455
  - 10517

and silence pulses are:
* remote
  - 17164
  - 17140
  - 17165
* esp
  - 7310
  - 7352
  - 7327

which already seems like a promising start to tweak the numbers on the ESP to match the remote more closesly

rstrouse, 2023-04-08:
The numbers I put in there were an average from the ones I have collected.  You can give it a try and see if it makes a difference.  You will find the first value on line 1647 of Somfy.cpp and the silence value on line 1654.  There is about a 30% tolerance as I understand it.

nymanjens, 2023-04-10:
Great news!

I gathered some more data, and confirmed that the numbers from https://github.com/rstrouse/ESPSomfy-RTS/issues/8#issuecomment-1500975679 are consistent for both ESP and the remote.

Representative extract from that the gathered data: These are the first 11 pulses starting from the wakeup pulse:

```
*** ESPSomfy ***

10.479 7.343 2.571 2.553 2.561 2.560 4.448 1.283 1.309 1.247 1.287 ms
10.424 7.353 2.576 2.568 2.542 2.545 4.460 1.269 1.322 1.260 1.280 ms
10.413 7.364 2.573 2.551 2.562 2.556 4.473 1.262 1.279 1.270 1.312 ms
10.437 7.329 2.593 2.541 2.579 2.571 4.448 1.274 1.287 1.269 1.307 ms
10.419 7.378 2.560 2.537 2.597 2.548 4.458 1.268 1.284 1.290 1.278 ms
10.443 7.350 2.579 2.548 2.587 2.540 4.448 1.262 1.291 1.299 1.291 ms
10.415 7.349 2.582 2.545 2.554 2.568 4.470 1.271 1.287 1.271 1.305 ms
10.436 7.340 2.589 2.529 2.571 2.579 4.439 1.289 1.280 1.266 1.294 ms

*** Remote ***

10.275 17.201 2.519 2.514 2.505 2.526 4.839 1.266 1.300 1.283 1.282 ms
10.282 17.194 2.505 2.518 2.529 2.503 4.837 1.287 1.300 1.279 1.289 ms
10.302 17.182 2.519 2.523 2.496 2.509 4.860 1.256 1.304 1.275 1.310 ms
10.276 17.193 2.513 2.524 2.512 2.519 4.842 1.266 1.293 1.266 1.321 ms
10.312 17.170 2.532 2.524 2.511 2.521 4.839 1.258 1.316 1.261 1.315 ms
10.312 17.156 2.525 2.536 2.501 2.531 4.831 1.261 1.314 1.280 1.291 ms
11.179 83.807 5.831 103.981 56.395 38.630 7.391 91.984 3.012 0.074 15.334 ms
10.287 17.171 2.536 2.521 2.505 2.501 4.861 1.260 1.316 1.259 1.315 ms
10.280 17.171 2.532 2.526 2.504 2.504 4.851 1.268 1.312 1.269 1.306 ms
10.271 17.196 2.525 2.504 2.510 2.528 4.837 1.263 1.314 1.258 1.318 ms
10.281 17.175 2.522 2.537 2.498 2.518 4.853 1.261 1.304 1.274 1.310 ms
10.290 17.180 2.524 2.543 2.498 2.518 4.851 1.261 1.307 1.272 1.311 ms
10.278 17.185 2.519 2.524 2.516 2.516 4.822 1.295 1.287 1.270 1.307 ms
10.289 17.161 2.523 2.527 2.509 2.526 4.837 1.260 1.315 1.259 1.315 ms
```

I then updated the code to have ESPSomfy use the exact timing from the remote:

```
diff --git a/Somfy.cpp b/Somfy.cpp
index 443d3b6..59ce8db 100644
--- a/Somfy.cpp
+++ b/Somfy.cpp
@@ -1644,15 +1644,13 @@ void Transceiver::sendFrame(byte *frame, uint8_t sync, uint8_t bitLength) {
     // Wake-up pulse
     //Serial.printf("Sending wakeup pulse: %d\n", sync);
     REG_WRITE(GPIO_OUT_W1TS_REG, pin);
-    delayMicroseconds(10920);
-    //delayMicroseconds(9415);
+    delayMicroseconds(10783);
     
     // There is no silence after the wakeup pulse.  I tested this with Telis and no silence
     // was detected.  I suspect that for some battery powered shades the shade would go back
     // to sleep from the time of the initial pulse while the silence was occurring.
     REG_WRITE(GPIO_OUT_W1TC_REG, pin);
-    delayMicroseconds(7357);
-    //delayMicroseconds(9565);
+    delayMicroseconds(17177);
     //delay(80);
   }
   // Depending on the bitness of the protocol we will be sending a different hwsync.
@@ -1666,7 +1664,7 @@ void Transceiver::sendFrame(byte *frame, uint8_t sync, uint8_t bitLength) {
   }
   // Software sync
   REG_WRITE(GPIO_OUT_W1TS_REG, pin);
-  delayMicroseconds(4450);
+  delayMicroseconds(4850);
   // Start 0
   REG_WRITE(GPIO_OUT_W1TC_REG, pin);
   delayMicroseconds(SYMBOL);
```

I've tested about 40 interactions now, and it has been working 100% perfect!!!

rstrouse, 2023-04-10:
Awesome.  I know this might be a pain in the butt but can I get you to try to pin down whether there is one value or the other that has an impact on the frame detection.  For instance change the software sync, wakeup silence, and wakeup pulse separately.  These are separate things and I need to get a handle on their tolerance.

Also what is the model number of your remote.  Surprisingly, I have found some differences between them

nymanjens, 2023-04-12:
>  I get you to try to pin down whether there is one value or the other that has an impact on the frame detection.

I'll do that one of these days

rstrouse, 2023-04-12:
Update to v1.4.7 and let me know if you have solid performance.  I think our issue was the need for the software sync to hold for one more symbol before starting to pulse the frame.

nymanjens, 2023-04-13:
FYI: Now that it works (at least 99%), I'm going to put away my experimental motor setup until we move to our new house (in about 6 months). If you want me to test anything before I stow everything away in boxes, now is the time :-).

closed 2023-04-13 by nymanjens

rstrouse, 2023-04-13:
If I am reading the table correctly that pretty much matches up with the things that I have seen and makes total sense.  The Software Sync for the motor is counting out after the end of the hardware sync.  It needs 7 * 640 for the software sync + half a symbol for 0 to fall in the start.

Good luck on the move and thanks for the help.

nymanjens, 2023-04-14:
Thanks so much for all your help and sharing and maintaining this awesome project!