rstrouse/ESPSomfy-RTS #9: Is it possible to change the frequency to 868.95 MHz?

MishoOo123, 2023-03-24:
Hello, I have a Somfy Situo 1 Var iO Pure II remote control that operates at a frequency of 868.95 MHz. I was wondering if there is a way to change the frequency within this program, so that I can use my remote control? Any help or suggestions would be greatly appreciated.

rstrouse, 2023-03-26:
Unfortunately I suspect there are some other fundamental differences with the IO protocol that go beyond the frequency.

rstrouse, 2023-04-16:
I am closing this for now.  I don't have any plans to support the IO functions for Somfy at the moment.  If by chance my shades were of this type, I would have beat the tokens to make those a reality.  Perhaps in the future if more info on the protocol is revealed I might dabble in it.

closed 2023-04-16 by rstrouse