rstrouse/ESPSomfy-RTS #10: 5 channel remote, and rolling code clarification.

Anashost, 2023-03-26:
I've just found out about this project  and it's really looks promising. I've couple of questions tho.

- I have 3 big balcony shades that i control using one remote that support up to 5 channels. So will that work with your controller, if i change the channel on the remote every time i pair a shade.
[image]

- after setting up the whole thing correctly (hopefully), will i be able to control the shades using the remote over time. As the rolling code is changing while using the shade how will the original remote keep track of this if im not using it. (Or It might be that i didn't understand the concept about the rolling code at all 😅).

- If i somehow missed up the rolling code settings, you mentioned something about resetting the motor memory. Is that the same procedure of resetting the motor using the plug:
unplug 2 sec -> plug 10 sec -> unplug 2 sec -> plug, until the motor jogs.
Or do you mean something else! That red bold (Beware, Danger banner) kinda freaked me out 😅.

Again thank for putting this together, really appreciate this project, and hopefully i will try it out, as i already ordered the hardware.

rstrouse, 2023-03-26:
Yes you can use the remote and ESPSomfy together.  Each defined shade has its own rolling code and will not interfere with the remote.  You then link the existing remote and it will keep the position even if you use the remote.

Anashost, 2023-03-26:
Thanks for clearing things up.

Anashost, 2023-03-26:
Do you have an answer regarding the last point, the motor memory resetting. I just want make sure not to ruin my shade in case i messed things up.

rstrouse, 2023-03-26:
By reset I mean the sequence you are referring to for most motors.  Mine are hardwired and require a whole lot of ladder work and unwiring.

Anashost, 2023-03-26:
Thanks a lot, thats a relatively easy task for me.

rstrouse, 2023-03-29:
I am going to close this issue for now.  If you have any other questions of issues open another issue.

closed 2023-03-29 by rstrouse