rstrouse/ESPSomfy-RTS #13: New Installation Using 1.4.5 Unable to Get IP Address on Local Network

FlaMike, 2023-04-03:
My transceivers arrived (early!) on the slow boat--bought 2, just in case...

I wired up a new 30 pin ESP Wroom-32 to the transceiver, loaded Somfy-RTS onboard.bin, version 1.4.5 using the  ESPHome Web installer.  I disconnected the device from my PC & powered it back up.  I connected to the ESP Somfy RTS network & went to 192.168.4.1 & hit the gear icon.  Scanned &  found my network & entered the password & hit save.  I got the "waiting" spiral dots & that was  about it...  I disconnected from the network, exited the broweser, & did a hard reboot of the ESP.

The device shows up on the router, but no IP address is assigned.  Unable to ping the device by name, of course, so I was dead in the water.    

I redownloaded the firmware & tried again from the beginning & had the same experience.

I am going to try with the previous version & am hoping there's just a bug in 1.4.5.

All suggestions are welcome to get past this roadblock :-)  Thanks!!!!!

rstrouse, 2023-04-03:
If it is not giving a DHCP IP address out of the router this seems to indicate an issue with either the passphrase or the router is rejecting it.  Do you by chance have 802.11b enabled on your router?

FlaMike, 2023-04-03:
Yeah, it's the strangest thing I've ever seen!  I have Ubiquiti equipment (USG router, 3 APs, 2 switches, 1 24 port and 1 24 port POE, running the Unifi controller in a docker container).  DHCP is running & I have about 70 WiFi clients, and a couple of dozen wired clients, all with static IPs, assigned through the router.

The controller shows the ESP on the correct network by name (I have 4 running with various firewall rules for each), the ESP's MAC address & signal strength are shown, but *no* IP address--the field is blank!  I've never experienced that before.  I tried 2 different networks to no avail, that's why I reached out to you.  I really don't want to bother you but, it's so strange!  I will try again tomorrow with a guest network because it has a short password, but I copy & paste the network passwords from a pdf list.

So bizarre!  I would think the router would reject the device if the password is wrong and not show the device in the client list.

I'm hoping it's just a case of the Mondays & all will be well tomorrow!  Hopefully, I'll touch base with you tomorrow & let you know what a fool I was today.

FlaMike, 2023-04-03:
The only other thing I can  think of at the moment is I've been up since 2 AM CST & am probably too punchy to be doing anything that requires a synapse.

rstrouse, 2023-04-04:
The esp32 has a limit on the length of an SSID to 64 and the passphrase can be no longer than 32.  If you are still running 802.11b frames on your network, I would turn them off.  It is eating your bandwidth.

It is possible that the Ubiquiti router is somehow seeing 192.168.4.1 and even though it released the address it still thinks it is there.  Are you using a /21 subnet and a 198.168.x.x mask?

FlaMike, 2023-04-04:
I think you just nailed it...  My "normal" passwords for the networks are 63 characters long!  Not an issue for Tasmota, ESPHome, etc., but apparently,  it *is* for an ESP!  I had no idea!  I will try using the guest network---it has a much shorter password.  Even if I  beef it up a bit, I will limit it to < 32 characters.

Thank you so much for your help!  I will give it a shot tomorrow (gotta get my 3 hours of sleep) & let you know about my success.

I *so* appreciate your time, patience, willingness to help, and all!  You are a rock star!!!!

FlaMike, 2023-04-04:
BTW, not using a /21 subnet.  All are /24.  I think the password limit is the problem.  Fingers crossed!

rstrouse, 2023-04-04:
So did you get it to connect?  Sleeping is for wimps!

rstrouse, 2023-04-04:
This is an issue with the guest network and the purpose of a guest network.  You will likely have an issue with HA if you try to cross that boundary into your guest network.

I can try to expand the password length by allocating the space to eliminate the chance of an overrun.  It only needs this while it is first connecting to the network.  I used the idf limits for the host name when allocating the space for the passphrase.  It may use the same restrictions as the SSID for its passphrase instead of the host name restriction.  At some point it becomes impractical though.

FlaMike, 2023-04-04:
I thought it being on a guest network could cause HA challenges.  

I could reduce the length of the IOT network--it has only a handful of devices I'd need to reconfigure, as opposed to my NOT network.  Alternatively, I may be able to create a firewall rule that lets HA listen & respond to the guest network (if Unifi allows rules to be applied to a guest network).

Please let me know if you make the change or if I need to.

Thanks from the always awake world :-)

rstrouse, 2023-04-04:
I will make those changes and it should not impact anyone else.  The only issue may be internally with the idf.  I cannot find any reference to what that max password length is.  Are you using a WPA or WPA2 network?  Some of these limits may be contained in the TKIP section of the idf and will only show their head on WPA/TKIP.

FlaMike, 2023-04-04:
The networks are WPA2.  I believe the passwords I am using currently are 63 characters long--more or less the limit of WPA2, I think.

Thanks again for your help!  If you let me know when I can pull the update, I'll give it a shot & let you know of my success :-)

rstrouse, 2023-04-04:
Actually, I believe the limit is based on the AES not the WPA2 and the original TKIP had a 31 character limit.  Nobody should be using WEP or WPA anymore but if they do they won't get past the WiFi setup in their router with a 63 character passphrase.  I just need to figure out if my storage can handle it.

rstrouse, 2023-04-04:
Pull the [SomfyController.ino.esp32.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin) file and the [SomfyController.littlefs.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin) file from the links provided and upload *.ino.esp32.bin file as the firmware and the *.littlefs.bin file as the application from the Updates tab.

The WiFi passphrase length has been increased.

FlaMike, 2023-04-04:
My response aren't showing up on github, so here they are, possibly duplicates nevertheless...

Irrigation/irritation what's the difference anyway?  I consider myself to be a bidet, of course!  At your service...

Working on the network change now, running 1.4.6 on the ESP.

The device is now on the preferred network (NOT) and I can access it from my PC!   Fabulous!!!  Assume all will be good for HA now as well.

Still tuning the radio.  Don't know if I'm going to get strong signals, but I have sufficient parts to build 2 if needed.

Thank you so much for your help & very quick responses & software updates!!!  As I said before, you're a rock star!!  Woo hoo.

I am getting the following error message on the WiFi tab in the app. 500: Network HTTP error.  The other tabs appear to be functioning correctly and I do get signal from the Somfy remote on the Somfy tab.

closed 2023-04-04 by FlaMike

rstrouse, 2023-04-04:
You probably need to hard refresh your browser after installing the Application.