rstrouse/ESPSomfy-RTS #14: No RSSI value when pressing remote button

bradphillips921, 2023-04-04:
I followed the hardwire guide and setup guide exactly. I have the same ESP32 board and transceiver from the hardware guide and have it wired per the guide.  I have the firmware installed and the ESP32 connected to my network. I'm able to access the GUI, I've set my RX bandwidth and Frequency deviation per the setup guide. But when I press the button on my Somfy remote I don't get any value for the RSSI. I've tried multiple Somfy channels, checked my wiring, etc. But still can't get an RSSI number when pressing the button on the remote while right next to the transceiver.

rstrouse, 2023-04-04:
Which CC1101 transceiver and ESP32 are you using?

FlaMike, 2023-04-04:
I have the same challenge (onto the next one??).  Hardware setup is exactly like yours, Dupont cabling until setup is complete.  ESP is a WROOM-32 Devkit1 Version 1 and the transceiver is a E07-M1101D-SMA with the duck antenna, it's connected to the board, of course.

I tried both 50 and 80 bits for the radio and the radio is saved and enabled.

I am running ESP Somfy RTS version 1.4.5.  The initial config I am using is exactly what's in  the Wiki for software configuation: 56 bit radio, RX bandwidth 96.96, Frequency deviation 11.43, TX power 10 db.  Pins set as in the Wiki.

rstrouse, 2023-04-04:
In your setup change TX to GPIO12 and RX to GPIO13.  These appear to be documented backwards.  I added the ETH boards and it has different pin assignments.  I think I got lost in the markdown.

FlaMike, 2023-04-04:
Will do.  Thanks!

rstrouse, 2023-04-04:
Just change it in the pin assignments page do not move the pins around.

FlaMike, 2023-04-04:
👍

FlaMike, 2023-04-04:
I changed the pin assignments on the Somfy Settings page, saved the radio, and  rebooted the device from the General tab.  The "Radio Not Initialized" banner is not disappearing when I save the radio.  I tried a hard reboot as well, but the radio does not appear to be initializing.  Possible dead board from the far east?  I have another I can try...

rstrouse, 2023-04-04:
Double and triple check your wiring.  The pins numbers are from the top of the board not the bottom.  The way that thing is marked is a menace.  Pin 1 is the left bottom pin with the pins facing away from you,

rstrouse, 2023-04-04:
> I apologize for being a major irrigation to your backside. I have no excuse for my failure to follow directions.

No worries sometimes I stick my backside out the window at the carwash when I need a good irrigation.

bradphillips921, 2023-04-05:
Changing those TX and RX pins didn't work for me. I have TX set at GPIO-12 and RX set at GPIO-13. I also tried the opposite with TX set to GPIO-13 and RX set to GPIO-12. Neither worked.

GPIO-13 is connected to terminal 3 on the transmitter. GPIO-12 is connected to terminal 8 on the transmitter

I'm using this transmitter
https://www.amazon.com/dp/B09CTRYV4L?psc=1&ref=ppx_yo2ov_dt_b_product_details

And this ESP board
https://www.amazon.com/dp/B09XDMVS9N?psc=1&ref=ppx_yo2ov_dt_b_product_details

It's wired per the instructions on the hardware page.
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device

rstrouse, 2023-04-05:
That looks like a fine bill of materials for this project.  Did the transceiver come with an antenna if it did not you need one?  

When you are connecting the transceiver the pin numbers are from the top where the pins are facing away from you.  Please make verify your connections with this in mind.

bradphillips921, 2023-04-05:
Good catch! I don't have the male pin in the antenna. I just ordered one. I'll report back if that worked.

rstrouse, 2023-04-05:
I thought that might be an RP-SMA antenna.  Make sure you get one tuned to 433Mhz.

bradphillips921, 2023-04-07:
I bought this antenna. Unfortunately, it's still not working.
https://www.amazon.com/dp/B072Q7319V?psc=1&ref=ppx_yo2ov_dt_b_product_details

Here are my settings
https://drive.google.com/file/d/1xGhHDzgG24nKCQGnnWmuu4ZhpVwab_20/view?usp=sharing

Anashost, 2023-04-07:
> I bought this antenna. Unfortunately, it's still not working. https://www.amazon.com/dp/B072Q7319V?psc=1&ref=ppx_yo2ov_dt_b_product_details
> 
> Here are my settings https://drive.google.com/file/d/1xGhHDzgG24nKCQGnnWmuu4ZhpVwab_20/view?usp=sharing

my esp32 boards has dedicated rx and tx pins other than GPIO-12 and GPIO-13.  so if your board have them try that. in the espsomfy settengs, these pins usually translate to TX: GPIO-01 and RX: GPIO-03.

rstrouse, 2023-04-08:
You can use the pins suggested by Anashost if you like but you should simply change your setup to TX = GPIO-13 and RX = GPIO-12.  I need to go in and edit the docs to reflect the pictures.  The pins can be switched so long as GDO0 is connected to the TX and GDO2 is connected to the RX pin.

bradphillips921, 2023-04-08:
It's still not working for me. I've confirmed that the transceiver pin 3 is connected to GD00-TX (labeled D13 on the board) and transceiver pin 8 is connected to GD02-RX (labeled D12 on the board). I have TX = GPIO-13 and RX = GPIO-12 on the configuration screen.
https://drive.google.com/open?id=1xGhHDzgG24nKCQGnnWmuu4ZhpVwab_20&authuser=bphillips%40phillipsstafford.com&usp=drive_fs

But I'm still not getting an RSSI signal when I push the remote button within 6 inches of the transceiver antenna.

rstrouse, 2023-04-08:
Change the pins to the following and try again.

[image]

Anashost, 2023-04-08:
This is the diagram of his esp32

[image]

I think 12 & 13 doesn't work as TX & RX in his case (they didn't work for my board either).
So maybe he should use the pins that supports that, so:
01 & 03 (the ones i used)
Or
16 & 17

I've an ESP32-WROOM-32D with SILAPS chip , and i had to use pins 01 & 03 as tx rx. I'm using the blue transceiver with the spring antenna. these are my configs:

[image]

I might be wrong here as I'm new to this world, and still learning. Hope this helps.

rstrouse, 2023-04-08:
I too am new to this world and the SiLabs folks don't make it easy with all the versions of the ESP32 and very little information on what the differences are.  Near as I can figure all the current WROOM models being shipped are using the ESP32-D0WD chip and all of them are dual-core.  This means that an ESP32S and an ESP32D are exactly the same.  At one point the S was supposed to mean single core and D was supposed to be dual code.  To make matters worse, they also say DEVKIT01 on them.

I listened to so many YouTube videos from the guy with the Swiss accent that I even started to pick up his accent but was still confused by the models.   From a layperson perspective, I would say the board pictured above is the same one I have.

The software does not use a serial UART so it never initializes any of the pins to the UART.  Using pins 1 and 3 are supposedly the same as using 12 and 13 or any of the other un-reserved pins.

Here is Andreas going through them.
https://www.youtube.com/watch?v=LY-1DHTxRAk

bradphillips921, 2023-04-10:
I tried the recommendation from Anashost but that didn't work.

I set my pin numbers how you showed in your latest post, rstrouse. Still doesn't work.

rstrouse, 2023-04-10:
@bradphillips921 I ordered the antenna that you purchased above and it works fine with several different transceivers that I tested.  I also made an attempt to review the wiring from your pictures and it looks correct.   I am starting to think you have a bad transceiver.  I have bought several of these (simply because they show up in a day) and only had one that was bad and acted like yours is acting.

https://www.amazon.com/dp/B01DS1WUEQ?psc=1&ref=ppx_yo2ov_dt_b_product_details

bradphillips921, 2023-04-10:
Yep, I'm leaning towards that as well. I just ordered transceivers from the link you posted. I should have them next week. I'll test with those.

rstrouse, 2023-04-10:
@bradphillips921 keep that antenna.  I did a bit of tuning on one of my devices and it increased the receive strength by about 30% even when it was already spectacular.

reindruu, 2023-04-11:
It looks like I have the same setup and the same problem.
Not picking up any buttons from Somfy remotes.
I bought them at Ali (maybe that is the problem?!?):
ESP 32 board 30 Pin CP2102 chip:
[https://www.aliexpress.com/item/1005001627605230.html?spm=a2g0o.order_detail.order_detail_item.8.31a96d76agcUp8]
Transmitter incl. antenna:
[https://www.aliexpress.com/item/33059343309.html?spm=a2g0o.order_detail.order_detail_item.12.31a96d76agcUp8]

Configuration is ok, Radio is initialized, header disappeared.

[image]
[image]
[image]

Anashost, 2023-04-11:
I just want to mention that i got mine from Aliexpress as well, but everything works great 👍🏻

ESP32-WROOM-32D - SILAPS CHIP
https://a.aliexpress.com/_EQlnZHl

CC1101 (Blue with spring antenna)
https://a.aliexpress.com/_EJV3HKB

You may got a faulty board, or it's something else!

Anashost, 2023-04-11:
@reindruu If you're going to order the same boards as mine, then here is the pins I've used.
[image]

Configurations:
[image]

rstrouse, 2023-04-11:
Are those RTS remotes or IO remotes?  If it is IO, they are not 433Mhz protocol RTS.

reindruu, 2023-04-11:
> 
@rstrouse 
Oops................ 
You are right, completely.
[image]

rstrouse, 2023-04-11:
I'm so sorry.  Unfortunately, this solution will not work for IO motors and your journey to control your own blinds will not end here.  What a terrible trend all this cloud based IoT stuff is.

reindruu, 2023-04-12:
> I'm so sorry. Unfortunately, this solution will not work for IO motors and your journey to control your own blinds will not end here. What a terrible trend all this cloud based IoT stuff is.

Big disappointment. I did not know there were different motors. Life's full of disappointments but is was an experience.
Thanks for your huge good manual and guidlines.
Maybe your next project is the IO motor................

rstrouse, 2023-04-16:
@bradphillips921 if your new hardware does not solve the problem please open another issue.  This one got a bit hard to follow and thanks for the antenna tip.

closed 2023-04-16 by rstrouse