rstrouse/ESPSomfy-RTS #15: Can't pair the blinds

miczlo, 2023-04-05:
I wanted to start by saying thanks for doing this as I think it's an amazing project.
So far I have had an ESP just shorting correct DI on the remotes to have my blinds controlled in HA but this would be a game changer for me.
Unfortunately I'm struggling with pairing process.
I have quintuple checked my wiring and tried swapping RX/TX without changing them in the configuration and swapping RX/TX and then also swapping them in the config.
Basically I put my blinds into pairing by holding the button on the back, they jog and blink green but then nothing ever happens when I try to pair them from ESPSomfy.
I do see all the commands I send from my remote and I do see them regardless of the RX/TX situation.
Strangely (for me) I'm still seeing the commands sent in the logs even if I swap the TX/RX without changing them in the config
I'm using the blue E07-M1101D with a Wifi ESP32-WROOM-32.
It's a 56 bit motor. I have tried 2 different blinds - same results.
Any solutions for debugging/diagnosing?

rstrouse, 2023-04-05:
First, what type of remote do you have and which model?  If I could get you to click on the configure transceiver button and open the logs screen.  Then press the prog button on the back of your remote 3 times (do not hold it in).  This should result in three prog commands showing up in the logs.  Press the copy button on the screen and paste those here in a code block.

miczlo, 2023-04-05:
Thanks for taking the time.
It's a Telis Situo RTS Pure. Tried 2 of these. 
[{"encKey": 160,
  "address": 9408556,
  "rcode": 7688,
  "command": "Prog",
  "rssi": -49,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [2494,2559,2469,2537,4797,1295,1286,1295,1261,665,643,644,639,661,626,620,669,633,635,665,619,1324,1255,693,578,1334,1246,1326,1244,697,585,1335,620,675,608,663,1273,1297,629,648,1269,684,601,1348,582,660,1265,699,602,1309,617,670,609,685,1268,1297,614,689,618,672,609,664,1251,670,636,665,626,645,624,1307,1265,1329,611,669,1255,664,623,665,630,661,631,665,617,647,636,664,624,669,628,658,609,664,644],
  "time": "2023-04-05T16:12:03.845+0100"
},
{"encKey": 161,
  "address": 9408556,
  "rcode": 7689,
  "command": "Prog",
  "rssi": -48,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [2495,2544,2460,2534,4811,1305,1265,1307,1258,668,630,659,631,650,643,1301,1265,672,611,1318,1244,659,633,1306,1275,640,641,647,615,723,585,1339,570,700,608,687,1247,1324,1244,670,609,688,600,1321,608,689,1251,670,634,1289,643,645,636,664,1250,1321,607,665,644,645,636,664,1250,671,635,665,625,645,627,1307,1247,1321,629,671,1272,664,625,645,623,682,608,667,618,669,636,638,628,669,623,655,633,655,642],
  "time": "2023-04-05T16:12:04.299+0100"
},
{"encKey": 162,
  "address": 9408556,
  "rcode": 7690,
  "command": "Prog",
  "rssi": -50,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [2495,2535,2471,2540,4802,1296,1285,1295,1287,639,644,645,635,1316,1268,658,609,678,617,1298,1265,643,660,1316,593,695,584,711,1225,697,584,1336,594,699,609,688,1224,696,612,1314,1244,683,606,1313,626,679,1261,644,626,1309,621,667,631,661,1281,1295,635,641,646,644,630,661,1281,643,631,660,631,650,644,1303,1252,1319,633,651,1268,660,633,650,618,675,635,641,648,641,630,661,631,650,644,651,636,640,647],
  "time": "2023-04-05T16:12:04.746+0100"
}]

rstrouse, 2023-04-05:
Thanks for the output.  Is this remote currently paired with the motor?

miczlo, 2023-04-05:
> 

Correct. When I hold the prog button by blind jogs happily as well it's just like if it wasn't getting prog from the transceiver

FlaMike, 2023-04-05:
Good Morning, my automation BFF :-)

I am also having problems pairing shades.  When I press the program button on the remote (a Situo 5 RTS Pure II) the shade jogs.  I then press the "prog" button on the web page.  The pairing screen goes away, but the shade does not jog.  I've unpaired and attempted to re-pair the shade several times, to no avail.  It's currently un-paired.  Below is the result for my remote when I used the procedure you recommended for miczlo above.  I'm using a blue E07-M1101D with a Wifi ESP32-WROOM-32

Separately, I seem to be unable to get a signal stronger than about -75 dBa even when very close to the transceiver.  I've tried with both of my slow boat models with the duck antenna.  Assuming I get the pairing issue resolved, do you recommend I try one of these (which appears to be the same but who knows???).

https://www.amazon.com/dp/B01DS1WUEQ?psc=1&ref=ppx_yo2ov_dt_b_product_details

Thanks again for your help & support!

```
[{"encKey": 166,
  "address": 2445379,
  "rcode": 1174,
  "command": "Prog",
  "rssi": -69,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10432,7173,2515,2518,2496,2516,4848,1270,1287,1271,1308,623,666,1257,684,620,1296,640,657,620,671,1257,1303,1274,659,640,1293,615,684,621,643,632,661,1280,1297,1291,1273,633,662,640,651,1270,1312,1274,652,636,641,647,643,631,661,630,1299,1282,1308,620,654,1265,670,621,1310,1268,669,609,662,642,648,644,1303,614,668,1278,1283,1269,661,641,650,643,1303,614,668,628,663,1265,1293,638,663,622,666],
  "time": "2023-04-05T11:10:44.440-0500"
},
{"encKey": 167,
  "address": 2445379,
  "rcode": 1175,
  "command": "Prog",
  "rssi": -68,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [145237,10426,7193,2503,2521,2504,2505,4862,1260,1309,1247,1311,619,670,1263,677,618,674,613,1313,618,675,1267,1301,1255,659,649,1293,1266,1298,633,661,1268,1320,1266,1301,628,660,1267,669,627,1283,1290,653,614,684,620,644,631,662,638,1296,1264,1305,624,659,1291,645,625,1309,1273,649,637,661,624,668,631,1284,650,639,1283,1308,1271,662,601,665,624,1310,621,666,631,660,1267,1319,613,660,623,669],
  "time": "2023-04-05T11:10:44.687-0500"
},
{"encKey": 168,
  "address": 2445379,
  "rcode": 1176,
  "command": "Prog",
  "rssi": -69,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [146333,10426,7194,2507,2508,2508,2502,4836,1262,1316,1254,1309,1272,1311,618,669,634,662,625,641,625,682,1266,1295,635,661,623,666,1254,1309,618,676,614,667,1275,1298,630,667,1270,660,614,1317,1267,1300,1281,658,622,666,609,661,641,650,619,1304,1293,1295,612,684,1246,677,614,1311,1268,661,641,650,618,676,614,1311,618,676,1267,1302,1253,684,620,669,609,1314,619,670,609,660,1293,1293,615,683,621,643],
  "time": "2023-04-05T11:10:44.938-0500"
}]
```

rstrouse, 2023-04-05:
Do you happen to know the model number and brand of the motor?

FlaMike, 2023-04-05:
It's a Somfy but I don't know the model.  And the motors are not available to examine physically without me tearing out a bunch of finished framing & drywall.  (They're hidden above the windows--built into the wall above with a slot for the shades).  I could reach out to the supplier & installer if that helps.

rstrouse, 2023-04-05:
Are they battery powered?

FlaMike, 2023-04-05:
Nope, 120 volts, powered by an outlet in the pocket above the window.  The brand of the shades is Alta.  I'll hit their web site & see if I can find out more info on the motors.

miczlo, 2023-04-05:
> Are they battery powered?

Battery powered ROLL-UP 28 WIREFREE

FlaMike, 2023-04-05:
Might be Sonesse® RTS motors???

rstrouse, 2023-04-05:
I have paired many Sonesse motors at this point and over 20 Wirefree.  What am I missing?  Mine are the LT60s.  @FlaMike from the link you provided I apparently have purchased 8 of those and they have been great with the exception of this one.  You can see the flaw if you look closely at the metallic rectangle on the right.  It has a dent in the crystal.
[image]

rstrouse, 2023-04-05:
Just for shucks set the address of the shade to 4624451 and then try to pair it.

FlaMike, 2023-04-05:
Will do in a little while...  I'm curious, where did you get the magic number from?  Not seeing it in the code block.  Eventually I'll have 6 more to pair, so I'm interested in being independent to pair them.

rstrouse, 2023-04-05:
I pulled that from a known remote that has paired 3 different shades.  Just trying to eliminate any possibilities.

FlaMike, 2023-04-05:
Understood.  I'll get back to & let you know how it goes.

Perhaps I selected my only "problem" first.  🤞  It's the only one that
controls a single shade...

FlaMike, 2023-04-05:
No love pairing with magic code 4624451.  The behavior was the same as before.  I pressed the program button on the remote, the shade jogged, I pressed the program button on the ESPSomfy-RTS web page, the shade did not jog, but the page went back to the edit shade page.  The former "pair shade" button then read "unpair shade."  To try to keep everything in synch, I then attempted to unpair the shade.  I pressed the remote's program button, the shade jogged, I pressed the "unpair" button on the web page, & then was presented with the edit shade page.

rstrouse, 2023-04-05:
Yeah I am currently revamping that page.  It was a pipe dream that the pair always occurs on the first try.  It did for me repeatedly but perhaps that is not the case for everyone else.  I am going to change the screen so that it shows an "It Jogged" button that you can press to dismiss the screen.   If it didn't actually pair then pressing the cancel button will close the screen without marking it as paired.

Then perhaps we can try multiple times without unpairing.

FlaMike, 2023-04-05:
Good plan!  Thank you.

BTW, I tried using a much larger antenna that I bought for use with an ESP32 that's outside (concrete & rebar construction, not WiFi friendly).  It's a male with the pin in the middle & same threading.  No love--couldn't get any signal at all from the remote.  Probably the impedance is wrong or or it's tuned for WiFi (really?).  Anyway, something is way off for the little transceiver.  Just a quick experiment that proved I know very little about antennas.  I was not surprised by the results but I had to try since it was sitting on my desk.

rstrouse, 2023-04-05:
You cannot use an antenna that is tuned for 2.4Ghz on a 433Mhz radio.  Antennas are frequency matched.  It will resonate at the wrong frequency and receivers will not be able to hear it.  The wavelength is important.

rstrouse, 2023-04-05:
Ok so pull the bin files from the main repo.  I have not created a release for this yet.  After you update the application you will need to hard refresh your browser to see the changes.  I also added a 0 bit to the end of the 56 bit protocol so that the motor recognizes that it has read the last bit.  I don't know if that makes a difference but my US telis remote sends this bit.  I just had to look for it.

FlaMike, 2023-04-06:
Dude!  What's mine say?  Sweet!  What's mine say?  Dude!...

Quick question (can't get to GitHub until late morning tomorrow). Does the
image I need to pull & try have the 1.4.6 long passcode code in it?  As you
know, I need the code for the network I'm using for ESPSomfy-RTS, HA, &
more.

Thanks & I'm looking forward to checking out the extra zero bit in the
frames.

BTW, I should receive the Amazon transceiver in about a week.  Hoping for
much better range, but will keep plugging along until it arrives.

rstrouse, 2023-04-06:
Yes it has all the most recent changes.  I just have not generated a release for it just yet.  I have installed 3 neighbors homes with this solution as well as my own and the range is stupid good.

FlaMike, 2023-04-06:
Sweet!  I shall get on it as soon as I can tomorrow & touch base with you 😁

miczlo, 2023-04-06:
> Ok so pull the bin files from the main repo. I have not created a release for this yet. After you update the application you will need to hard refresh your browser to see the changes. I also added a [...]

Is it the SomfyController.ino.esp32.bin?
I've just tried it but the ESP32 doesn't seem to happy with it and I can see the onboard.bin.
I'm not getting the wifi network and that's the logs:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

rstrouse, 2023-04-06:
@binu02 which remote do you have?  Also, do you know what model of motors you are pairing with?

rstrouse, 2023-04-06:
On the back of the remote does it indicate whether it is US or any other country?

FlaMike, 2023-04-06:
Just uploaded both files.  I selected the SomfyController.ino.esp32.bin file & got the following message:  This file is not a a valid firmware binary file.

Not sure how to proceed from here.  I welcome your thoughts.

FlaMike, 2023-04-06:
I understand that.  I am trying to follow rstrouse's suggestion from yesterday but the update procedure failed.

FlaMike, 2023-04-06:
I think I pulled the wrong file...  Will try again with the correct one!

rstrouse, 2023-04-06:
Use the *.ino.esp32.bin to update the firmware and *.littlefs.bin to update the application.  You will get this message if you use the wrong file.

FlaMike, 2023-04-06:
Just realized that!!  Sorry.  I guess that's what happens when I do this while on a phone call :-(

rstrouse, 2023-04-06:
Thanks I just wanted to be sure because mine says US on it.  I am trying a couple of things.  I think there is potentially an oddball thing with the Glydea motors with regard to the last bit transmitted.  I have already figured out that the expectation is that the protocol while it is only 56 bits for some motors the bit after that must be pulled to 0.  That is what fixed @miczlo's issue.  Perhaps with Glydea it needs to hold the carrier during the interframe silence period.

On 80 bit protocols it is a bit different but so far I only have one user with 80 bit remotes.  These appear to be the remotes that have the nudge thumbwheel on them.

binu02, 2023-04-06:
An option to copy the code from the log file and paste it on the corresponding Up, My, and Down buttons can be an option. can it be implemented?

FlaMike, 2023-04-06:
It's working!!!!  I finally had an opportunity to update 1.4.6 (extended password beta) using the repo updates you created for pairing "problem" blinds & I was able to pair my blind on the first try!!!  And even with my crappy transceiver signal strength (around -80dBm) I am able to open & close the blind with the transceiver at the opposite end of the house.  So fun (at last!).

Question:  I use only one remote per room--each remote controls only the blind(s) located in that room.  Assuming I retain that setup, is there a need to link remotes?  I ask because I tried to link the remote I used when pairing the motor & nothing happened when I pressed the buttons on the remote--the link screen never disappeared.  I'm  guessing I don't need to link the remote used when pairing the motor to ESPSomfy-RTS, but I would appreciate if you could clarify that I don't need to link remotes in my situation.

If it's all good as-is, I will pair the other shades & move on to the HA integration.

rstrouse, 2023-04-06:
@binu02 I am a bit confused with your output.  There should be a bit that enables the carrier between each command if the firmware file was uploaded or the radio failed to send the pulse.  Can I get you to give me a log where you simply hold the up button down for a full second.  This should result in a few repeats in the log.

rstrouse, 2023-04-06:
@FlaMike yes you should link the remote.  The reason for this is so that ESPSomfy RTS knows about any uses of such remote.  I will do some testing to make sure I didn't induce any timing issues with the link detection by resurrecting our missing bit.  You should be able to use the remote or ESPSomfy RTS and is will keep track of its position.

You should know that the reason you have a remote for each room and there isn't a bunch of remotes that can control motors throughout the house, is because the Somfy remotes have really poor range.  That typically isn't an issue though because you hang a remote on the wall in each room.  I'm sure much of this is so you don't have to change batteries all the time.

The transceiver is able to hear signals down to about -106dBm but even so that is about 30% better than the RX strength from the remote to the motor.  You will find that the TX strength is really powerful.  I can move my patio shades from about a 1/4 mile away which is excessive.  If it comes down to ESPSomfy RTS not hearing the remotes from a distance you can install more than one device.  This will work beautifully with HA.

binu02, 2023-04-06:
> @binu02 I am a bit confused with your output. There should be a bit that enables the carrier between each command if the firmware file was uploaded or the radio failed to send the pulse. Can I get [...]

[{"encKey": 173,
  "address": 14173654,
  "rcode": 1523,
  "command": "Up",
  "rssi": -46,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10454,7502,2509,2526,2514,2520,4867,1271,1307,1265,1296,1292,666,626,1313,1272,657,640,1305,644,635,635,652,1293,661,642,648,642,1279,1291,1295,640,652,641,652,1288,1296,1307,652,640,1278,1293,644,656,662,641,644,620,1309,650,642,654,639,1291,1293,1291,1296,632,661,630,660,627,663,640,649,1293,635,667,644,619,1310,649,642,1281,1303,1297,1289,646,634,1311,644,626,1313,645,645,1286,649,642],
  "time": "2023-04-06T22:17:16.575+0400"
},
{"encKey": 173,
  "address": 14173654,
  "rcode": 1523,
  "command": "Up",
  "rssi": -44,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [631,26985,2530,2507,2513,2527,2517,2543,2490,2553,2495,2538,2494,2537,2502,2555,4812,1311,1290,1285,1298,1276,665,642,1278,1308,642,634,1285,667,627,661,639,1295,635,659,640,643,1307,1292,1270,666,625,643,651,1312,1293,1282,651,641,1283,1287,644,658,636,666,644,645,1285,649,644,654,636,1292,1294,1292,1293,634,660,651,640,653,637,631,670,1270,665,645,645,635,1285,669,627,1288,1296,1306,1275,649,661,1291,643,627,1313,644,646,1286,648,643],
  "time": "2023-04-06T22:17:16.718+0400"
},
{"encKey": 173,
  "address": 14173654,
  "rcode": 1523,
  "command": "Up",
  "rssi": -45,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [630,26979,2523,2542,2483,2553,2513,2521,2512,2518,2515,2531,2519,2517,2496,24,2544,4827,1287,1299,1298,1282,1294,661,640,1294,1285,648,642,1280,666,649,639,635,1291,643,648,661,642,1275,1300,1282,643,659,639,654,1269,1310,1275,652,661,1282,1294,636,640,670,644,634,641,1302,648,633,659,642,1291,1287,1294,1286,647,643,657,635,641,646,25,644,659,1291,642,649,636,659,1290,628,663,1294,1291,1291,1295,633,661,1291,645,634,1286,643,650,1289,645,645],
  "time": "2023-04-06T22:17:16.861+0400"
},
{"encKey": 173,
  "address": 14173654,
  "rcode": 1523,
  "command": "Up",
  "rssi": -45,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [633,26953,2519,2536,2501,2558,2484,2541,2487,2545,2490,2550,2492,2538,2495,2541,4826,1283,1302,1298,1289,1296,639,644,1298,1291,643,653,1288,645,645,634,662,1277,649,634,659,652,1290,1288,1293,635,661,640,645,1272,1309,1296,636,641,1297,1280,653,643,649,636,660,654,1289,639,656,620,670,1284,1299,1278,1287,649,637,667,625,667,627,662,633,1298,655,636,656,644,1272,666,625,1296,1287,1297,1309,638,644,1296,639,643,1300,639,641,1294,635,666],
  "time": "2023-04-06T22:17:16.998+0400"
},
{"encKey": 173,
  "address": 14173654,
  "rcode": 1523,
  "command": "Up",
  "rssi": -44,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [619,26967,2521,2554,2474,2558,2497,2516,2524,2543,2483,2545,2483,2540,2514,2546,4811,1305,1271,1311,1269,1314,620,669,1285,1275,652,662,1284,643,656,636,630,1296,658,636,655,644,1299,1291,1268,663,633,650,640,1310,1276,1298,641,642,1302,1289,644,653,635,641,646,644,1309,640,644,645,635,1290,1295,1289,1297,654,636,643,645,643,658,639,655,1295,633,640,653,644,1298,641,649,1292,1287,1295,1286,649,643,1279,655,643,1299,640,649,1292,636,642],
  "time": "2023-04-06T22:17:17.137+0400"
},
{"encKey": 173,
  "address": 14173654,
  "rcode": 1523,
  "command": "Up",
  "rssi": -44,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [622,27015,2517,2538,2496,2542,2498,2543,2506,2516,2511,2550,2488,2548,2483,2545,4844,1268,1312,1270,1309,1291,647,633,1285,1296,661,641,1298,626,661,641,649,1294,639,631,665,646,1287,1300,1278,640,665,645,633,1286,1291,1310,641,646,1286,1299,628,663,632,649,641,657,1286,644,649,635,658,1300,1278,1291,1294,639,654,644,648,635,660,649,642,1280,653,646,646,636,1309,643,655,1290,1269,1310,1274,650,660,1293,643,628,1312,645,646,1285,649,643],
  "time": "2023-04-06T22:17:17.279+0400"
},
{"encKey": 173,
  "address": 14173654,
  "rcode": 1523,
  "command": "Up",
  "rssi": -46,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [630,26996,2503,2565,2483,2526,2514,2545,2490,2524,2521,2537,2494,2538,2499,2533,4836,1286,1289,1306,1290,1279,658,639,1298,1295,639,629,1313,645,645,635,661,1278,649,635,660,650,1293,1291,1294,633,636,666,645,1271,1309,1293,635,655,1293,1283,648,643,655,642,654,645,1271,641,651,642,651,1288,1296,1308,1275,648,661,642,652,639,629,663,633,1298,655,640,630,646,1295,641,655,1292,1286,1299,1280,654,642,1299,631,645,1295,661,638,1293,635,642],
  "time": "2023-04-06T22:17:17.417+0400"
},
{"encKey": 173,
  "address": 14173654,
  "rcode": 1523,
  "command": "Up",
  "rssi": -45,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [645,26965,2503,2554,2495,2538,2497,2538,2501,2545,2506,2529,2511,2524,2511,2522,4854,1273,1307,1293,1271,1291,643,650,1314,1269,657,639,1304,645,634,640,653,1295,635,666,625,643,1301,1291,1295,634,664,644,627,1313,1289,1286,648,642,1281,1292,645,658,635,666,644,646,1286,649,643,655,635,1291,1294,1293,1289,638,656,650,643,652,636,655,642,1296,633,647,644,657,1286,644,649,1287,1296,1307,1276,648,661,1292,641,628,1313,646,645,1287,645,643],
  "time": "2023-04-06T22:17:17.556+0400"
}]

FlaMike, 2023-04-06:
I thought I should link the remotes, but when nothing happened when I attempted to do it, I began to second-guess my understanding of how you keep track of each shade's position regardless of how it is controlled.  When you have an update I'll be happy to be your beta platform.

BTW, I showed my wife the one blind I have paired up & your animation of the shade's position & she was duly impressed :-)  Kudos to you!  (Recall I've been using a ZRTSI to enable shade automation until now.  While they're minimally functional & have a less-than-ideal mechanism to integrate with HA (z-wave), not knowing the blind's position and the inability to keep blinds, remotes, & HA in synch is a hassle.  Not to mention the outlandish price of the ZRTSI boxes!  My wife's sisters will be here in about a week for a visit.  I look forward to not having shade synchronization challenges!

Your project is a big win for us & I thank you so much for sharing & supporting it!!!!

rstrouse, 2023-04-06:
@binu02 there is our missing bit.  That ~630 timing in front of each of the repeats is what I was looking for.  I assume you only have one ESPSomfy RTS device so I will look at these timings to see if anything is off.

binu02, 2023-04-06:
> @binu02 there is our missing bit. That ~630 timing in front of each of the repeats is what I was looking for. I assume you only have one ESPSomfy RTS device so I will look at these timings to see if anything is off.

Hi, Yes I have only one device with me now.

rstrouse, 2023-04-06:
Just wait until you install the HA integration.  It is stupid simple.  The feeling you should experience is... It's done now what do I do?  For me, I showed it to a few neighbors and ended up with a desk full of ESPSomfy RTS devices.  You have been warned!

FlaMike, 2023-04-06:
Just installed it.  I now have one device in it--the blind we both know & love to death.  I'll get the others going & switch my node red automation targets for opening & closing the shades as well as the entities in the UI to your system.  Then the ZRTSIs shall be history.  Can't imagine who else could use the bloody things.  Ugh!

rstrouse, 2023-04-06:
After you add and pair up the motors you can simply click reload on the integration and it will pick all those up as well.

rstrouse, 2023-04-06:
@binu02 I went through your timing pulses and they are amazingly consistent with those put out by ESPSomfy RTS.  There must be something else going on here.  Can you take a couple of pics of your connections for me?  Also, just to be sure you did not reassign the shade address to the same address as your remote.

FlaMike, 2023-04-07:
I have paired all of my blinds except for 1 of them.  I believe it has a different motor because it's a very small window.  It's still powered by 120 volts & is on the same remote as the others in that room Situo 5 RTS Pure II.

I don't know the model of the motor but I can reach out to the installer, if that will help.  Below is the output from the logs when I did the 3 program button press, as you've requested from others.  

If you have any suggestions, I'll all in.

Many thanks!

```
[{"encKey": 161,
  "address": 1061441,
  "rcode": 485,
  "command": "Prog",
  "rssi": -69,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [503,3055,50,43652,65714,236162,10458,7062,2513,2519,2509,2499,4847,1275,1299,1267,1316,614,659,646,643,632,661,1291,1295,636,661,1267,1303,1275,647,636,1312,1267,1311,620,669,1260,1309,1273,661,624,1296,637,661,1275,649,636,1313,615,674,1266,1298,631,659,1267,666,630,661,633,1298,1282,659,619,1296,639,654,1271,657,636,677,618,647,637,1311,1268,1314,1271,654,639,658,621,1295,1292,1296,1259,1304,1275,1309],
  "time": "2023-04-07T12:24:44.542-0500"
},
{"encKey": 162,
  "address": 1061441,
  "rcode": 486,
  "command": "Prog",
  "rssi": -69,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [137638,10430,7072,2509,2527,2514,2498,4841,1267,1323,1258,1315,627,659,631,667,1268,1287,643,647,636,660,1276,1297,1266,668,629,661,640,1279,649,660,615,654,1297,1287,1270,678,611,659,648,643,629,661,640,650,643,1279,653,645,1273,1310,642,652,1264,648,646,654,639,1283,1297,638,652,1292,636,660,1277,646,635,660,625,644,651,1287,1297,1277,1291,644,632,659,640,1300,1281,1282,1295,1287,1295,1285],
  "time": "2023-04-07T12:24:44.782-0500"
},
{"encKey": 163,
  "address": 1061441,
  "rcode": 487,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [133056,10468,7050,2513,2545,2509,2497,4847,1272,1307,1276,1298,630,664,621,654,1290,640,650,1284,647,642,1283,1291,1296,635,640,671,622,655,638,1308,621,655,1290,1296,1283,641,644,653,635,1284,1294,635,652,1292,636,663,1276,1298,640,652,1269,661,632,651,642,1307,1276,646,635,1287,643,648,1289,650,644,652,635,633,647,1294,1292,1296,1285,653,616,649,636,1309,1268,1313,1269,1308,1276,1298],
  "time": "2023-04-07T12:24:45.027-0500"
}]
```

rstrouse, 2023-04-08:
I just published v1.4.6.  This includes some bit changes to the transmit protocol that may allow some troublesome shades to pair.

FlaMike, 2023-04-08:
Even got the bad boy blind working late yesterday.  It's the only blind with a different (smaller) motor because the window & shade are so small.  It uses the same remote as  the others in the room.  I'm guessing it has  a weaker radio than the other shades--often sketchy reception with the ZRTSI.  

I moved the ESPSomfy-RTS to the location where I expect it will hang out forever; closer to the center of the house than my office.  Bad boy paired up immediately from there & linked without a problem as well.  All shades & HA automatons working as planned.  Solid performance, even for bad boy!  ZRTSIs are history--even offered them to the installer, 'cause I don't ever care to see them again :-)

I will migrate the ESP & transceiver to a PCB as in your "More Polished Version" in the hardware section.  Project box should be here early next week & everything else is ready & waiting for me to do some soldering.  I hope to be extra careful with the wiring layout since it's a mirror image of the top of the board, as you noted in the Wiki.

Were there any changes from the most recent beta I was playing with (update files that were in the main repo a couple of days ago)?  My ESP is showing 1.4.6, but I don't know if the packaged version differed from the beta.

Thanks again (and again!) for a killer & incredibly well-supported project!!!   Happy Easter!

rstrouse, 2023-04-08:
Yes there were some tweaking of the transmit bits and a fix for the My favorite setting.  How many shades did you pair?  Also, if you have any remotes where you grouped up the shades do not forget to link these as well.  ESPSomfy RTS will keep track of that as well.

FlaMike, 2023-04-08:
I have a total of 8 shades paired up & fully integrated in HA, complete with Node Red automatons & in the HA UI too.  7 shades are in the main house & 1 in the pool house.  All appear to be working perfectly.  I have not controlled any as groups within the RTS world, but will keep you suggestion in mind.  Thee only "double" shades are on the same channel, courtesy of the installer. 
 (Really, why mess with the remotes once your system has been installed?  So much better & easier to deal with.  Remotes are going into drawers except for the guest rooms.)

I will pull the 1.4.6 update files.  I thought you might have made a few more mods since I uploaded the repo updates, but wanted to confirm.

FlaMike, 2023-04-11:
Cool!  Thanks--just ordered the antenna.  Not sure I need the extra range, but a little bit of wiggle room never hurts.  May add to reliability.

Did you "re-tune" the transceiver after attaching it or just leave things as they were?

rstrouse, 2023-04-11:
I didn't need the extra range either but that is why monster trucks exist.  I did mess around with the tuning but it is very close to where it was.

FlaMike, 2023-04-11:
👍. Cool!  It should be here later this week.  I'll let you know how the
monster antenna compares in my setup.

Have a super day!

FlaMike, 2023-04-13:
Just added the "Monster Truck" antenna & without any additional tuning the signal went from -86dBm to -81 dBm.  Not a monster improvement, but half an order of magnitude--log scale, right?  Anyway, my build now looks like the John Holmes of the RTS world :-)

I may play with tuning later, but even at the current tuning I can reach all of the blinds in the house & the pool house, so I'm a happy camper.

Have a super afternoon (still morning in your world, I think.

rstrouse, 2023-04-13:
Yep still morning for a couple more hours.  That increase in performance while less than the improvement I got, is still respectable with a 3dbi gain.  Before adding the antenna I had to put my device in my office but after adding it I can push it back to my server rack with no dropped frames.

FlaMike, 2023-04-13:
I actually have a 5 dBm improvement--almost 1 dB per dollar spent :-)  
EZ PZ improvement & I'll take however many fewer dropped frames I can 
get.  The Monster antenna was a superb suggestion!

Thanks!  I think *now* you're on the verge of being able to enjoy the 
afternoon.  1:40 in the FL Panhandle--we're on CST, an open secret.

rstrouse, 2023-04-16:
I think we have gotten everyone paired in this issue.  Open a new issue if this is not the case.

closed 2023-04-16 by rstrouse