rstrouse/ESPSomfy-RTS #16: Wiring different than table

reindruu, 2023-04-07:
Hi,

I think the wiring is not the same as the text in the table.
The blue wire on PIN-8 from M1101D is connected to ESP-GPIO13 (according to photo).
PIN-8 is the RX pin according to table.
In the ESPSomfy settings RX is configured to GPIO-13
So I think the table is a little bit confusing.
[image]

rstrouse, 2023-04-07:
Just to be sure.  Are you are referring to the picture of the back of the board?  It will work either way just so long as you have matched the pin configuration with the way you have it wired.

I will make these be the same either way.

binu02, 2023-04-08:
> Just to be sure. Are you are referring to the picture of the back of the board? It will work either way just so long as you have matched the pin configuration with the way you have it wired.
> 
> I will make these be the same either way.

Hi,
Great, it started working after changing the TX and RX pin configuration to GPIO 13 and GPIO 12 respectively. Thank you all.
I have noticed that most of the time I have to press the button three or four times to make it work. I have also tried through integration also, but the result is the same.
I really appreciate your support in this regard.
Thanks a lot

rstrouse, 2023-04-08:
@binu02 there another user who has a similar issue #8.  Out of the 7 devices that I have created so far, none of them experience this behavior.   Interestingly, the other user has created 2 ESPSomfy RTS devices and while the two devices pick up the signal reliably between themselves, the motors don't always respond to the command.   

The range for every device I have created so far as well as quite a few others have been ridiculously far.  Try placing ESPSomfy RTS close to the motor and see if the reliability improves.  If it does then we need to be looking at the TX.  If it does not then perhaps there is something else going on.

reindruu, 2023-04-08:
> Just to be sure. Are you are referring to the picture of the back of the board? It will work either way just so long as you have matched the pin configuration with the way you have it wired.
> 
> I will make these be the same either way.

Yes, I'm referring to the picture of the back of the board. It will work if the configuration is also adjusted.
But I like it to follow the example :)

binu02, 2023-04-08:
> @binu02 there another user who has a similar issue #8. Out of the 7 devices that I have created so far, none of them experience this behavior. Interestingly, the other user has created 2 ESPSomfy [...]

To capture the signal, I have created one more device and I am pasting the log file here. In the below log, Rcode 659 works. After I sent the stop code, only the last Rcode 662 works. All stop codes above do not make any action.

[{"encKey": 163,
  "address": 374797,
  "rcode": 659,
  "command": "Up",
  "rssi": -68,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10458,7355,2550,2565,2570,2548,4472,1270,1305,1275,1270,639,639,640,643,1280,640,650,617,649,1286,640,644,627,636,1291,1295,1261,655,618,669,609,1313,619,644,633,660,1264,1294,633,660,1274,1272,659,615,644,625,1309,624,643,1301,1267,1270,655,636,26,640,650,621,647,1288,652,618,647,637,634,666,621,644,655,614,1285,1297,1287,625,644,1274,661,626,643,623,656,637,1279,1270,1291,1273,1300,1279,1268],
  "time": "2023-04-09T00:15:39.659+0400"
},
{"encKey": 163,
  "address": 374797,
  "rcode": 659,
  "command": "Up",
  "rssi": -69,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [28055,2609,2545,2550,2576,2558,2540,2597,2545,2549,2562,2561,2546,2590,2550,4464,1249,1322,1268,1294,633,639,641,623,1269,657,641,653,619,1298,640,640,644,646,1285,1265,1275,662,624,644,624,1307,623,664,603,656,1285,1270,656,663,1266,1269,611,661,641,646,1293,635,638,1291,1261,1280,643,658,613,662,621,669,1252,660,622,668,627,636,641,654,622,667,611,1288,1275,1296,616,668,1272,662,615,650,643,630,635,1289,1272,1284,1298,1281,1266,1293],
  "time": "2023-04-09T00:15:39.782+0400"
},
{"encKey": 164,
  "address": 374797,
  "rcode": 660,
  "command": "My",
  "rssi": -67,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10426,7345,2580,2545,2571,2557,4440,1293,1294,1261,1304,619,683,1243,1296,630,637,1266,1322,1263,627,649,665,610,661,615,655,643,650,633,642,639,1275,1274,662,600,669,647,636,636,1293,1267,1309,614,667,1258,1282,1268,1288,639,668,1253,1288,643,647,1261,1307,1271,663,610,1290,634,666,1252,1295,640,634,1275,1294,1266,1294,635,639,1258,1317,616,661,1274,646,613,1288,646,670,1260],
  "time": "2023-04-09T00:15:43.472+0400"
},
{"encKey": 165,
  "address": 374797,
  "rcode": 661,
  "command": "My",
  "rssi": -67,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10409,7389,2567,2523,2595,2549,4467,1275,1270,1292,1294,633,638,1267,1269,1309,625,643,1301,1263,642,624,656,640,654,620,643,630,1288,1290,1305,1266,617,649,662,615,655,618,1321,616,658,619,643,657,636,1261,1318,1266,1275,648,634,1267,1293,619,683,1242,1296,1256,659,622,1318,611,662,1274,1272,641,654,1246,1326,1267,1272,658,616,1264,1320,616,664,1268,660,616,1282,642,655,1267],
  "time": "2023-04-09T00:15:47.240+0400"
},
{"encKey": 165,
  "address": 374797,
  "rcode": 661,
  "command": "My",
  "rssi": -68,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [648,27417,2573,2571,2553,2543,2579,2548,2572,2569,2565,2525,2598,2552,2545,2574,4451,1283,1282,1268,1289,622,667,1281,1285,1268,659,615,1308,1271,637,616,676,618,646,613,661,642,1295,1280,1285,1242,685,640,626,643,622,657,1287,620,644,656,662,602,672,1245,1287,1274,1299,616,676,1268,1286,625,642,1274,1312,1269,635,636,1281,644,631,1284,1290,630,660,1258,1290,1284,1290,628,633,1284,1290,634,636,1293,642,626,1289,647,642,1252],
  "time": "2023-04-09T00:15:47.372+0400"
},
{"encKey": 166,
  "address": 374797,
  "rcode": 662,
  "command": "My",
  "rssi": -69,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10413,7355,2594,2549,2539,2585,4427,1309,1277,1271,1311,617,645,1263,666,618,1295,1293,1269,1287,635,622,668,628,639,637,1293,1269,664,610,1281,1294,634,640,647,643,653,610,661,623,649,645,1287,625,644,1275,1284,1295,1282,624,639,1276,1285,644,649,1290,1275,1274,664,616,1275,646,659,1267,1270,657,636,1265,1269,1310,1275,623,654,1290,1298,605,660,1267,668,602,1312,622,668,1254],
  "time": "2023-04-09T00:15:50.926+0400"
},
{"encKey": 166,
  "address": 374797,
  "rcode": 662,
  "command": "My",
  "rssi": -67,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [660,27402,2593,2565,2549,2573,2564,2543,2579,2549,2566,2543,2563,25,2551,2593,2546,4466,1268,1284,1266,1295,615,660,1272,649,636,1287,1268,1305,1276,620,630,664,615,678,643,1278,1268,644,626,1283,1292,648,624,638,635,677,619,644,613,666,636,1300,601,659,1266,1297,1279,1302,620,652,1266,1279,645,663,1267,1294,1261,656,619,1322,606,657,1268,1282,640,647,1267,1310,1245,1302,641,624,1268,1311,620,668,1254,661,622,1294,634,661,1251],
  "time": "2023-04-09T00:15:51.046+0400"
}]

rstrouse, 2023-04-08:
That is so very bizarre given the frames received between the two ESPSomfy RTS devices are complete and contain no weriness to them whatsoever.  Can I get you to capture the same sequence using your remote.  I assume the motor responds 100% of the time to the remote.

binu02, 2023-04-09:
> That is so very bizarre given the frames received between the two ESPSomfy RTS devices are complete and contain no weriness to them whatsoever. Can I get you to capture the same sequence using your remote. I assume the motor responds 100% of the time to the remote.

Hi, yes, my Somfy remote is working fine and not missing any pushes. I have attached the log file to show the sequence of operations for the Somfy remote
[From Situo 2.txt](https://github.com/rstrouse/ESPSomfy-RTS/files/11184870/From.Situo.2.txt)

011V32, 2023-04-09:
I had the same problems with ESP-WROOM-32 and LilyGO T-Internet POE.
This one didn't work (only receiving).
[image]

This one worked:
[image]

binu02, 2023-04-10:
> Hi, yes, my Somfy remote is working fine and not missing any pushes. I have attached the log file to show the sequence of operations for the Somfy remote [From Situo [...]

Hi all,

I have spent a significant amount of time troubleshooting the issue by performing tasks such as pairing/unpairing, adjusting the distance between the transceiver and the motor, downgrading the firmware to version 1.4.4, and adjusting transmitter power and radio settings. However, the issue still persists, and I find myself having to press the button 2 or 3 times for it to work properly. I would greatly appreciate any support to help resolve this issue.

Thank you and regards

rstrouse, 2023-04-11:
@binu02 having to press the button more than once is not normal.  There is another user that has tweaked the timings with success on the transmit.  

I am waiting to hear which remote they are using because the reported timings are different from what I am seeing.  I know you have a Situo 2 but does it say Pure or Pure II on the back of it or reference a country code?  If motors contain differences I can simply allow some tuning if I know what the differences are.

binu02, 2023-04-11:
> 

> @binu02 having to press the button more than once is not normal. There is another user that has tweaked the timings with success on the transmit.
>
> I am waiting to hear which remote they are using [...]

Hi, Behind my remote it mentioned "Pure". Here, I attached a few more data from my Somfy remote control.
[image]
[image]
[Data From Remote.txt](https://github.com/rstrouse/ESPSomfy-RTS/files/11205458/Data.From.Remote.txt)

rstrouse, 2023-04-11:
Mine is also a Pure and it has similar timings.  Are you running the most recent release?

How close was the remote to the ESPSomfy RTS device when you were pressing the buttons?  I see consistently strong signals in your output then right in the middle of it there is a -78dBm.  Like all of the sudden there is interference.

binu02, 2023-04-12:
Today, I tested my device on a roller Sonesse 50 RTS blind motor, and it worked perfectly without any missed pushes. At the same location, they also had a Glydea Ultra 60, but it only responded to 50-60% of the pushes and mostly missed the "My" button. I also tried using the PURE and PURE 2nd remotes, but the results were the same. I believe that the Glydea Ultra RTS may require some fine tuning in its code.

rstrouse, 2023-04-12:
I had been expecting that the Glydea and its clones were just a bit different.  Can I get you to update your ESPSomfy RTS device with this firmware and report back.  
https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin

Another user with these motors did some testing but I need to know which parameter the motors are sensitive to.  Sonesse seems to not care so much about the start of the frame but Glydea is a bit of an outlier.  I added a full symbol width to the software synchronization because on the surface, this appears to be the one value that may make sense from the changes tested by @nymanjens.

binu02, 2023-04-12:
> I had been expecting that the Glydea and its clones were just a bit different. Can I get you to update your ESPSomfy RTS device with this firmware and report back. [...]

I have uploaded and tested the file you provided, and it is working great!! with the Glydea ultra 60 RTS motor. The motor responds promptly to all commands without any delays or the need for repeat presses. I hope that it will work with roller blinds too. I don’t have a roller blind motor with me now to test. I would appreciate it if you could release the file as a new version. Thank you very much for your support.

rstrouse, 2023-04-12:
Yes it works great with Sonesse motors as well.  I will put together a release shortly and thank you for working through this with me.

rstrouse, 2023-04-16:
I am closing this for now.  If you experience any other issues please submit another issue.  I do have some documentation updates coming.

closed 2023-04-16 by rstrouse