rstrouse/ESPSomfy-RTS #17: Tilt time is not appropriate

011V32, 2023-04-09:
One time for the gradual change of the steps does not apply to me.
I determined the following values for my external venetian blinds.

kitchen 1
Drive up time: 31000ms
Drive down time: 30700ms
Steps up: 313
Steps down: 320

kitchen 2
Drive up time: 39000ms
Drive down time: 38200ms
Steps up: 397
Steps down: 408

Probably there is a better way to for configuration of the steps?

011V32, 2023-04-11:
Unfortunately I can't say for sure. I can only say that there are 2 outdoor venetian blinds from the same manufacturer. However, there are 7 years between installations of these 2.
If relevant I could try to get to the motors?

What I also asked myself: The tilting movement is made with the up and down movement. What is my problem if I always see 0%? (Did I choose the wrong type?)

011V32, 2023-04-14:
I made another video and sent it to you via PM.
I only know the tilt function in our area and there it is always the same.
The angle is adjusted with the up or down movement. When you go down, the slats close, when you go up, the slats open. It doesn't matter in which position. You can move up and down or make a step up or down...

rstrouse, 2023-04-16:
I am finishing off the integrated tilt function.  I'll let you know when it is complete.  After that I will expand the internal percentages to accommodate your nudge function.  Right now there are more steps than can be represented by an integer percentage.

rstrouse, 2023-04-20:
Install firmware v1.5.0 and the associated application file.  Then change the shade configuration to use `Integrated` as the tilt type.

011V32, 2023-05-08:
With v1.5.2 it is now correct. Thx.

closed 2023-05-08 by 011V32