rstrouse/ESPSomfy-RTS #18: Documentation of My-Position is missing

011V32, 2023-04-09:
I can't find any information about the My-position anywhere in the "Configuring the Software" documentation.
Maybe something could be added here?

rstrouse, 2023-04-09:
Here you go.

https://github.com/rstrouse/ESPSomfy-RTS#setting-a-favorite

011V32, 2023-04-10:
Thanks, unfortunately I overlooked that. I just searched the wiki.
Maybe you could adopt that in the wiki, or at least refer to it?

Furthermore, I find a long button press in a software not very intuitive.
(I know that this is based on the original control, but probably almost nobody knows that either...)

rstrouse, 2023-04-16:
The long button press is all we have at this point.  After I get v1.5.0 released I will be doing some documentation updates and putting a TOC outline on the main wiki.

closed 2023-04-16 by rstrouse