rstrouse/ESPSomfy-RTS #20: Clarification of pairing instructions

youngsyt, 2023-04-11:
Hi. I'm having a problem pairing to a Somfy motorised blind, Basically I don't understand what the first line of the pairing instructions wants me to do. It says "Open the shade motor memory with an existing remote". I'm using a single channel Somfy remote - Situo RTS. I've followed the instructions as far as pairing and have tuned the receiver. I have an RF Explorer spectrum analyser and can see that both the remote and RTS RF module are transmitting at approximately the same level and frequency. I'm also using the latest 1.4.6 update. Any help would be much appreciated and thanks for a great project.

youngsyt, 2023-04-11:
Thanks for the quick reply. I've been pressing the prog button on the back of the remote and the battery powered Somfy motor responds by jogging and the switch on the motor starts to flash green. When I press the prog button within the app, the Somfy motor does not respond. The RF module is about 2 metres from the motor, I'm using Dupont leads to connect the RF module to the ESP32 mounted on a bread board and am wondering if they are too long at about 20 cms although I don't see any corruption receiving the data from the remote. I don't have another RF module to try as they are still on the boat. So I may move on to the tidied up version with shorter wiring.

rstrouse, 2023-04-11:
Double check your wiring.  GDO0 (pin 3 on the transceiver) should be connected to the TX Pin.  If you followed the guide that will be GPIO 13.  GDO2 should be connected to RX (GPIO 12) or not connected at all.  If you omit this wire you simply set RX and TX to GPIO 13 or 12 depending on where you wired it.  The transceiver can receive from either pin and the software controls which mode it operates in.

rstrouse, 2023-04-11:
Yeah I need to update the pictures.  You can build it either way and is just so happens that the picture was built differently than the table.

youngsyt, 2023-04-11:
Unfortunately I now have another problem. I have three blinds all bought at the same time. All three are outwardly identical with the same Somfy battery powered motor. Two of them are now paired but I can't get the third to pair. The motor responds to the single channel remote in the same way as the other two and it goes into pairing mode using the remote's prog button but it won't respond and pair to the ESP Somfy. Perhaps I should open a new issue?

rstrouse, 2023-04-11:
Is this third motor fully charged?  Also, try to increment the address by 10 and see if it will pair.  If it does then something occurred during the pairing process and the address ended up getting reserved in the motor.  If this happens with a remote you need to reset the motor to factory settings and start over but in this instance, you can simply change the address.

youngsyt, 2023-04-12:
Yes, all the motors were fully charged. Anyway, changing the address by 10 allowed the blind to pair so something must have gone wrong with the original pairing attempt. I plan to factory reset the motor and start again from scratch. Thanks for your support, much appreciated.

rstrouse, 2023-04-16:
Closing this for now.  Open a new issue if you have further problems.

closed 2023-04-16 by rstrouse