rstrouse/ESPSomfy-RTS #21: No fallback AP ESPSomfy

mozzaLB, 2023-04-14:
Hi there, 

I am using a ESP32-C3-DevKitM-1 which I want to use for this project. The flashing works like a charm using the ESPHome web but afterwards I do not get a Wifi AP named ESPSomfy RTS to set my home Wifi credentials and setup the cc1101. I have tried using the latest version 1.47 and 1.46 to make sure this was not the problem. Any idea what I am doing wrong?

Any help much appreciated, 

kind regards, Mozza

rstrouse, 2023-04-14:
If you installed any pin jumpers to place the ESP32 into firmware mode then remove these jumpers and reboot.  If no jumpers have been installed then disconnect power completely (do not simply press the boot button).  If the device is boots it will show the ESPSomfy RTS as an AP.

mozzaLB, 2023-04-14:
Thanks for the quick response. I indeed disconnected the power and reboot the ESP32 module. I have flashed v1.47 to another ESP32 (same type), Succes after flashing but still no AP. Not sure what I am doing wrong here

mozzaLB, 2023-04-14:
I have installed ESPHome on the device and then I do see the ESPHome AP, so Wifi module seems to be working correctly

mozzaLB, 2023-04-14:
yes that one indeed

rstrouse, 2023-04-14:
Hmmm... I do not have a board with the mini on it.  I wonder if it needs to be compiled differently.  Do you use the Arduino IDE?

mozzaLB, 2023-04-14:
I am not familiar with Arduino IDE. I just use a MacBook and flash the ESP32 with ESPHome web using Chrome which work fine for ESPHome and use this connect to Bluetooth Temperature and Humidity sensors around the house. 
I did find one page where they report a similar problem: https://github.com/espressif/arduino-esp32/issues/6551

mozzaLB, 2023-04-14:
https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html This is the ESP32 module I am using

rstrouse, 2023-04-14:
I read the thread and the solution where they call out setting the hostname before turning on WiFi mode is already being done.  Contrary to what the poster is saying, the AP will not fire up the strict AP mode if it does not have this.  The last post states something interesting where it says that they do some weirdness with the WiFi pins by dropping the TX power to get the AP mode to start.

When you install ESPHome do you have to identify that it is an ESP32C3 chip?  Sorry I do not use ESPHome.  If it does, I am pretty sure I know what is going on with it.  The processor on the C3 is actually akin to an ESP8266 and only runs at 160Mhz.

You cannot set the WiFi credentials in the code as this is part of the onboard process and are not stored in the code.

mozzaLB, 2023-04-14:
Thanks for your efforts in this. Pity I cannot include my WIFi credentials immediately. When I install ESPHome I just go to https://web.esphome.io/, connect to the USB port were my ESP32 is connected with and than prepare for first use. That is it and starts installing. After the installation there is a ESPHome AP to which I can connect to.

rstrouse, 2023-04-14:
This is exactly the process but perhaps not with this board.  Are you following these instructions exactly and ignoring the Prepare for first use link?
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware#using-esphome-web

mozzaLB, 2023-04-14:
Yes, initially I ignored the ' prepare for first use' with both ESP32 chips I have tried (they were not used before and completely new) and just followed these instructions in your Wiki. It follows the proces like normal and at the end I do get a "installation succesfull" after the flash. Than I disconnect and reconnect the ESP32 but there is no AP (waited for >30min)

rstrouse, 2023-04-14:
I initialized the hostname separately as suggested by the link you provided to see if this makes a difference.  In the release download the `SomfyController.onboard.esp32C3.bin` file then flash it to see if this makes a difference.  If it does not then I will see if I can compile with a C3 bootloader.

mozzaLB, 2023-04-14:
Thx rstrouse! I flashed the updated version without any problems but unfortunately still no AP after disconnecting and reconnecting the ESP32

rstrouse, 2023-04-14:
Hmmm.  Please disconnect the transceiver completely to see if perhaps the strapping pins are not as described.

mozzaLB, 2023-04-14:
Completely connected, only disconnected the tx pin of the transceiver or completely disconnected, still no AP

rstrouse, 2023-04-14:
I tried recompiling with the C3 bootloader and it will not compile.  I will look deeper into why that is.  I am starting to believe that only the older libraries are supported for the C3.  I will try a few different board definitions to see what I can find out.

rstrouse, 2023-04-14:
I just posted another SomfyController.onboard.esp32C3.bin file to the release.  It appears so much of the library for the C3 is out of date that it does not even include the math functions that are used to calculate the timing but I did remove the pin assignments from the SoftAP mode.  When you are done flashing can I get you to use the online esptool linked above and watch the console.  Keep in mind the flash should be done to address 0x0000.  

To watch the console on boot perform the following steps
1. Click the disconnect button if it exists.
2. Press the start button
3. Select the serial port that the ESP is connected to
4. Press the reset button

This should reset the ESP and log the output settings.  This should tell you what the ESP is currently doing.

mozzaLB, 2023-04-15:
Thanks again for your help. I have done as you instructed. The log output after flashing the SomfyController.onboard.esp32C3.bin:

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x5 (DOWNLOAD(USB/UART0/1))
waiting for download

And then nothing. I have also looked at the output once I have flashed the original SomfyController.onboard.esp32. The output continuously shows invalid headers 

invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c0dc
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

Hope this gives you a clue to solve the issues. I did not realise these chips were so old

mozzaLB, 2023-04-15:
Now also the new SomfyController.onboard.esp32C3.bin gives me the same continuous header error. If I flash ESPhome It works just fine. I have included the bin file for ESPHome, perhaps it helps you to understand
[test-factory.bin.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/11238870/test-factory.bin.zip)

mozzaLB, 2023-04-16:
I took another ESP32 (WROOM-32) and now I can see the ESP AP and fill in and save my WIFi credentials. Strange that the ESP32C3 module would not work. So far so good. However, I bumped into another problem during the setup. After I filled in the pin layout in the configuration with my E07-M1101D-SMA I cannot log any of the buttons of the delis 1 remote (Radio not Initialized header disappeared after I clicked saved). I rechecked the pins and even used GPIO1 and GPIO3 on the ESP32 for RX/TX but still no logs of the remote still. Is there any way I can check the E07-M1101D-SMA (bought from Ali and same as the one from the Wiki)? Or is there something else I am doing wrong here?

I can start a new issue because this is not related any more the the ESP32C3 problem. If you want I can still do some test on the ESP32C3 module

mozzaLB, 2023-04-17:
I just went crazy and almost touched the antenna of the CC1101 for the logging of the buttons of the remote, surprisingly then it did work! After that I was able to pair my blind on the first try :-) 
I also installed the home assistant add-on and everything just works amazing. Thx again for you work rstrouse!

I only have one more question: can I invert the open and close functionality for the cover entities in home assistant? For our folding arm sunshade it’s more logical to “open” it and shield us from the sun on our patio. Now the status is 100% open while the folding arm is completely closed

rstrouse, 2023-04-17:
> I only have one more question: can I invert the open and close functionality for the cover entities in home assistant? For our folding arm sunshade it’s more logical to “open” it and shield us from the sun on our patio. Now the status is 100% open while the folding arm is completely closed

I don't know where the braindead idea of the backwards cover position came in but I agree wholeheartedly with you 100% is not completely open when it comes to a cover, shade, curtain, or blind.  It is completely backwards.  If you read the discussions on the topic you will see folks protesting and the devs blowing them off.  There are statements like "think of it as 100% of the light"... huh?  You can keep cutting all the corners off that box and it will never make a circle.

Short of completely re-writing the cover code in Home Assistant (because you cannot extend existing domain components), or creating new templates, this issue is shut down every time.  Perhaps this originally comes from a garage door or gate but it makes no absolutely no sense for anything else.  Even then you have to stretch the imagination.

Perhaps I may dig in my heels with the powers that be at HASS with this at a later date but I have some things that I need to get to first wrt integrated blind slats.  It is flat out silly.  The code to fix this would be trivial at the cover domain level so I may see if I can get a pull approved for it in the future. </end rant>

mozzaLB, 2023-04-17:
I found a solution on a HA community forum post. I made a covers.yaml file with the following code 
[covers.yaml.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/11254769/covers.yaml.zip) and added the include at the end of configuration.yaml: cover: !include covers.yaml. Works perfect for me

rstrouse, 2023-04-17:
Glad you found a solution but this should really be done by type of cover in the integration not by using templates.  Don't get me wrong I am a Home Assistant believer but this one keeps showing up over and over again in their forums and issues log.  While I can complain about it what I really need to do is contribute and fix the issue.

I'll point folks to this solution for now and push for options on the integration code for covers in the future.

mozzaLB, 2023-04-17:
I completely agree, this should not be necessary by using templates and I think this can easily and should be implemented in HA. Nevertheless I really appreciate you help in this. Amazing work!

rstrouse, 2023-04-17:
On the original issue, I think the C3 is a bit of an outlier.  It is using a different instruction set, bootloader and some of the underlying physical addresses.  This would require more effort than I can place on creating different builds for this series.  Besides, I don't have this hardware to test.  The other thing that is very odd with it is that the C++ libraries seem to be incomplete when you simply select the chip type in the Arduino IDE.  Standard constructs for math functions appear to be missing.

rstrouse, 2023-04-21:
I am going to close this for now.  I believe the libraries I am using to compile the low level comms also use some math functions that do not appear to exist in the standard libraries for the RISC-V chip.  The invalid header you are seeing is on the C3 is because of this.  I don't own one of these and at every turn I have been thwarted even after stubbing some of the missing libraries.

For now I am going to shelf this and maybe in the future when there is nothing else to add I can come revisit it.

closed 2023-04-21 by rstrouse