rstrouse/ESPSomfy-RTS #22: Blind tilt position is incorrect

jfparis, 2023-04-14:
Hi

I have an issue with tilt position of the blinds.

Let me first describe how the blinds with tilt work
* There is a single motor to actuate the blinds
* When the blind goes down, the motor start by tilting into a "closed tilt" position then the Venetian goes down 
* if we want to tilt them open, the motor is actuated in the opposite direction for a few steps
* Similarly when the blinds goes up, the motor start by tilting into another "closed tilt" then pull the venetian up
* if at that point you want to tilt open you move a few step in the opposite director

In addition, the user point of view could be totally inverted depending on which side in/out the blinds have been installed

Based on that I believe that the logic that reports whether the tilt is open or not is not correct. 

anyway, not sure whether this is clear or not. Would a video help?

rstrouse, 2023-04-14:
The code is written the tilt and lift motors which are actually two motors in one.  To tilt the slats you long press on the remote and to lift you short press.  I am working out how to support the tilt method that you have as it is much different and will need some configuration parameters added to each blind.

jfparis, 2023-04-15:
Let me know if a video showing how the work is useful to you or not

rstrouse, 2023-04-15:
I believe I have a handle on it now.  The confusing part is that there are several tilt methods for the motors.  My assumption was that it only ever involved a motor for the tilt and a motor for the lift.  Funny enough I cannot find a motor that has integrated tilt anywhere online.  Or even one word of documentation illustrating it.  Everything is based on the tilt and lift dual motor system.

jfparis, 2023-04-15:
> Funny enough I cannot find a motor that has integrated tilt anywhere online

Maybe it's me misunderstanding that both motors seems to move at the same time (when I tilt, the blind start rising)

rstrouse, 2023-04-16:
Actually the only operation info out there says that the lift function operates with a short press and the tilt with a long press.  One motor connects to the lift mechanism and the other to the tilt mechanism.

rstrouse, 2023-04-16:
I am zeroing in on the integrated tilt function.  v1.5.0 will have the ability to identify the motor as having integrated tilt.  I am testing it now but I will need to revamp the my positioning so that it stores the tilt position for the my command then moves the motor up to acquire the position.  I will release it instead of producing it as beta since I know the tilt function for integrated tilt does not operate correctly right now but the other types work correctly and I am pretty confident in my simulation.

I'll post back here when it is ready.

rstrouse, 2023-04-20:
Install the firmware and application files for v1.5.0.  Then go into the shade configuration and set your shades to the `Blind` type and the Tilt Type to `Integrated`.  This should now capture and set the proper tilt for an integrated tilt mechanism.

rstrouse, 2023-05-09:
@jfparis did this sort out your issues.  Also, there have been additional changes related to integrated tilt motors so update to v1.5.3