rstrouse/ESPSomfy-RTS #23: Configure Transceiver - GPIO not selectable

d82k, 2023-04-15:
Hello,
Amazing work! Huge effort very much appreciated.

I'm using a Wemos S2 mini (based on ESP32-S2FN4R2). I use the wiring below. Unfortunately the web interface does not allow to select pins 6-11. To overcome this limit I used chrome and forcing the values needed via the inspect function. I reviewed the code, cannot find the filter on the pinout, where is it possible to change it please?

```
Wiring CC1101 on Wemos S2 Mini

+=============+===================+
| CC1101 Pin  |   ESP32-S2 Pin    |
+=============+===================+
| 1 GND       | GND               |
+-------------+-------------------+
| 2 3V3       | 3V3               |
+-------------+-------------------+
| 3 GDO0 (TX) | GPIO-5            |
+-------------+-------------------+
| 4 CSN       | GPIO-12 (FSPICLK) |
+-------------+-------------------+
| 5 SCK       | GPIO-7            |
+-------------+-------------------+
| 6 MOSI      | GPIO-11 (FSPID)   |
+-------------+-------------------+
| 7 MISO      | GPIO-9  (FSPIHD)  |
+-------------+-------------------+
| 8 GDO2 (RX) | GPIO-3            |
+-------------+-------------------+
```

rstrouse, 2023-04-15:
Pins 6-11 should be connected to the SPI for the integrated flash.  Did they re-route these on the mini?  If they didn't then littleFS will fail.  Does everything actually work with the pins you forced?  I originally filtered these out because I didn't want anyone to destroy their board with a bad SPI choice.

If it works I can remove the filter.  It is in the index.js ~line 1240.

d82k, 2023-04-15:
I'm still new to ESP, I did modify index.js but the filter was still there, so was wondering if there was another check. I had some trouble with uploading the data, and cannot remember if I did with littleFS or spiffs.. need to verify.. the board looks fine, did not explode so far. I will recompile and upload again.

rstrouse, 2023-04-16:
Let me know if you get this working on the S3 mini and I will pull the pin restrictions from the UI.  I may just do that anyway.  If you are dabbling in setting pins to something other than the standard pins then you probably have a better handle on your board anyway.  It's too bad the ESP folks didn't standardize on these within the chip versions especially with unprotected pins.

rstrouse, 2023-04-20:
If you update your device firmware and application files, ESPSomfy RTS no longer restricts the use of pins 6-11 for the SPI selection.

rstrouse, 2023-04-22:
I removed the restriction on pins 6-11.  However, please be careful with this.  Even looking at the table you supplied above I can see you tagged them with FSPIxxx.  To me this indicates the flash SPI interface.

closed 2023-04-22 by rstrouse