rstrouse/ESPSomfy-RTS #24: Static IP not detecting

binu02, 2023-04-16:
I have installed firmware on my WT32-ETH01 device and am attempting to assign a static IP address to it, but it always defaults to DHCP automatically. Could you please take a look at it?

rstrouse, 2023-04-17:
I removed the initial code that sets the static IP on the ethernet module because it was causing an occasional reset and did not have the time to figure out why.  That being said, is there a reason why you aren't doing DHCP reservations for fixed IPs?  Static IPs at the device level can be very challenging since they can still be issued by the router to other devices.

Originally, I added this code simply because it was there but I have found the PHY chip to be a bit unstable with a fixed address.  Let me know your use case and I will take another stab at it.  The preferred method is to marshal IPs from the DHCP server but reserve by MAC.

rstrouse, 2023-05-09:
I am closing this for now and will be removing the UI for the static IP on the next release.

closed 2023-05-09 by rstrouse