rstrouse/ESPSomfy-RTS #25: 3rd linked remote shows undefined, cannot be removed

EDelsman, 2023-04-17:
I tried to link multiple remotes to one shade. This resulted in the third shade showing undefined, undefined. (see attached picture) I cannot remove the entry - the garbage icon asks yo delete, if I confirm then nothing changes.

By the way, how many remotes can be linked? I have three telis4, one on every floor. All of them can control the blinds, in total 6 codes per blind because all remotes use the fifth group to control all blinds.

[image]

rstrouse, 2023-04-17:
You can link up to 5 remotes per shade.  First go to the updates tab and generate a backup then post it here.  This will let me know if I somehow got the indexing incorrect.  Then try restarting the ESPSomfy RTS device and refresh the browser.

EDelsman, 2023-04-18:
Hi, I tried rebooting, refreshing the browser, ipad and PC but the undefined one remained. I never could get more than three items visible in the list, one of which "undefined". I added the backup, if I interpret it correctly the first blind has 5 remotes, but I'm not sure which ones anymore, and remove does not seem to work anymore.
[ESPSomfyRTS 1970-01-01T01_39_56.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/11266975/ESPSomfyRTS.1970-01-01T01_39_56.zip)

rstrouse, 2023-04-19:
I'll have a look.  Interestingly, I see a valid address for the linked remote in the output so this might be the buffer size for the screen display.

EDelsman, 2023-04-20:
Thank you very much!

rstrouse, 2023-04-20:
Please pull v1.5.0.  I also increased to max number of linked remotes to 7 from 5.  If you use Home Assistant do not forget to update that component as well.

EDelsman, 2023-04-21:
I updated both HA and the firmware. It now works properly, I was able to remove remotes and add all of them back.
One small thing though, I think you have a similar problem with buffer in the linking process. If I link more than say 5 remote codes at once, the list shows undefined still. When I save and edit the shade again, it is shown correctly.
Thanks again for all your effort.

closed 2023-04-21 by EDelsman

rstrouse, 2023-04-21:
You are correct the output from the link and unlink remote needed to be changed as well.   This will be fixed in the next update.  Thanks for reporting it, I do not have that many remotes so I had to make all ESPSomfy RTS devices link to a single roller shade to even test it.