rstrouse/ESPSomfy-RTS #26: Add the ability to close the shade even if it thinks it

bradphillips921, 2023-04-19:


bradphillips921, 2023-04-19:
I'm having some issues with my shades getting out of sync if I use the remote to control them but the remote isn't picked up by the transceiver. So, the shade opens but Home Assistant still thinks the shade is closed. Because the shade is closed it won't let me press the close button in Home Assistant. Vice-versa in the open position. Can this be changed so that I can still send a close command via Home Assistant even if HA thinks the shade is closed?

rstrouse, 2023-04-19:
Unfortunately, I do not have control of that in HA it automatically disables the button and doesn't execute my code.  Are you running the most current release v1.4.7?

bradphillips921, 2023-04-19:
Bummer!

Nope, I'm on v1.4.6

rstrouse, 2023-04-19:
Update your device.  There were updates to the protocol for more remotes.  Also, mine never misses a beat from very far away.

muppet99, 2023-08-10:
You can create a button that’s calls the correct service:

show_name: true
show_icon: false
type: button
tap_action:
  action: call-service
  service: cover.close_cover
  target:
    entity_id:
      - cover.1
      - cover.2
  data: {}
entity: cover.1
icon: mdi:close
show_state: false
name: Close covers

hope this helps. This button closes the both covers on the back side of the house