rstrouse/ESPSomfy-RTS #27: Receiving signal but not Transmitting

andy-81, 2023-04-21:
I have tried this with 3 C1011 boards so far and each time I am able to receive the remote control commands but unable to send them.

I am sure it must be something I am doing wrong but for the life of me can not figure out what it is. I went through the manual and could not see anything I missed so thought I would see if I have missed anything.

I currently have 2 of the 8 pin E07-M1101D boards, one hooked up to a lilygo PoE esp32 and the other connected to a standard Wi-Fi esp32 board. Both connect to the Wi-Fi and allow me to see when a remote is pressed for any of the blinds I have but when I try to pair with a blind nothing happens. 

I have tried checking the logs from the opposite esp-Somfy set up and can see no record of the opposite device when I press the prog button on the webpage. 

I have also tried following the tuning but this makes no difference on the transmition of the signal. Is it possible I have followed the wiring wrong? I thought it at first but as I was receiving figured I had to have the tx & rx wires the right way.

I also have problems with getting the Ethernet port working on the lilygo but can work on that later, keen to get the connection to the blinds working first. 

I have wired it the same as the instructions on the wiki. Is there anywhere I can pull logs from the portal to see if anything is getting mixed up in the back end?

rstrouse, 2023-04-21:
Please double check that pin marked as GDO0 (pin 3 on the E07-M1101D) is connected to the pin you have selected as the TX pin.  I do not have a LilyGO and this information comes from another user.  This very much looks like what you get when the TX pin is not connected to the GDO0 pin of the transceiver.  The transceiver is capable of receiving on either pin.

As an alternative you can try to hook up GPIO4 to GDO0 then set both the RX and TX to GPIO4.  On some ESP32 boards GPIO35 only allows input and LilyGO doesn't seem to list any restrictions on their pins.

andy-81, 2023-04-21:
Ok, thank you so much, and thank you for coming back so quickly.

I can now see most the commands being sent by Espsomfy, for some reason it doesn’t show the prog command when I click the prog button for the shade, all the other commands I can see. 

Is there another way of getting the portal to send the prog button? I’ll upload some screenshots of what the second esp is seeing being transmitted

andy-81, 2023-04-21:
This is what I see when I use the actual remote connected to the blind. 
[image]

andy-81, 2023-04-21:
And this is what I see when I use Espsomfy to transmit codes including clicking the prog button on the web interface to try and program the controller to the blind
[image]

andy-81, 2023-04-21:
As you can see, I see the prog button being pressed on the physical remote but not the Espsomfy remote.

rstrouse, 2023-04-21:
Thanks for troubleshooting this.  I added filters for the up/down/my buttons and didn't add one for the prog command.  This needed to be done to allow the integrated tilt blinds to work.

Please download the SomfyController.ino.esp32.bin and SomfyController.littlefs.bin files from the main repository (not the release) and update your device.  I will be publishing v1.5.1 later today.

andy-81, 2023-04-21:
Amazing, thanks rstrouse.

That has resolved the issue with the prog signal being sent and I can now pair espsomfy with my blinds.

After powering off for a while and changing some pins I can also confirm the Lilygo Ethernet is now working, it looks like there is an overlap with some pins in the documentation which was causing the issue, when I fixed that and powered the device down for a bit it has all come up and connecting via Ethernet. 

Thanks again

rstrouse, 2023-04-21:
Thanks for replying back.  I will be posting a v1.5.1 shortly.  Essentially, this has no impact for you unless you have an 80-bit remote with a thumbwheel but it will include the changes I made earlier.

rstrouse, 2023-04-21:
Thanks for the help.  I am closing this issue so if you have further issues please submit another ticket.

closed 2023-04-21 by rstrouse

nobbyneb, 2023-04-24:
Thanks for fixing so quickly, had the same issue but now working perfectly on our Somfy garage door 😄

love espSomfy - what a great development, thank you.