rstrouse/ESPSomfy-RTS #28: Motors sometimes get stuck

ticarki, 2023-04-23:
Great project, with detailed documentation and easy-to-set-up hardware; thanks a lot!

I have had issues with my motors before; Sometimes, when pushing the up or down button at the remotes, they are stuck, but they seem to receive the signal because the motor clicks. The same happens after I set up a CC1101 with your project. As a result, the status of the shades in ESPSomfy RTS gets out of sync. The remotes are 'Somfy Telis 1 RTS', but I can't tell which motors are used; it is a rental apartment. This is not an issue with your implementation, but do you know this issue and a solution?

Thanks a lot!

rstrouse, 2023-04-23:
Are these battery powered?

ticarki, 2023-04-23:
the remotes, yes, the motors, no, as far as I know.

rstrouse, 2023-04-23:
From what I know of the click on the motors, that is the relay that energizes the motors.   The reason I asked whether they were battery controlled is that the start for the motor requires a current inrush and if the batteries are weak that becomes more difficult.

However, if they are hardwired and you hear the click then it probably is one of the components on the motor going bad.  I am one of those folks that would tear it apart to see if I could fix it but in a rental that might not be a good idea.  There are some videos online regarding the disassembly of some motors and a potential part that fails.  This is called the condensateur but as I look at it, it appears to simply be a starter capacitor which would be a failure point for the symptoms you describe.

ticarki, 2023-04-24:
Okay, thank you! Another issue I observed; I use to shut down my wifi at night and in the morning the ESP32 does not connect to the Wi-Fi again, only after I restart, it works as gain. Is there anything I can do about that?

rstrouse, 2023-04-24:
That is odd since the code will check the connection and reconnect if it loses connection.  I just tried that by rebooting my router and it reconnected as soon as the router came back up.

rstrouse, 2023-04-24:
One other thing.  If you shut down your network and the SSID goes away the device will open its AP mode until the network comes back on-line.  This can take up to 3 minutes after the SSID starts.

ticarki, 2023-04-25:
I turned the Wi-Fi off for 5 hours, maybe it stops retrying to connect after some time? I will try shorter breaks and see if it works.

ticarki, 2023-04-28:
I think the ESPSomfyRTS SSID was there once, but not always. Your test also works for me; After a restart of the router the ESP connects to my router directly. For now I only had the issue when the Wi-Fi was off for 5 hours. I will now steadily increase the Wi-Fi shut downs and see when it stops working.

rstrouse, 2023-04-29:
Let me know what you find.

ticarki, 2023-04-30:
Tonight I shut it off for 2 hours and in the morning it worked. Then sometimes, I can not navigate the shades in HA, but then it works again. I get the following error:

```
Logger: websocket
Source: custom_components/espsomfy_rts/controller.py:120 
Integration: ESPSomfy RTS 
First occurred: 28. April 2023 um 22:25:25 (13529 occurrences) 
Last logged: 10:04:27

[Errno 104] Connection reset by peer - goodbye
Connection to remote host was lost. - goodbye
websocket connected
ping/pong timed out - goodbye
[Errno 110] Operation timed out - goodbye

```

In the last nights I did not shut it off or only for 30 minutes and it worked, but it seems like I got the error frequently.

rstrouse, 2023-05-01:
Yes if you shut off your network then it will loose the connection to the websocket and your shades will become unavailable.  It should reconnect whenever the network comes back.

Out of curiosity why do you turn off the network?

ticarki, 2023-05-02:
Yes I would expect the error while the Wi-Fi is off. But now I also get it while the Wi-Fi is on every few milliseconds. I now don't get it running again in home assistant with the following error:

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 293, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/espsomfy_rts/cover.py", line 42, in async_setup_entry
    for shade in controller.api.shades:
  File "/config/custom_components/espsomfy_rts/controller.py", line 301, in shades
    return self._config["shades"]
KeyError: 'shades'
```

The router is right next to my bed, so I thought it might be better to turn it off. Thanks for your feedback, maybe I will try to get an ESP with Ethernet and try it again.

rstrouse, 2023-05-02:
That message has nothing to do with the connection.  I'll have a look at the HA component.  It looks like it is not receiving any shades from the request.

rstrouse, 2023-05-02:
Please check that you are running the latest version of ESPSomfyRTS-HA in Home Assistant.  I am suspecting that you do not have an IP address reservation set up in your router for the ESPSomfy RTS device and your router issued a different address.  The current version of the integration will attempt to connect using the host name you set up for the device.

ticarki, 2023-05-03:
I updated the integration, firmware, application and reserved the IP Adress yesterday. For now it seems like that fixed the issue, thanks a lot!

rstrouse, 2023-05-07:
I am going to close this.  If you have further issues please submit a new issue.

closed 2023-05-07 by rstrouse