rstrouse/ESPSomfy-RTS #29: Could your code be adapted to work with 915 MHz transceivers?

FlaMike, 2023-04-23:
Obviously, this post is not creating an issue with ESPSomfy-RTS, which is awesome & appreciated greatly!

My wife asked me last night if it's possible to create simple way to turn on/off our whole house vacuum.  The system uses a 915 MHz remote that's integrated into the handle to turn the power unit on/off.  However, when she is just using tools & doesn't have the handle with her, she needs to go to a hose outlet, turn the system on/off, & then return to where she's cleaning.  I have zero experience with 915 MHz systems. E.g., I don't even know if all 915 MHz devices are LoRa

Of course, Vacuvision, the company that makes the power controller for our unit, offers a fob that can be added to do exactly what my wife is looking for, but it's priced like a ZRTSI, $80.00 for a 2 button key fob.  OMG!  And it's yet another dongle to misplace.

I started looking around for transceivers, & of course, they're available with antennas for around $2.50 from the slow boat company.  I assume one could connect it to an ESP32, figure out how sniff codes, & hopefully stumble upon away to use HA to turn the beast on & off.  

I also found this development board which is a combo ESP and 915 MHz transceiver:  

https://www.amazon.com/MakerFocus-Development-Bluetooth-0-96inch-Display/dp/B076MSLFC9/ref=sr_1_2?crid=3077DGV7YX1NF&keywords=esp32+915mhz+transceiver&qid=1682265536&sprefix=esp32+915mhz+transceiver%2Caps%2C92&sr=8-2

My question is, with minimal guidance & assuming I could sniff the codes with the transceiver, could I create a toggle in HA to turn the big  sucker on & off?  It looks like there are some Arduino libraries to help sniff packets, etc.  Alternatively, should I just take my lumps & get her the ridiculously priced fob.  Of course, my wife is lobbying for the latter.

Your thoughts & suggestions are most welcome!!!

FlaMike, 2023-04-23:
Is the CC1101 capable of 915 MHz?  I'm thinking it might be...  I have an extra E07-M1101D & antenna that could possibly be used in a separate box than the Somfy unit.

rstrouse, 2023-04-23:
The CC1101 is capable of receiving that frequency.   I have to say I have never seen such a thing.  Why would anyone use wireless when you have to run 2 inch pipes all over the house.  Mine is hardwired and the communication comes from the handle and the dustpans in the baseboards.  I am just confused as to what that advantage is.

I digress though.  How are your C/C++ skills?  With some of those, you could go in and modify the code to receive and transmit on the that frequency.  On line 194 of Somfy.h is where this initial frequency is set.  

The bigger issue will be decoding and encoding the frames.  Line 1792 of Somfy.cpp is the decoding algorithm and the transmit is on line 1711.  The Somfy protocol was a bit of a challenge due to the way it pulses the preamble.  So all of the code uses a bit-banging strategy.  This could be LoRa and use some protocol standards though in which case this codebase would be the long road.

Perhaps if you are tinkering.  https://randomnerdtutorials.com/esp32-lora-rfm95-transceiver-arduino-ide/

FlaMike, 2023-04-23:
Thanks for your quick & helpful response!

I think they used wireless for our vacuum because it has "hide a hose" (hideahose.com) where the hose(s) get sucked into the wall when not in use.  You just pull out as much hose as you need from the wall plate, vacuum away, & when finished, the vacuum system pulls the hose back into the wall (PVC pipe, actually).  You don't have to lug the hose around the house-- we have 2 hose outlets, one  has a 50' hose & the other 40'.  The hoses are just that--no power in them like the Beam system we had in our previous house.  So, I guess the easiest way to turn the vacuum on/off is via a wireless switch in the handle.  It's a trade-off, like everything is.

Once upon a time, many decades ago, I programmed in C quite a bit.  Rusty as heck, but I  may give it a whirl.  I also found an inexpensive LoRa tap remote that my wife is excited about. 

https://www.amazon.com/LoraTap-915MHz-2-button-Remote-Receiver/dp/B07MXWNYCG?ref_=ast_sto_dp

Since it's returnable to Bezos, I may order one today & fall back to doing the ESP Somfy hack as well.  We shall see.

As always, I appreciate your helpful suggestions!

rstrouse, 2023-04-24:
I guess it has no beater head on it.  Ours has power to every port and there are 8 of them strewn around the house.  The retractable hose is a neat idea but the maids would have torched that for sure by now.

FlaMike, 2023-04-24:
Yep.  The carpet beater bar is "air" powered, not with an electric motor, which is certainly a more powerful and better option.  Since the "maids" here are my wife primarily and me secondarily, we've not had any issues with the retractable hoses.  When my wife saw the retractable hoses were an option, she said it's the only one for her.  She wasn't a fan of lugging the hose around our previous house & really hated storing it.  To each, their own, of course.

With two adults, two small dogs, and mostly tile flooring with a few area rugs, the retractable hoses work pretty well overall for us.  We have a couple of baseboard dustpans, too.  My wife said she rarely uses them, however.  Having a car kit that can be attached to the power unit in the garage is really useful.  I hate sand in the cars!  We also have a Dyson stick vacuum for small cleanups & a shop vac, of course for outdoor/wet/messy stuff.

closed 2023-05-01 by rstrouse