rstrouse/ESPSomfy-RTS #30: LAN modules have no network connection after updates

011V32, 2023-04-24:
Since version V1.5.0, the LAN modules no longer have a network connection after an update. This means that after each update (firmware and application) you have to remove the voltage from the module so that the module becomes visible in the network again.
This worked without problems prior to v1.5.0.

rstrouse, 2023-04-24:
I have a WT32-ETH01 on my bench.  I do not experience this with the firmware update.  In the past I have had some goofiness but that was simply because the bootstrapper pin was connected.

The code for updating the firmware simply marks ESP32 reset after the update.  On boot it then enables the PHY.  This boot process is no different whether coming from the reset or a power down to the software.

011V32, 2023-04-24:
I am using the LilyGo T-Internet PoE.
It's weird that before I didn't have this behavior with any update and now with every...
What else could it be?
(I haven't changed anything on the hardware.)

rstrouse, 2023-04-24:
Update the firmware to the current release.  I did not change anything and it will not change anything for you.  Then wait after it is finished to see if you get a wait spinner as the device reboots.  This is the process of the socket reconnecting after it has been made unavailable during boot up.

If this fails you can watch the boot up logs by connecting to the console with this tool.  https://espressif.github.io/esptool-js/

011V32, 2023-05-05:
I can't pinpoint the error any further.
Maybe it's the temperature.
So I'll close it until I have more specifics.

closed 2023-05-05 by 011V32