rstrouse/ESPSomfy-RTS #31: Possible to add signal Led?

nobbyneb, 2023-04-24:
Cheeky request - It would be great if there was an option to show when a 433mhz signal is received or sent (as with traditional Somfy remotes) via a hardware led. Could a spare pin/pins be added as default led indicators please for 433mhz transmission?

rstrouse, 2023-04-24:
I originally hooked GDO0 to the same pin as onboard LED of the ESP32.  This actually flashes when it sees a 433Mhz signal.  However, this was very early on in the development and I suspected that it was messing with my timings.   Since then however, I have determined that the widely accepted timing parameters were not entirely correct so it may have been coincidental.  You could try to use GPIO-02 for TX and RX to see if it interferes.

prbtt, 2023-04-25:
I have tried using an led and it works fine. I used the same pin for SCK. It flashes when a signal is being received and transmitter. My board was having issues with transmitting data but after removing the LED my problem persisted. It turned out that I was missing some power filtering caps.
You can also put a transistor connected to sck to power the led, that also worked for me.

nobbyneb, 2023-04-26:
Thank you! I will try both. Much appreciated

closed 2023-05-09 by rstrouse