rstrouse/ESPSomfy-RTS #32: Receiving but can't transmit. Battery Somfy

KSKNok, 2023-04-27:
Hi, 
Great project and I see many people benefitting from this so good job and thank you! And I like the writing style in the instructions :D 

Myself, I have had no luck to move my shades. I can't manage to get the E07-M1101D to send/transmit. The logs do recognize the button presses accurately from the remote Situo 5 RTS Pure II. 
I believe the motors are Sonesse ULTRA 30 WireFree RTS Li-ion.
I have ESP32-WROOM-32 (usb-c version).

I have used the wiring from the the instructions:
1 | GND | GND
2 | VCC | 3v3
3 | GDO0 - This is the TX Pin | GPIO 13
4 | CSN | GPIO 5
5 | SCK | GPIO 18
6 | MOSI | GPIO 23
7 | MISO | GPIO 19
8 | GDO2 - This is the RX pin | GPIO 12

Is there something that I am missing?

rstrouse, 2023-04-27:
Please double check your wiring to ensure that pin 3 of the E07-M1101D truly is connected to GPIO 13.  Then check the Transceiver configuration to ensure you have GPIO-13 selected for TX and GPIO-12 selected for RX.  Unfortunately, the picture is opposite of the table because I took the pictures from a different setup and haven't had the time to update the wiki with new photos.

KSKNok, 2023-04-27:
Aaaaaa. okay now I got it. I didn't realize this was the place to do that. I had here earlier TX as GPIO-12 and RX as GPIO-13. Now swapped them and it is working. It is amazing! Thank you! 
[image]

closed 2023-04-27 by KSKNok