rstrouse/ESPSomfy-RTS #34: Open MQTT Gateway

FlaMike, 2023-04-30:
I found this while looking for ways to sniff, transmit, & receive 300ish-433 MHz signals.  There are ESP32 boards that have transceivers & OLED displays builtin, similar to the one I sent to you last week for 915 MHz signals.  Then I saw that Open MQTT Gateway has firmware available for Somfy controllers.  I think you may still need the C1101 to get everything going, but I thought you might be interested in knowing about it.

Open MQTT Gateway:   https://docs.openmqttgateway.com/

It looks like you need to be very selective about the ESP32 board, as many versions are not supported.  Of course, I ordered one via slow boat delivery because I couldn't tell with certainty what board the Bezos would actually ship.  It may be here in a month or so.  More on this story as it develops :-)

https://www.aliexpress.us/item/2251832685763835.html?spm=a2g0o.productlist.main.5.581aSL5eSL5ebn&algo_pvid=38c54624-829d-40f3-a535-792780bfe6c2&algo_exp_id=38c54624-829d-40f3-a535-792780bfe6c2-2&pdp_npi=3%40dis%21USD%2116.59%2116.59%21%21%21%21%21%40211be59e16828861692765954d0858%2112000031557075306%21sea%21US%210&curPageLogUid=aAumdtpSST2c

BTW, feel free to close my other issue regarding the 915 MHz option.  Doesn't look like it's going to work for me, but I found a workaround for my beloved that earned WAF points.

Have a super evening & week ahead!

rstrouse, 2023-05-01:
Looking forward to hearing what it does.

FlaMike, 2023-05-01:
Not much information on the pre-compiled binary.  Just says it needs a C1101, so there's likely no advantage to your solution and it's more expensive.  Still it may be a reasonable sniffing tool.  My board is weeks away, but I'll let you know what I think of it once I get my feet wet.

 Somfy is #29 in the list:

https://docs.openmqttgateway.com/upload/web-install.html

Somfy binary:

https://docs.openmqttgateway.com/prerequisites/boards/esp32dev-somfy-cc1101.html

C1101 Pinout they use is here.  Could be the same as yours... I haven't checked but Dupont cables make it easy enough.

https://docs.openmqttgateway.com/setitup/rf.html#pinout

rstrouse, 2023-05-01:
Yeah you can give it a shot.  I found that to be one of the ones that use somfy_remote_lib and it only sends commands for Somfy anyway.

FlaMike, 2023-05-02:
I guess that's not too surprising, if a little disappointing.  I prefer to keep things as general  purpose as practicable.

I'll close this thread for now.  If I learn anything "interesting" once I get the beast & play for a while, I'll let you know what I find out.

closed 2023-05-02 by FlaMike