rstrouse/ESPSomfy-RTS #35: EspSomfy shade state issue

Anashost, 2023-04-30:
So I'm using espsomfy for over a month, works perfectly but this week something strange is happening. 

So I keep my shades open 100% most of the times, but at random times once or twice a day, the state of all the shades in the software (also in home assistant) becomes closed, without the shades moving at all.

So, every time this happens, i need to put them back to 100% using remote or espsomfy software (which technically they already are at 100% because thet didn't move in the first place). Then they will act normal until it happens again.

I'm running latest v1.5.1, same issue with v1.5.0.

My configuration:
[image]

rstrouse, 2023-05-01:
That is very odd.  I wonder if we are picking up a signal that it thinks is a remote.  Open a browser and leave the frame log open.  Lets see if there is a frame from coming from somewhere.

Anashost, 2023-05-01:
Ok, i will keep the frame log open in my laptop over night (as im based in Europe), and report back tomorrow.

Anashost, 2023-05-02:
So, this problem didn't happen again just yet, but i did notice a lot is going on in the fram log mostly unknown signals in red. I wonder how would an outside signal make my shade state change, if it doesn't have the same address.
(I didn't use my shade or remotes at all during this time).
Here is some code captured from the frame log.

`
[{"encKey": 0,
  "address": 4720640,
  "rcode": 4688,
  "command": "Unknown(0)",
  "rssi": -82,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [1090,697,1569,459,2215,77,257,309,1845,51,126,151,2431,26,173,272,2189,75,4837,25,205,260,3484,103,155,181,14009,104,155,403,14010,99,173,198,247,41170,24,5026,75,6310,465,10048,75,85545,27,76,7123,258,3948,127,180,513,97829,274,95406,99,7265,251,1326,150,175,449,1949,101,150,177,91399,99,7202,276,2230,99,174,521,5220,101,152,204,74168,50,13100,50,4795,51,2291,457,444,568,48,146,94267,100,7115,269,3359,131,183,260,91964,24,7396,253,2216,101,200,277,92764,72,7107,256,2242,258,80062,135,231,467,78,604,25,101,203,1126,100,10780,152,6962,573,1991,346,395,692,26,4709,25,149,299,348,5419,24,74,6280,76,5435,51,5381,76,260,311,64962,74,7099,358,102488,166,95202,48,7117,200,5856,125,175,274,4041,201,1698,48,122,244,5776],
  "time": "2023-05-01T14:28:01.059+0200"
},
{"encKey": 0,
  "address": 8421376,
  "rcode": 32896,
  "command": "Unknown(0)",
  "rssi": -84,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [852,94501,7024,959,101724,25661,76642,2242,71,143,217,3024,25,74,89889,7194,102708,3836,517,2012,25,75,374,2618,51,229,305,3004,103,154,180,2885,79,5075,51,2359,77,5327,101,231,257,334,4430,51,62822,97,345,827,24,49,595,101,302,5944,582,2349,52,128,254,383,742,884,150,90466,149,6562,365,9910,124,200,664,85250,128,7211,354,1117,151,204,512,1495,127,176,249,8365,207,258,862,1257,250,275,398,422,544,1166,80,1436,149,78442,126,6668,499,95250,75,7050,272,2257,154,180,206,359,93113,51,7067,229,5860,63,190,265,2519,75,176,328,1505,73,838,52,77,327,376,933,23,124,279,305,435,540,41583,592,40933,124,6344,518,84033,24,49,73,171,11536,100,7017,325,426,502,750,255,333,94152,26,7067,255,2208,103,155,281,82546,150,222,467,76,593,125,151,886,103,8573,149,7090,504,45075,49,49751,121,7077,256,94836],
  "time": "2023-05-01T14:28:17.731+0200"
}]

[{"encKey": 0,
  "address": 9470088,
  "rcode": 1172,
  "command": "Unknown(0)",
  "rssi": -81,
  "bits": 56,
  "valid": false,
  "sync": 7,
  "pulses": [8264,2226,109,154,281,18107,73794,8257,94124,9567,5798,93137,4395,1359,2903,47,315,339,2852,77,231,282,2862,73,2878,75,223,298,2899,78,257,357,2889,49,220,294,1866,76,151,176,5776,55,1940,98,218,242,267,2772,126,6162,75,420,498,1362,129,4554,103,2722,46,172,197,297,373,2866,74,248,297,2891,76,254,353,1900,102,37950,124,8320,329,2272,101,175,567,2189,128,154,204,89068,103,8297,295,94094,74,10186,223,5825,102,153,255,75660,75,103,203,440,518,550,1014,98,786,222,7832,179,8310,582,452,330,103234,280,1122,101,125,174,517,1940,181,284,361,465,23,126,254,2689,50,282,308,1859,51,127,254,5228,26,2192,50,2665,146,272,347,2896,76,274,299,2661,54,196,318,1687,98,66316,102,8332,400,2532,126,176,572,2917,72,173,276,1993,26,86057,122,8337,268,10067,154,206,462,12299,99,173,198,71236,73,8317,310,2215,126,178,257,102031,243,1365,105,183,462,2440,99,147,220,8238,25,3077],
  "time": "2023-05-01T15:38:33.361+0200"
},
{"encKey": 4,
  "address": 1916192,
  "rcode": 32768,
  "command": "Prog",
  "rssi": -80,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [1528,606,10489,1084,1038,9837,591,573,716,1800,49,327,352,3825,452,2480,75,275,325,2893,77,230,305,2906,74,230,309,2898,76,230,307,4427,68,651,153,1122,177,5287,51,3556,152,16065,24,220,294,534,1308,152,8311,550,5550,128,205,691,8958,103,78345,75,8258,378,3954,589,6035,101,83564,25,8329,251,1164,256,92916,78,10141,278,5833,253,86439,28,8275,271,1265,25,260,101140,251,3752,98,90346,100,7870,279,94527,24,10175,180,92205,74,10187,203,92219,76,8268,253,52273,51,102,21598,206,537,153,609,206,487,26,60,643,147,247,662,177,253,792,278,329,7737,130,674,321,370,638,7456,251,8260,576,3222,147,171,195,413,90374,75,8790,272,2211,310,91436,26,6931,277,95432,76,8294,227,94119,51,8765,175,93623,74,8240,295,102458],
  "time": "2023-05-01T15:38:35.069+0200"
}]
`

rstrouse, 2023-05-02:
Something out there is transmitting on 433Mhz.  If they are in red then it failed the decoding and will not be processed.  Are you aware of anything nearby that may be transmitting?

Anashost, 2023-05-02:
I live in a crowded 9 floors building, sure lots of neighbors has there own blinds and balcony shades. While most signals are in red, small amount of these signal are not in red, but of course has different addresses than mine.

Anashost, 2023-05-02:
Thanks, i will keep you posted in case something weird happens, as im still monitoring the frame logs in a 24/7 vm.

Anashost, 2023-05-08:
So unfortunately this issue keeps happening.

The state of all three shades i have become closed without them actually moving. This happens at random times when I'm not using the shades.
[image]

I've managed to capture the frame logs while its happening, the FIRST row was captured exactly the same time when the state of my shades becomes "closed". It's marked as DOWN, and thats what happened to my shades, the went "virtually" down.

(The last 5 frames at the bottom are related to my 3 shades, if that helpful!)
[image]

[image]

I was not using espsomfy or the somfy remote when this happened, as it happened at night.

Here is the frame the frame logs:

```
{"encKey": 9,
  "address": 0,
  "rcode": 18696,
  "command": "Down",
  "rssi": -81,
  "bits": 56,
  "valid": true,
  "sync": 5,
  "pulses": [457,1808,75,149,224,73652,11925,588,623,1146,4622,6325,631,1401,741,4210,1891,52,2145,76,2633,51,2605,78,2743,51,5016,52,234,283,2636,51,152,177,31916,127,228,461,76,654,2855,125,1297,104,2364,50,1367,74,1550,268,393,418,495,50,12451,366,539,23,1778,52,11692,123,6354,502,101895,174,1173,545,725,125,228,277,379,429,1128,25,75,245,294,92488,100,6355,397,2344,151,229,638,3567,102,2359,50,4931,57,2984,75,2130,77,1929,51,1958,77,105,203,73204,98,6532,327,1329,531,1765,74,99,148,172,221,61348,25,30879,125,6344,346,1571,77,179,456,3153,129,337,465,1828,69,147,200,88609,99,6348,297,102404,179,96037,77,6367,224,2241,277,25942,25,107,152,562,51,5819,102,61507,50,6308,335,2158,126,201,432,665,99,1804,169,195,266,289,337,385,435,48511,50,42918,102,6351,377,3455],
  "time": "2023-05-08T22:31:56.442+0200"
}
```
Hope this info helps debugging the issue.

rstrouse, 2023-05-08:
Ok so I think I know what is going on.  It is one of the following so keep reading until the last one.

1. You have a mystical spirit running around your house pushing buttons.
2. All at once and in the middle of the night all of your neighbors have decided to decode the Somfy protocol and are sending 433Mhz signals at random.  ESPSomfy RTS is just listening in and playing along.
3. I was not filtering an address of 0 when the rest of the frame was valid.  This meant that any time something out there sent a frame with a 0 address, ESPSomfy RTS though it was linked to all your motors since the linked remote slots are initialized to 0.

My guess is number 3.  This feature while fun was not intended.

If you update your firmware and application to v1.5.3 the spirits will have be banished to the underworld.

Anashost, 2023-05-09:
I'm glad no mystical spiritss were involved, lol. I've just updated to 1.5.3, hopefully it fix the issue. Really appreciate your work, Thanks.

rstrouse, 2023-05-27:
I am going to close this for now.  I hope the spirits have stayed away.

closed 2023-05-27 by rstrouse

Anashost, 2023-05-27:
@rstrouse spirits are gone for good, been working flawlessly since 1.5.3 really thanks for the integration and support.