rstrouse/ESPSomfy-RTS #36: Having trouble getting a strong signal

bradphillips921, 2023-05-03:
I'm using this hardware
ESP board - https://www.amazon.com/gp/product/B09XDMVS9N/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1
Transceiver - https://www.amazon.com/gp/product/B09CTRYV4L/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
Antenna - https://www.amazon.com/gp/product/B072Q7319V/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I can't get a dBm better than -66 and I can't get anything with my RX bandwidth below 125kHz.  Also, if I press the RTS remote 30 feet or more away from the ESP Somfy RTS device it doesn't register. I was under the impression that the remote should work from 75-100 feet away. Am I just being unrealistic? I have a decent size house (5,500 sq ft) so I need the ESPSomfy RTS to pickup remotes from a decent distance in order to keep the shade position in sync in HA.

Should I be getting better results than this? Any reason I need my RX bandwidth so high in order to pick up a signal?
[image]

rstrouse, 2023-05-03:
This should not be.  I am dealing with a much larger distance and I have many obstacles that should impact the signal.  In every instance I have found the remote to have less distance than the device.  That being said, I do have some transceivers, even from the same batch that perform better than others.  Maybe that is why you often find them in two packs.

If it were me I would buy another transceiver to see if I could get it to improve.

rstrouse, 2023-05-03:
You could have an interference issue but you could also have power issues.  It is very odd that you are not getting better receive numbers.   I am adding some additional filters for the pulse detection to reject frames that have been clobbered earlier but that won't change the RSSI values.

rstrouse, 2023-05-04:
I posted a beta firmware to the repo but it is not yet released.  In this versions I reset the pulses faster when there is interference.  Load the firmware and application bin files from the repository (not the release) and see if the capture improves.

As an aside you can also install multiple ESPSomfy-RTS devices if there are some remote areas.  My roller shades are outside in an outdoor living space and the transceiver is located inside the house on the second floor and it never misses a command.  That antenna you turned me on to allowed me to move the transceiver into my server closet.

bradphillips921, 2023-05-04:
Thanks for all the help. I tried the beta software, it helped a little. 

The transmission distance is great! It will open/close a shade from across my house.  It just doesn't pick up the remotes from very far. Maybe it's the remotes I have ([Situo 1 RTS](https://springblinds.com/motorized-shades/somfy-motors-sensors-controllers/somfy-remote-situo-1-rts-pure-remote-ii-1-channel-mpn-1870571/?gad=1&gclid=CjwKCAjwjMiiBhA4EiwAZe6jQ3_xs8mVWMMD6GIbkoTTjMpazTRjwI1PbEtUiHjnK4USiuGAcxazkhoC-CkQAvD_BwE)).  For now, I just decided to build 3 more and put them in different places around my house.

rstrouse, 2023-05-06:
Somfy remotes do not transmit a very strong signal.  In my case I have no other 433Mhz stuff floating around but I do get really good receive abilities.  I did install in a neighbor's house that is just under 5k sqft and it has no problem hearing the remotes that are in each room.  Those are all Situo 5s and Telis 16s spread over 30 shades.  I built them a second device but they didn't need it so it sits here on my desk as a test device.

My shades are down one floor and outside through 1.25inches of solid travertine, a teak ceiling, and two stucco walls.  All told just over 7400sqft and it never misses a beat.  Funny I only have 3 shades but they span an area of over 40 feet.  I'm starting to think that I need more shades.

I wonder if there is a lottery for the transceivers since they vary or if it has something to do with interference.  Either way it is pretty cheap to build multiple devices and they are not very big.

rstrouse, 2023-05-07:
@bradphillips921 create another issue if you do not get sorted out but I suspect multiple devices will solve it for you.  I will keep track of these where the receive is weak.

closed 2023-05-07 by rstrouse