rstrouse/ESPSomfy-RTS #37: slow hass integration communication and web-ui unresponsive

mihaiaka, 2023-05-05:
Hi. I stumbled upon an issue.
A few days later deploying this, I am unable to open the web-ui anymore, two circles are spinning, but nothing happens:
[image]
Eventually, after a few minutes, I get this message:
[image]

From the beginning, I've also installed the HACS integration (ESPSomfy RTS) and that still works, I'm still able to control the (4) shades I have but has a noticeable 3 seconds delay from the moment I press something and the motor actually moves.

Is there something I can do? Or should I reflash the ESP board? (of course, I didn't back-up anything, and I might have to re-pair the shared/remote/etc, which I might want to avoid by reaching out here for help).

Thank you,
Mihai.

rstrouse, 2023-05-05:
Which ESP32 version do you have?  In every installation I have seen, this is instantaneous.

The webserver on ESP32 only supports 5 clients at one time.  Under normal circumstances this should be enough but you could potentially run out if you have the GUI up on multiple devices.  Check to ensure you do not have multiple browser windows open and check other machines that may have it open in the background.

mihaiaka, 2023-05-05:
ESP32 version: esp-wroom-32 (written on the chip)

what I also tried (to narrow our hardware issues):
- to boot without the CC1101 module
- change the power cable and power adapter
- disable the HACS integration
unfortunately, nothing changed.

As per the number of clients... I'm trying to open (personally) the web-ui from 1 laptop and/or 1 phone. But e HACS integration counted as a client as well? Would that count as 1 client for the 1x home assistant server? or X amount of clients for X amount of tables/phones that are running the home assistant companion app?

Is there any way to grab a log or something from it (via USB/etc) before I would re-flash it? (if that would even be useful for something?) ?

Thank you.

mihaiaka, 2023-05-05:
Just to add, I'm using a mesh system for my WIFI, could that cause any issues to it?
(long story short, 2x netgear R7000 flashed with a Merlin AsusWRT clone firmware = xwrt >> running Aimesh)

rstrouse, 2023-05-05:
1 HASS server counts as 1 connection and I am running AIMesh as well.

I was just going to tell you to connect the serial console via esphome or https://espressif.github.io/esptool-js to see what is reporting on those logs. 

Do you have an MQTT server set up.  It is not required nor desired for Home Assistant integration.  That is eating at least one connection.  Double check your phone to see if you have the ESPSomfy-RTS client currently running in a browser.  If you do uncheck the Enable MQTT client for now.  The connection parameters are clearly not right or the identified server does not exist.  Perhaps I need to disable MQTT if it fails repeatedly.

mihaiaka, 2023-05-05:
thank you very much for taking the time to write back.
So, in the end, I've just started from scratch. I've refreshed the ESP and re-paired the blinds. Once everything was successfully done and confirmed working I made a "day0" backup. :) This should teach me a lesson.

About MQTT, yes I'm running the mosquito broker as I have another ESP for the gas and electricity meter via a P1 plug (a geek alternative to the "homewizard p1". That one works fine, so I would assume that the MQTT server is fine (address/credentials/etc), but I could've done a typo during the configuration of the ESPSomfy-RTS.
Nevertheless, the ESPSomfy-RTS was initially planned to make use of MQTT as well, only to find the HACS integration a bit later (which is great, btw). I never took the time to disable that but I also did not think this will affect the controller that badly.
Maybe this can be fine-tuned a bit.

All is fine now, the fact that 4 blinds are not that many, I went with the clean reinstallation varian.
Let me know where one can "buy a coffee" / support you/your work.

rstrouse, 2023-05-05:
No need for coffee.  I will make the reconnect for MQTT not be so aggressive.  This will provide some guardrails against oversized fingers.  Right now it continually tries to reconnect when that connection is lost.  This makes that poor little D0WD processor burn more cycles than it has and steals any chance of high speed socket communications.

Normally, pub/sub for MQTT is not an intensive operation.  However, the initial connection negotiation appears to be.  This could also be related to waiting on responses from MQTT during the initial connection.  If that server is busy it would wait to find a response slot.

I connected to a mosquitto on a Pi4 and on my homeserver with no issues with drag on the device.  I even put in an invalid hostname.  However, if I hosed up the port or the authentication it spun ESPSomfy RTS out.  The aforementioned fixes above will take care of that.  I will simply not attempt to reconnect if I have tried within the last 15 seconds or so.

Also, like all `things` on your network you should reserve the IP address in DHCP on your router.

closed 2023-05-05 by mihaiaka