rstrouse/ESPSomfy-RTS #38: Cannot pair my blinds. Command not recognized.

FredericoCBarbosa, 2023-05-07:
Here's de log of a up signal from the remote: 
[{"encKey": 134,
  "address": 2113079,
  "rcode": 551,
  "command": "Unknown(15)",
  "rssi": -45,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [500,607,626,585,653,584,851,640,511,681,642,464,644,864,473,2851,2604,2503,2612,4862,1316,1268,663,638,650,647,651,638,1284,650,652,1287,666,629,1309,613,672,631,659,642,641,1285,1288,648,653,1293,1292,637,666,638,645,626,656,1285,666,631,1311,1293,1281,1298,1305,612,672,623,679,615,669,1269,659,631,1304,651,656,632,652,642,655,633,638,647,670,1256,667,642,1300,1286,649,635,664,643,642,650,659,641,665,620,649,635,1316,619,674,613,665,1290,1298,644,647,645,652],
  "time": "2023-05-07T10:17:26.322+0100"
}]

And here's the same signal picked-up by de RTL_433 software

time = 2023-05-07 10:18:44.537032
protocol = 167
id = 3620384
control = Up (6)
counter = 553
retransmission = 0
mic = CHECKSUM
mod = ASK
freq = 433.38912
rssi = -0.0989304
snr = 5.71867
noise = -5.8176

It seems that the command is not recognised by the ESPSomfy-RTS. 
Any tips?

rstrouse, 2023-05-07:
Which remote do you have?  The pulses do pass the checksum.  The response from the RTL_433 looks like the up+down buttons were pressed at the same time.

If you install the firmware and application files from the main repository I have added some additional filters for the pulses.

rstrouse, 2023-05-09:
If you install v1.5.3 release there are additional checks for stray frames.

rstrouse, 2023-05-12:
It looks like these while they use the same encoding as Somfy do not use the same command numbers.  Can I get you to press each button in the following order.  Press it only once and no need to hold the button down and it doesn't matter if the shade does anything or which channel it is on.

1. The program button on the back of the remote.
2. UP
3. STOP
4. DOWN

FredericoCBarbosa, 2023-05-12:
Here:

[{"encKey": 140,
  "address": 699434,
  "rcode": 2862,
  "command": "Unknown(15)",
  "rssi": -49,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [28092,4366,2566,2580,2542,2575,2540,2592,2533,2574,2550,2568,2562,2572,2546,2567,4891,1267,1319,632,638,646,657,1265,671,647,1285,644,668,635,660,1267,668,608,663,652,1298,1261,1330,626,662,630,663,1273,659,627,668,627,661,640,650,620,676,637,633,648,1294,1293,1295,1288,1302,629,663,639,649,1296,1289,1296,1285,1298,651,637,1309,1271,660,638,642,632,659,640,651,643,653,639,1307,1272,659,625,669,627,661,641,649,619,1327,1276,645,637,665,613,1324,1259],
  "time": "2023-05-12T19:51:06.648+0100"
},
{"encKey": 134,
  "address": 699434,
  "rcode": 2863,
  "command": "Unknown(15)",
  "rssi": -43,
  "bits": 56,
  "valid": false,
  "sync": 7,
  "pulses": [1118,1983,77,308,383,2004,25,51,305,354,2016,49,297,2934,2590,2531,2572,4875,1302,1307,620,655,640,656,642,648,1288,648,646,1302,627,667,1281,656,620,670,634,1313,1267,1312,1270,1305,1275,673,613,661,646,645,631,659,631,670,621,1309,646,642,1286,1306,1272,1311,642,655,638,665,1267,1308,1274,1302,1291,644,631,1312,1269,659,633,664,645,651,635,666,616,650,636,1309,1272,659,643,663,618,677,638,641,649,1296,1290,644,622,660,637,1318,1270],
  "time": "2023-05-12T19:51:08.355+0100"
},
{"encKey": 133,
  "address": 699434,
  "rcode": 2864,
  "command": "Unknown(15)",
  "rssi": -43,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [2576,2571,2545,2569,4885,1265,1318,639,659,617,670,636,661,1275,1299,1293,1292,1292,666,624,661,633,640,643,1305,1277,648,637,1310,1272,663,639,644,652,1287,648,667,634,639,654,644,651,659,1269,1316,614,666,647,644,630,661,655,645,647,659,632,640,1295,1310,618,675,1264,1322,1258,1330,1275,658,620,669,633,1288,644,671,1265,647,643,1305,1279,669,613,1309,644,655,1268,669,622,1309,647,667],
  "time": "2023-05-12T19:51:09.052+0100"
},
{"encKey": 136,
  "address": 699434,
  "rcode": 2865,
  "command": "Unknown(15)",
  "rssi": -42,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [2568,2571,2544,2571,4891,1276,1297,640,673,619,656,1291,1295,634,660,647,643,632,663,1291,641,633,660,642,647,643,1307,1275,1300,655,642,1275,659,646,643,632,1311,644,646,637,661,1271,1329,1267,1317,615,680,616,677,612,658,646,670,632,664,614,669,1273,1310,644,651,1266,1320,1262,1329,1275,662,618,666,635,1286,643,674,1264,649,643,1305,1274,675,612,1312,642,654,1266,669,623,1307,647,666],
  "time": "2023-05-12T19:51:09.826+0100"
}]

Thx

rstrouse, 2023-05-12:
Wow look at that.  The command byte is replaced by the encryption key.  Do you know who the manufacturer is?  This is definitely not RTS but it is already decoded.  Google is failing to come back with any references to Jove that is a shade motor?

FredericoCBarbosa, 2023-05-12:
Here is the log again:

[{"encKey": 140,
  "address": 699434,
  "rcode": 2870,
  "command": "Unknown(15)",
  "rssi": -55,
  "bits": 56,
  "valid": false,
  "sync": 6,
  "pulses": [649,608,1168,1382,462,1747,2249,75,1720,2025,23,246,342,1983,48,268,2829,2591,2533,2599,4874,1281,1310,620,655,640,658,1273,679,609,1330,625,659,643,648,1271,658,629,666,648,658,610,661,645,1306,1266,1323,1267,667,628,663,640,1292,1275,687,623,1295,638,657,1273,1308,646,643,633,660,654,647,648,659,615,678,617,678,1258,1317,635,661,1277,1300,1290,1297,1289,640,651,660,633,1292,648,658,1283,641,631,1313,1295,633,633,1306,650,660,1267,666,634,1315,617,675],
  "time": "2023-05-12T22:29:25.376+0100"
},
{"encKey": 134,
  "address": 699434,
  "rcode": 2871,
  "command": "Unknown(15)",
  "rssi": -50,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [2562,2585,2550,2573,4872,1276,1309,646,651,640,658,643,645,1290,649,643,1305,625,668,1277,665,617,676,610,659,646,644,631,1314,643,647,638,660,1274,675,613,660,645,1294,1291,644,651,658,619,1324,1278,1308,622,654,641,656,642,652,633,658,646,669,632,661,1266,1320,639,651,1271,1311,1270,1306,1271,677,637,660,620,1320,633,648,1272,659,642,1299,1284,654,643,1299,639,649,1294,663,614,1322,632,662],
  "time": "2023-05-12T22:29:27.413+0100"
},
{"encKey": 133,
  "address": 699434,
  "rcode": 2872,
  "command": "Unknown(15)",
  "rssi": -49,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [2577,2568,2562,2574,4871,1287,1300,633,663,629,671,621,683,1269,1293,1290,1300,1283,655,644,646,637,1312,620,674,1266,673,620,1307,1275,677,611,659,647,642,633,1313,642,649,638,660,648,642,1282,1305,651,659,614,677,617,676,642,655,618,669,635,661,1276,1297,653,645,1275,1309,1294,1288,1292,660,615,677,620,1325,627,668,1252,684,618,1319,1266,670,627,1311,620,669,1286,638,644,1306,625,666],
  "time": "2023-05-12T22:29:28.197+0100"
},
{"encKey": 133,
  "address": 699434,
  "rcode": 2872,
  "command": "Unknown(15)",
  "rssi": -49,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [33087,2568,2576,2544,2576,2540,2593,2536,2599,2520,2591,2538,2568,2545,2579,4903,1270,1306,624,669,629,661,629,665,1275,1316,1276,1306,1270,656,640,655,642,1299,641,648,1297,662,626,1294,1290,649,643,652,639,660,618,1321,641,651,619,674,636,658,1271,1307,647,642,633,661,630,669,648,660,632,650,644,653,1265,1322,634,662,1274,1297,1291,1292,1289,666,628,659,641,1302,627,660,1268,667,632,1311,1269,657,636,1302,650,660,1268,667,629,1311,620,671],
  "time": "2023-05-12T22:29:28.359+0100"
},
{"encKey": 136,
  "address": 699434,
  "rcode": 2873,
  "command": "Unknown(15)",
  "rssi": -48,
  "bits": 56,
  "valid": false,
  "sync": 7,
  "pulses": [2272,75,1731,2257,75,1743,2259,74,1743,2250,100,2039,50,1699,2019,26,1999,52,280,330,1994,52,80,2688,2581,2551,2571,4868,1273,1314,651,657,615,677,1269,1312,642,645,637,660,649,642,1281,663,619,675,640,1309,618,655,1286,1316,637,660,1272,654,639,658,619,670,636,1311,618,676,1266,1297,1284,1307,649,658,633,650,643,653,639,657,645,646,634,665,1276,1299,651,640,1276,1318,1269,1312,1272,655,658,640,646,1304,622,668,1283,656,619,1320,1291,643,637,1311,646,650,1263,672,620,1311,647,643],
  "time": "2023-05-12T22:29:29.095+0100"
}]

In the meantime I've found the box and the manual of the remote. 
The box advertises it as a Somfy , but apparently it's not.

Product code:  SMF-ASA 2211118 

[image]
[image]
[image]

Thx

FredericoCBarbosa, 2023-05-12:
It's not chinese but Italian.  See   www.asamotor.com

rstrouse, 2023-05-12:
Ahah so this is Somfy RTW not RTS.  All is not lost as it appears the command byte is not encoded and is also used to encode the frame.  All of the commands repeat appropriately so we will need to do this in two steps.  The first step is to translate the incoming frames then make sure we are sending them back out correctly.

So now can I get you to capture the following combined button presses where both buttons are depressed at the same time in the following order. 
1. Up + Down
2. My + Up
3. My + Down
4. My + Up + Down

rstrouse, 2023-05-13:
After you capture the commands above and post them here.  Load the firmware from the code repository by clicking on `SomfyController.ino.esp32.bin` and the application by clicking on `SomfController.littlefs.bin`.

After that go into the gears menu then select the Somfy tab.  Then click Configure Transceiver.  Change the protocol from RTS to RTW and click Save Radio.
[image]

Then for each shade created edit the shade and change the protocol  to RTW from RTS and save it.  It should then start talking RTW instead of RTS for your shades.  At that point you should be able to pair them.  But I do need the other commands or it can spin out the controller when it sees these and doesn't know what to do with them.
[image]

FredericoCBarbosa, 2023-05-13:
Up + Down
My + Up
My + Down
My + Up + Down

[{"encKey": 138,
  "address": 699434,
  "rcode": 2879,
  "command": "Unknown(15)",
  "rssi": -34,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [514,2695,2571,2548,2568,4886,1290,1295,639,654,643,648,1289,1317,1252,1310,646,654,1285,665,627,661,614,1327,1280,1309,619,658,639,654,1297,657,617,677,618,674,612,658,647,643,632,664,641,1316,1253,1312,643,655,641,654,643,647,637,659,649,640,633,662,1279,1318,641,652,1270,1311,1270,1311,1271,677,635,640,642,1299,652,665,1277,658,622,1317,1267,667,627,1311,620,670,1285,639,645,1303,640,644],
  "time": "2023-05-13T07:59:40.213+0100"
},
{"encKey": 135,
  "address": 699434,
  "rcode": 2881,
  "command": "Unknown(15)",
  "rssi": -41,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [2563,2592,2547,2569,4880,1278,1308,622,659,640,677,618,673,1264,673,619,656,639,1308,1272,659,622,666,631,661,640,648,644,1304,1275,1324,1251,671,637,662,625,1295,1287,664,622,1308,647,669,632,663,1250,670,635,1310,1268,661,630,664,623,1307,646,670,1258,679,617,650,636,666,623,1320,1289,645,621,1309,647,668,1262,1303,650,659,616,676,1269,661,641,1301,1279,659,640,648,634,1311,1269,663,640],
  "time": "2023-05-13T07:59:50.633+0100"
},
{"encKey": 137,
  "address": 699434,
  "rcode": 2882,
  "command": "Unknown(15)",
  "rssi": -34,
  "bits": 56,
  "valid": false,
  "sync": 9,
  "pulses": [1703,2002,50,280,354,1998,155,306,357,2005,51,130,154,282,357,1984,77,299,324,374,2026,27,128,2737,2592,2545,2559,4889,1273,1326,628,642,651,658,1283,1294,633,655,1293,1312,1268,654,635,666,616,676,636,658,620,671,633,1313,616,673,1266,664,623,671,614,1318,1268,1315,1270,1309,646,668,1262,666,617,1321,1267,668,635,663,623,1322,637,657,1271,652,640,659,617,670,635,1311,1270,659,641,1300,630,660,1267,1318,642,654,643,649,1286,650,644,1305,1274,648,638,659,623,1318,1266,669,647],
  "time": "2023-05-13T07:59:52.550+0100"
},
{"encKey": 139,
  "address": 699434,
  "rcode": 2883,
  "command": "Unknown(15)",
  "rssi": -35,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [2593,2563,2553,2572,4876,1275,1333,619,657,640,653,1270,1336,1270,653,639,1306,1272,660,625,664,630,662,639,651,645,650,614,686,618,1318,1267,669,626,659,635,1297,1281,1305,649,657,615,680,616,675,1268,670,623,1305,1275,673,619,656,644,1294,639,679,1270,660,614,677,618,676,611,1311,1271,663,639,1318,627,661,1283,1293,634,664,644,652,1288,637,650,1312,1273,651,634,665,618,1327,1278,645,637],
  "time": "2023-05-13T07:59:58.328+0100"
}]

FredericoCBarbosa, 2023-05-13:
Upgraded  to the beta version.
I guess it should work now. 
I've not tested the blinds, but the log reports the keys presses perfectly:

[image]

[{"encKey": 140,
  "address": 699434,
  "rcode": 2889,
  "command": "Prog",
  "rssi": -28,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [42904,4582,581,30485,66749,469,4014,24189,73736,4545,530,97339,4742,65434,10106,95768,2553,2598,2542,2563,4886,1290,1295,635,666,625,667,1283,664,621,1323,611,680,616,674,1267,671,619,659,637,1294,1297,1291,1295,1311,1271,652,641,639,646,645,631,664,641,664,618,1303,652,644,651,657,1266,667,635,1315,1270,660,618,662,639,1299,639,672,1271,664,627,642,651,661,614,1319,1273,667,624,1322,612,664,1274,1328,625,667,630,659,1282,641,630,1314,1292,660,630,640,646,1305,1275,647,638],
  "time": "2023-05-13T08:05:36.427+0100"
},
{"encKey": 134,
  "address": 699434,
  "rcode": 2890,
  "command": "Up",
  "rssi": -28,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [465,666,648,690,450,690,585,610,471,568,25799,465,2684,49,40244,8097,10087,95753,2567,2593,2538,2573,4897,1266,1321,632,662,626,668,627,660,1268,668,628,1314,641,645,1289,648,644,653,638,1309,1271,662,624,1319,613,684,1274,651,639,660,619,669,610,686,628,1294,1289,1298,631,660,1292,642,632,1313,1270,681,609,679,617,1326,624,669,1278,658,620,669,633,662,615,1321,1261,681,615,1299,27,653,665,1277,1288,650,658,640,652,1267,662,640,1300,1280,658,642,644,637,1311,1269,658,642],
  "time": "2023-05-13T08:05:38.189+0100"
},
{"encKey": 133,
  "address": 699434,
  "rcode": 2891,
  "command": "My",
  "rssi": -30,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [458,659,1362,2600,2561,2563,2571,4871,1287,1299,631,663,652,644,651,657,1266,1318,1266,1322,1259,680,616,674,615,1309,1296,634,644,669,619,1308,1275,673,613,661,646,642,633,660,639,1300,631,661,655,643,650,659,1256,667,629,1313,1267,684,632,640,645,1303,624,667,1280,664,616,677,613,658,647,1294,1292,642,651,1311,618,671,1260,1329,627,661,640,650,1272,659,631,1321,1286,649,621,674,635,1290,1277,657,645],
  "time": "2023-05-13T08:05:38.995+0100"
},
{"encKey": 136,
  "address": 699434,
  "rcode": 2892,
  "command": "Down",
  "rssi": -30,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [558,741,2413,23,38255,937,54417,4520,850,82329,10107,95746,2564,2575,2546,2573,4892,1271,1312,643,653,638,657,1270,1307,646,643,634,659,641,673,1270,663,639,645,624,1308,645,668,633,662,615,670,623,683,1267,645,649,659,616,676,617,1328,1277,644,636,1317,637,655,1259,670,623,1307,1273,675,641,658,623,1318,615,679,1268,661,639,651,619,675,614,1310,1296,660,627,1295,635,660,1275,1327,624,670,627,658,1292,642,629,1312,1269,656,634,680,617,1300,1277,670,635],
  "time": "2023-05-13T08:05:39.827+0100"
}]

I will try  to pair the blinds later in the day. Thanks.

FredericoCBarbosa, 2023-05-13:
I still can't get it to pair. 
But i suspect that there is something wrong with my CC1101. The emiter is not picked up by the RTL_433.
It should show up, wright?

How can i diagnose emiter of the the CC1101?

rstrouse, 2023-05-13:
In the software switch the RX and TX setting.

FredericoCBarbosa, 2023-05-13:
Got it working!   

I've been able to pair all of my 5 blinds. I've done what you said, switched the pins in the software.
I really appreciate your effort. Thanks for your work and insights. 

I'll update to the latest beta  and give you the feedback.

rstrouse, 2023-05-13:
Please let me know if all the combination commands register properly.

FredericoCBarbosa, 2023-05-14:
Reading through the manual, it seems that the ASAMotor remote controls (the RTW elegance series) do not have the same functions as the other Somfy RTS.
There is no MY button, but only a STOP button. In fact, I was unaware of the possible existence of the MY function, since it was not mentioned to me by the installer.
So the button combinations do nothing (I tested both on the remote control and in the application). It looks like these additional functions have been suppressed by ASAMotor.
Here is the link to the remote control manual, which explains the different functions: 
https://www.asamotor.com/system/be/documentViewer.stub.php?xfile=652bed2f0088d74f26009425d3c5fe7dc19dd1de3b2ab10c35ef4245114cc6637f3462a3d28060e704048d950d8761ec7ab6be7f7a708acc36db15a8d309b48064f0ce3418fab0586c3f31904bf30e7659&mode=download

I don't know if there are other functions reserved for installers - the manual does not mention them - such as the setup of endstop points, but, as far as I can see, RTS combinations do not work with RTW remote controls.

rstrouse, 2023-05-16:
I have published release v1.6.0 which contains the support your your motors.

FredericoCBarbosa, 2023-05-17:
Again, thanks for your effort. 
I'll update to the new version. If I spot anything wrong, I'll post it here.
Regards.