rstrouse/ESPSomfy-RTS #40: Not able to receive nor transmit signal

Gerigot, 2023-05-08:
Hi everyone I really love that project but unfortunately I've tried to make it work with a somfy shade but I can't even receive the signal from the remote controller.

I've tried with two esp32 module and 2 different 8 pin cc1101 module but without success.
I've checked if every connection is solid and if everything is connected to the proper pin many times.
I've also tried to change the pin for GDO0 (TX) fist to the pin 13 then tried on pin 4 both the RX and TX on the same

I'm really out of ideas can someone give me some advice? I've also ordered another cc1101 to test but I've already tried with two so I'm not sure this could help.

Just for info, My somfy remote is a Situo 1 RTS Pure II.

thank you

rstrouse, 2023-05-08:
Typically, if there is no response out of the transceiver it is either a wiring issue or someone is trying to control Somfy IO.  In this case you indicated RTS above.   Double check to ensure you have SPI pins correctly set.  Specifically, CSN, and SCK.

While it could be the transceiver, if you tried two already I suspect that there is either a setting or wiring that isn't right.  Which transceiver do you have.  Is it the one pictured in the wiki.  If not they may have re-arranged the pins and if it is pay close attention to the order in which the pins are situated.  Pin 1 is on the left bottom when looking at it with the pins facing away from you.

Gerigot, 2023-05-08:
Thank you for your fast reply, I'm using the one in the wiki (E07-M1101D written behind, on the pin side). 
Unfortunately I've already checked lots of time the wiring so I suspect if there is nothing else that probably it's another faulty module. :(

rstrouse, 2023-05-08:
There are two things.  First, if the module cannot be initialized it will show as Radio not Initialized in the header.  This will be true if the registers of the module cannot be written to or read from.  Most often this is caused by incorrect pin settings for MOSI and MISO on the `Configure Transceiver` page.  Next if you click the log button it should show it if is receiving pulses that match the hardware and software sync for the RTS protocol.

I have never seen the printing of the E07-M1101D with the pin ids written on the pin side.  Do you have such a picture or is it printed as below?  With the module pin side down, pin 1 is typically on the bottom left.  Unfortunately when you flip it over to actually connect it they are reversed.
[image]

This will also show if a frame is receiving and how strong the signal is in the `Configure Transceiver` page.

Gerigot, 2023-05-08:
Yes I've the same transceiver and the SPI connection is working I think something is probably wrong with the GDO0 pin but I've tested it with a tester in continuity mode between the pin 6 of the little chip and the pin on the microcontroller and it's beeping so I'm out of ideas...

The Radio can be initialized but if I click the Logs I can't get anything even if I click a button on the Remote already linked  to the shade.

My transceiver the front is same as yours the back is this:
[image]

rstrouse, 2023-05-09:
Hmmm.  There have been bad transceivers out there (I have had a couple in more than 20 of them).  Lets focus on your settings.  Shoot me a screenshot of your settings and a picture of the back of the ESP board with your connections.

Gerigot, 2023-05-09:
[image]

This is my configuration and this is a pic of the back of may esp32 devkit module:
[image]

I couldn't do a better pic and unfortunately I've only those color for dupont cable so I had to use same colors for some of them but the two alone on the esp32 are connected to the pin3 and 8 of the transceiver:
black D12 on ESP32 -> pin 8 transceiver
green/blue D13 on ESP32 -> pin 3 transceiver

rstrouse, 2023-05-09:
Show me the front and back of your remote.  If it says IO on it then it will be 868.95.

Gerigot, 2023-05-09:
Yes it is a Situo 1 IO Pure II.
Do you think, if I order a 868MHZ cc1101 and compile your project modifying the frequency, will it work?
I'm sorry for wasting your time I was not aware there was also a 868 version of the Somfy

rstrouse, 2023-05-09:
> Do you think, if I order a 868MHZ cc1101 and compile your project modifying the frequency, will it work?

Unfortunately, the problem isn't with the frequency it is the protocol.  Somebody felt that they needed to stand up cloud services because it makes sense to control your home by routing it through a server in a far off land.  Here is some further reading in case you are interested.

https://github.com/Velocet/iown-homecontrol

Gerigot, 2023-05-09:
ok thank you so in my shade they are using a totally different protocol not only a different frequency :(.
that's a pity.
I will close the issue.

thank you again