rstrouse/ESPSomfy-RTS #41: Weird Home Assistant Controls behaviours

isngofoz, 2023-05-08:
Firstly, thanks for your efforts on this solution. It certainly helped with automating my Somfy roller shutters with Home Assistant. 

However there are sometimes some weird behaviours from the Home Assistant controls. For example, I push the Up button in HA to raise the shutter, it goes all the way up, stops for a few seconds and then for some reason comes back down to an almost closed position on its own. The Up button appears still masked out, so the only way to gain control/sync again is to press the Down button in HA, which then closes the shutter fully and then I can raised the shutter again using the Up button. 

The weird part is that it does not happen all the time, just on occasions and usually when I have left the ESP on idle for some time (ie not used it for a day or logged into the webUI). 

I have just updated the firmware and application to v1.5.2 to see if it fixes the problem as I recalled updating the HA Add-on recently so maybe there is some fixes. 

On that note, it is difficult to add another entity to show that there is a new firmware update available?

afmaniacc, 2023-05-08:
Hi i have the same Problem, do you found a solution for that?

rstrouse, 2023-05-08:
Can I get both of you to try something for me.  With the shades completely stopped.  Press the my button on the remote.  If the shade moves I may know what it going on.  Are these shades set up as roller shades, draperies, or blinds?  If they are blinds what type has been selected for the motor type?

> On that note, is it difficult to add another entity to show that there is a new firmware update available?

I'll have a look to see if I can figure something out here.

isngofoz, 2023-05-08:
To be honest, I usually only used the MY button as a 'Stop' button on the remote. 

Anyways, I tried according to what you said and interestingly enough, I believe the shutter moved to that position I described above, almost closed but not closed. Mine are metal roller shutters so I choose Roller Shade as the type.

isngofoz, 2023-05-08:
> Ok so press the my button and let the shutter move to that position. After it is done moving hold down the my button on the remote until it jogs.

Okay, done. How does this solve the issue?

rstrouse, 2023-05-08:
This removes the my position that ESPSomfy-RTS is not currently aware of.  Next you need to update your firmware and application to 1.5.3 which does not issue a stop (my) command when the motor is expected to go to its limits.

rstrouse, 2023-05-09:
If you want to set a my position, set it first with ESPSomfy RTS by holding down the my button on the ESPSomfy RTS app until the Set My Position screen opens.  Move the slider to the new my position then press Set My Position.  The motor will move to this position and stop then it will jog indicating the my position is set.

From here on out if the motor is idle and you press the my button, it will move to the set position.

isngofoz, 2023-05-09:
> If you want to set a my position, set it first with ESPSomfy RTS by holding down the my button on the ESPSomfy RTS app until the Set My Position screen opens. Move the slider to the new my position [...]

For some reason, after clicking the Set My Position button, the motor moves to the position that I usually like the shutter to stop at but it does not jog. The My Position does get updated though but when I click My button the on-screen shutter simulates a move but the actual shutter stays put. This is running v1.5.3 .
[image]

I also tried it today on v1.5.2 and it also did not jog as well when setting the My position. It seems to me that there is an issue setting the My position using ESPSomfy. I just tried setting the My position with the remote and it worked.

rstrouse, 2023-05-09:
Thanks, I will have another look at that code.  It may not be holding the signal long enough.  It will be fixed in the next release.

isngofoz, 2023-05-09:
Thanks for responding back so quickly. 

I hope you don't mind me thinking out loud. I have learnt from above that the My Button should only be pressed (held) when the shutter is at idle to set the My Position. However with the ESPSomfy, the Set My Position is done in one click. Is it possible that the code is sending the My Position too early while the shutter is still moving?

Either way, thanks for checking. I will monitor the post and test the update when it is available.

rstrouse, 2023-05-13:
v1.5.4 fixes your My button woes.

isngofoz, 2023-05-13:
Hi there, thanks. Yes, it definitely did solve the My settings. 

Not a big deal but I did noticed that 55% in Home Assistant works out to be 45% when I was setting the My position. But I guess My Position isn't exactly used HA anyways so once it is setup, it doesn't really matter.

rstrouse, 2023-05-13:
That is correct.  The Home Assistant cover is displayed in percentage of open and cannot be changed without completely rewriting the cover domain component.  I cannot find words to describe how ridiculous that is but at some point I will recode the Home Assistant interface to use my own cover domain.

This means the 55% closed it 45% open in Home Assistant.  Hard to complain when it is open source and I need to rework it so it works logically.  I love Home Assistant but that really does put the ass in the assistant.  Is my type A showing?