rstrouse/ESPSomfy-RTS #42: Blind positions are incorrect, used to be OK

FlaMike, 2023-05-09:
I have 8 blinds controlled with ESPSomfy RTS and integrated with HA using the HACS integration.  When I set up the system, I timed the open & close times, added & paired the shades & linked the remotes.

I don't know when or how it happened, but I noticed this morning that 6 of the 8 shades do not show the correct position in HA or on the ESPSomfy RTS web interface.  It appears HA and the web interface are in agreement with the "incorrect" positions, however.

Can you please let me know how to correct the reported blind positions?  Do I need to delete & re-add the blinds & remotes?

Attached is a screenshot of the web interface.  All of the blinds are currently open 100%, but only the Master_BR_West and Casita blinds reflect the position correctly.

I was running 1.4.6 but updated a few minutes ago to 1.5.3.  I am running HACS integration 1.0.12.

Thanks for your help!

[image]

rstrouse, 2023-05-09:
No don't delete anything just press the up button on the ones that are out of sync in the ESPSomfyRTS interface and wait for them to finish.  Then update to v1.5.3

FlaMike, 2023-05-09:
And Home Assistant now shows all of the shades as "open."  I'm going to close this issue to save you the hassle.

Thanks & have a super day!

closed 2023-05-09 by FlaMike