rstrouse/ESPSomfy-RTS #44: Rogue shade activity

bradphillips921, 2023-05-10:
I have 3 shades on a remote channel. I paired them as one shade by pressing the program button on the back of the remote, all 3 jogged. I then pressed "prog" on the ESPSomfy-RTS app, the 3 shades jogged again. No other shades in the house jogged during this. But, when I try to operate that "shade" in the app I expect the 3 shades to open. Instead, 2 of them open as well as 2 other shades in a nearby room. I've tried rebooting the ESPSomfy-RTS, unpairing the shade, deleting the shade, etc. Each time I re-pair the shade it seems like it paired correctly but then I get the same unwanted behavior.

If it matters, the two shades in the other room are also paired to the ESPSomfy-RTS as individual shades.

rstrouse, 2023-05-10:
If the shades indicated that they moved on ESPSomfy RTS then that might make sense but the addresses for each channel should be unique.  Is there a difference if you use the remote.  Do the other two shades move as well?  Double check the addresses in ESPSomfy RTS between the two devices and the linked remote channels.

bradphillips921, 2023-05-10:
When using the remote I get the expected behavior, the 3 shades I'm trying to control move and the 2 shades in the other room do not. Also, the 2 shades in the other room do not jog when I press the program button on the remote or the "prog" button on the ESPsomfy RTS.

I wiped my ESPSomfy RTS device as part of my range troubleshooting. I'll try it again with a fresh install.

rstrouse, 2023-05-10:
Pay attention to the addresses.  I use the bottom half of the MAC address for the ESP as a starting address.  So far these have been unique for me but there is a chance.  Also, make sure you have updated to the latest release as there have been some new filters for frames sent with 0 addresses.

bradphillips921, 2023-05-14:
I'm still having issues with this. I have a set of 6 roller shades on 1 channel. Every time I try to pair them with the ESPSomfy RTS I get all 6 shades to jog as expected. But when I use the ESPSomfy RTS to control the shades they never work as intended. Sometimes only 5 (of the same) shades work, sometimes it's just two of the same shades, sometimes it's 5 of the 6 shades plus an additional shade in another room that wasn't paired.  I've tried unpairing, deleting shades, trying different Remote Addresses, I wiped the software from the ESP and reinstalled (v. 1.54), I even removed it from my 3D printed mount and created a PCB mount with case exactly like you did.

rstrouse, 2023-05-14:
That is bizarre.  ESPSomfy RTS is sending a single address and never more than one for any command.  It is very odd that it could even cause a shade in another room that doesn't have the address stored to move.  Is it the same shade in another room each time?  To get that to happen the address would have to match, it would have to pass the checksum in the motor, and it would need to match within the window for the rolling code.

The spotty movement could simply be that it just needs a couple more repeats.  ESPSomfy RTS always sends on plus one repeat as you have probably witnessed with the logs open.  I think you set up multiple ESPSomfy RTS devices so maybe we can capture the frames on the device closest to the rogue shade(s) to see if something is getting scrambled in between.

bradphillips921, 2023-05-15:
The first time I had the issue, when I made this post, shade 1 (we will call it) in the other room was moving when it wasn't supposed, and it was always shade 1. After a bunch of troubleshooting, including a complete rebuild of the ESPSomfy RTS device, it's now shade 2 in the other room that moves when it shouldn't.

Yep, I do have two devices. Here are the logs on device 1 when using device 2. I hit the up button on device 2 three times. This should have moved 6 shades up. It only moved 2 of those 6 shades, but no rogue shades moved.

```
[{"encKey": 162,
  "address": 25410,
  "rcode": 18,
  "command": "Up",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10431,7337,2596,2551,2539,2577,4873,1260,1279,1270,1311,621,667,608,661,1271,1264,1310,1273,625,656,662,625,1271,1303,626,644,623,655,1267,1295,632,661,608,652,1297,1257,654,643,626,657,1291,1271,654,610,1316,618,648,1289,1277,1271,1290,1270,633,661,1257,667,627,1284,1296,1277,1258,666,628,633,665,625,644,623,655,1287,644,620,1307,626,644,1272,1291,643,647,633,639,1280,1269,660,641],
  "time": "2023-05-14T21:37:00.896-0500"
},
{"encKey": 162,
  "address": 25410,
  "rcode": 18,
  "command": "Up",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [639,27411,2570,2594,2543,2541,2585,2550,2569,2567,2562,2531,2597,2548,2558,2562,4851,1290,1275,1272,1262,671,645,610,660,1266,1294,1260,1305,625,654,639,629,1320,1259,628,651,666,608,1290,1300,628,634,634,639,1294,1258,652,667,628,633,1290,1276,649,611,1317,617,648,1286,1275,1272,1290,1294,609,661,1266,669,603,1308,1273,1277,1290,645,648,635,638,630,649,644,631,1286,650,642,1283,632,640,1269,1285,636,668,627,633,1288,1271,655,638],
  "time": "2023-05-14T21:37:01.034-0500"
},
{"encKey": 163,
  "address": 25410,
  "rcode": 19,
  "command": "Up",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10446,7349,2596,2547,2553,2563,4834,1285,1290,1280,1265,646,631,661,658,1271,651,610,660,623,1293,634,660,624,641,1275,1285,644,648,1288,615,644,1276,659,625,645,648,1288,1276,647,634,1289,618,669,1259,666,619,1268,661,655,1266,1282,1268,1270,1309,623,669,1253,659,622,1291,1286,1274,1275,665,625,644,623,655,638,635,639,1299,636,628,1298,630,662,1257,1293,634,639,658,645,1271,1260,664,624],
  "time": "2023-05-14T21:37:07.049-0500"
},
{"encKey": 163,
  "address": 25410,
  "rcode": 19,
  "command": "Up",
  "rssi": -19,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [654,27432,2569,2558,2572,2549,2559,2561,2553,2545,2580,2568,2554,2572,2548,2546,4860,1289,1294,1259,1290,647,638,634,655,1271,630,661,641,625,1295,636,635,640,650,1268,1288,651,643,1282,607,665,1269,656,636,633,666,1271,1261,647,645,1305,627,645,1272,614,665,1274,646,637,1288,1269,1282,1266,26,1319,615,659,1274,647,637,1262,1297,1283,1265,666,629,634,633,665,645,626,633,1285,643,625,1311,624,644,1275,1291,645,622,655,638,1278,1295,636,628],
  "time": "2023-05-14T21:37:07.186-0500"
},
{"encKey": 164,
  "address": 25410,
  "rcode": 20,
  "command": "Up",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10433,7355,2592,2546,2541,2588,4858,1266,1273,1277,1308,622,646,1289,1264,650,635,1290,1294,609,661,657,619,1294,635,640,622,668,1277,1266,1270,657,659,625,640,1275,660,625,645,623,1305,1274,1273,665,626,1270,653,639,1280,1269,1286,1300,1279,633,626,1294,633,633,1315,1254,1286,1269,656,639,634,647,644,653,611,667,1252,668,608,1311,616,670,1259,1291,643,631,633,658,1291,1254,659,623],
  "time": "2023-05-14T21:37:08.450-0500"
},
{"encKey": 164,
  "address": 25410,
  "rcode": 20,
  "command": "Up",
  "rssi": -20,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [644,27447,2550,2569,2566,2570,2535,2572,2571,2553,2563,2562,2545,2576,2557,2540,4873,1268,1298,1273,1287,644,622,1282,1307,620,653,1265,1290,646,658,608,665,1268,662,606,654,619,1318,1265,1292,633,635,640,621,1295,634,665,615,646,1295,1261,1316,608,660,1281,642,632,1283,1276,1301,1259,1291,633,635,1289,625,658,1286,1270,1307,1264,619,650,661,615,652,643,645,636,1285,618,670,1261,666,619,1295,1292,619,643,655,613,1286,1295,636,635],
  "time": "2023-05-14T21:37:08.593-0500"
}]
```

Even odder, of the 6 shades that I'm trying to control, only 2 of the shades worked when I hit the button all 3 times. And those 2 shades aren't on the same remote channel, as in I would not be able to make only those 2 shades move with the remote.

rstrouse, 2023-05-15:
The receiving pulse for the up command is exactly what I would have hoped for.  Each frame has exactly 1 repeat and the pulse widths are consistent to the symbol length.  The receiving ESPSomfy RTS device heard this with great clarity (rssi in the -20s).

How this works is that every motor is listening for the frame and every motor within range will receive it.  It there are 6 or 600 it will not matter because the communication is only one way.  This has nothing to do with the number of shades.  If the shade acknowledged and paired with that 25410 address it should have responded when it heard the frames above.

If you have a shade that moves when it hears the address then I am certain that it has this address in its memory.  The act of un-pairing should remove it.  However, this is where it can be confusing.  Un-pairing and pairing are the same command.  If you attempt to un-pair a remote that is not paired it will instead pair it.  If you attempt to pair a remote and it is already paired it will un-pair it.  There really is no way of knowing.   Maybe that is why the instructions say to remove power from all other shades and pair one at a time.

bradphillips921, 2023-05-16:
Thanks for the response. That helps me understand the mechanics of it a little more. I'll go through all the addresses and make sure the shades are unpaired then start from scratch.

If I'm going to use multiple ESPSomfy devices around my house should I make sure they each use a different range of remote addresses? For example, if I use remote address 25410 on ESPsomfy 1 and the same remote address 25410 ESPsomfy 2 will there be an issue? Should I instead reserve remote addresses 25350-25370 for ESPSomfy 1 and 25371-25390 for ESPsomfy 2, etc? Is there a range of remote addresses I need to stay in?

rstrouse, 2023-05-16:
There is a factory reset procedure for the shades that is supposed to clear the memory.  ESPSomfy RTS uses the lower 18 bits of the MAC address as the starting address and adds 1 for each new shade added.  Theoretically, this should be unique but there is a chance that something might overlap.  You will find that your remotes do the something similar.   Some thing to remember is that the shade will jog to pair it and un-pair it and the commands to do it are identical.  

To see whether the jog paired the shade or unpaired it try moving the shade.  If it moves then it was paired.  If it doesn't then it un-paired.

bradphillips921, 2023-05-18:
I've been playing with this for a couple of days with no luck.  I made sure to unpair all the shades that I had on the remote address 25326 then I attempted to repair. Each time 3 shades would jog when I put them in pairing mode from the button on the back of the remote, 3 shades would jog when I hit "Prog" on the ESP Somfy RTS but only 2 shades would move when being controlled by the ESP Somfy RTS.  These shades are 19 feet off the ground, so resetting them is not a great option.

I tried updating to 1.6.0 but the update hangs. I've tried 4 times and it always hangs at a different percentage. Not sure if v 1.6.0 would solve my problem.

rstrouse, 2023-05-18:
That is so very odd.  If the remote moves them then ESPSomfy RTS sending the exact same pulses should move them.  Can I get you to capture the remote signal where it actually moves the shade in question.

bradphillips921, 2023-05-19:
I was able to access one of the shades that was causing an issue and reset it. That appeared to work. It appears that the Somfy roller shades can hold 12 remote addresses. I had apparently filled those 12 slots up in that one shade from all my experimenting with different devices to control the shades. Even though the shade jogged when pairing, it must not have stored the remote address since they were all full.

I need to buy a ladder to access my high shades, or rig up some kind of pole with a tiny needle on the end to hit the reset button.

rstrouse, 2023-05-19:
Yeah it isn't helpful that is actually jogs and there is no remote way to reset it.  Mine are big rollers that are hardwired.  To reset them you have to do the power cycle thingy and that requires taking them down to just do one.  Not a simple operation.  I have had to do it twice while I was trying out different addresses until I figured out that it eventually stops accepting new addresses and the code was right all along.

At 20 feet a long pole might be like docking on the ISS.  If you have a pool, pool poles aren't very flexible so it might be easier than rigging multiple sticks like you were competing on Survivor.

rstrouse, 2023-05-27:
Did you get your shades reset?

bradphillips921, 2023-05-27:
Yep. I made a resetting contraption using an extendable painters pole, brad nail, 3D printed mount, and a pen scope in order to get to the reset buttons 20ft up. It worked!
[pic 1
](https://drive.google.com/file/d/1OyDVvGapCn_AEDeZ212oSTms8MWRtrRd/view?usp=sharing)
[Pic 2](https://drive.google.com/file/d/1zTGBpTq2myWsvEnhFDyd1zv0VzLBb5ly/view?usp=sharing)

I still have some issues with rogue shades being paired to a channel. But I think I can get around that by removing their power prior to programming.

Thanks!

rstrouse, 2023-05-27:
Well I wouldn't fly it to the moon or do remote surgery with it, but that looks like a fine tool for resetting a shade.  I have determined that if the shade reset requires effort to get done, just staring at it for a long time doesn't make it happen.  I had to pull mine down when I filled up the memory.

Glad you got it working.  Somfy recommends removing power to other shades when pairing remotes.  Not sure what the conditions are but it appears that the memory is open for some period after power is applied and after a reset.

rstrouse, 2023-06-01:
I am closing this for now.  If you have any other issues please submit another issue.