rstrouse/ESPSomfy-RTS #47: Update from phone

Motrail, 2023-05-14:
When updating from my phone (iPhone either chrome or safari), the firmware loads fine but the application tries to download the backup which results in a network error. Is it possible to separate download of the backup and upload of the application file so upgrades from phones are possible?

PS. Really great job on this device!

rstrouse, 2023-05-14:
I guess I never tried to update from a phone.  I'll take a look and perhaps I can simply disable the download for these devices.

Velocet, 2023-05-19:
Maybe have a look at a totally different solution?
> [OTA Update via GitHub Releases for ESP32](https://github.com/Fishwaldo/esp_ghota)

rstrouse, 2023-05-20:
Yeah that is a whole lot of overhead.  I looked at that in the past but maybe I will branch to see I can get it to coexist.

rstrouse, 2023-06-14:
What error are you getting?  I just updated the firmware and application on an iPhone and it succeeded.  It even downloaded a backup file to my phone.

Motrail, 2023-06-14:
After it opens the download dialogue, it states  "500: Network HTTP Error" and does not continue with the update.

rstrouse, 2023-06-14:
That is very odd.  What happens when you click the backup button.  Does it download a file?

rstrouse, 2023-06-17:
@Motrail have you reserved your ip address and hostname in your router.  Sometimes this helps as the mDNS on ESP32 is not very good.

Motrail, 2023-06-19:
> @Motrail have you reserved your ip address and hostname in your router. Sometimes this helps as the mDNS on ESP32 is not very good.

Yeah I have reserved it. When I click the button, it asks to download the file, afterwards I get the 500 error. When I click 'close' it shows the upgrade screen but the progress bar doesnt move.

[image]
[image]
[image]

rstrouse, 2023-06-19:
v1.7.2 has been released.  This does browser detection on the client and omits the backup process prior to uploading the file.  Interestingly I have determined that there is no request made to the server as if the mobile browser leaves the ESPSomfy page during the backup process momentarily.

You will still need to update 1.7.2 from a non-iphone browser but from here on out you will be able to update with your phone.