rstrouse/ESPSomfy-RTS #48: The shade does not jog

ok11, 2023-05-14:
Hi, first of all really an outstanding job you have done!

I have assembled and configured all per your tutorial, all worked great, just the last step, pairing, is not working as I expected.

The remote (Smoove 1) sends the prog command, the shade jogs, all as it should be. But the "Prog" button in the app, "Pairing" screen does not cause any reaction. And, consequently, the commands in the app do not work, too.

I double checked the wiring and the fact that the transciever receives as expected (the events from  the remote are logged).

Is there any way to see what is happening on the controller-transmitter side? Do commands go out at all?

Thanks!

ok11, 2023-05-14:
Hm. Do I get it right that `server.on("/pairShade", []()` is commented out in the web server?

rstrouse, 2023-05-14:
Yeah that should be commented out.  It has been deprecated for removal when I get around to refactoring.

Change the settings for RX and TX in the software.  Change the RX setting to the TX setting and the TX setting to the RX.  You likely have GDO0 attached to the RX.

ok11, 2023-05-14:
Indeed!!! 😃 thanks man, closing the issue

closed 2023-05-14 by ok11