rstrouse/ESPSomfy-RTS #49: Timezones

ok11, 2023-05-14:
Hey! There are timezones not considered in the index.js, e.g. "Europe/Berlin" (that's where I am ;)). Is definitely not a pressing issue though. I could add some and open a PR if you want, would take less important work from you 😀 

Thanks!

rstrouse, 2023-05-14:
Is the posix code for Europe/Berlin `CET-1CEST,M3.5.0,M10.5.0/3`?

ok11, 2023-05-14:
CEST-1CET,M3.2.0/2:00:00,M11.1.0/2:00:00

rstrouse, 2023-05-14:
Thanks I added it to the tz list and will come in the next release that I will post this week.

rstrouse, 2023-05-16:
v1.6.0 now supports the Europe/Berlin timezone.

cesarvoggithub, 2023-05-19:
Could you please also add America/Sao_Paulo (Etc/GMT-3)?

TIA

rstrouse, 2023-05-19:
It will show up on the next release.  Btw the posix code for the timezone is `<-03>3`

cesarvoggithub, 2023-05-19:
Many Thanks!
Have a nice weekend.

Cesar

rstrouse, 2023-05-23:
Update to v1.6.1 I added the Sao Paulo timezone.

cesarvoggithub, 2023-05-23:
Thanks for the update. I'm afraid the Application is still 1.6.0. I first tried updating by uploading the firmware first (worked as the frmware "took" and changed to 1.6.1) and then the application (did not "take" as it remained 1.6.0). Then I decided I would start from scratch by reloading the original ESPhome firmware (worked as everything went back to default) and then by installing the newonboard file (worked, but firmware showed 1.6.1 and application remained 1.6.0). And since the application did not "take", America/Sao_Paulo is still not an available choice in the dropdown list.

cesarvoggithub, 2023-05-23:
Scratch the above. I tried re-downloading the SomfyController.littlefs.bin file and this time it worked!

rstrouse, 2023-05-23:
Unfortunately you probably caught it before I changed the release token.  A hard refresh of the browser would have fixed the issue.  I am doing some browser tricks to clear the cache on for the app and unfortunately I missed one spot on the release files earlier.

pfak, 2023-05-24:
`America/Vancouver` is also missing.

rstrouse, 2023-05-24:
`America/Vancouver` is the same timezone alias as `America/Los_Angeles`.

getselim-stack, 2023-05-26:
Can you please add Asia/Dubai UTC+4 (no daylight saving)?

rstrouse, 2023-05-26:
That is the same timezone as `Asia/Baku`  posix: `<+4>-4`.  The reason I am not duplicating the them is that this is one of the more costly string arrays in terms of storage and transfer in the software.  However, please keep them coming and I will let you know if there is a setting already in the same ntp timezone.  

Unfortunately, I am not smart enough to gather which duplicates are the more logical selection.

getselim-stack, 2023-05-26:
> OR (preferably) where there is overlap, i.e. US and Canada, you could do "Pacific", "Central", "Eastern", etc.

Good idea. Generic time zones would be a good approach as I appreciate the need to keep it as light as possible. Happy to help define for this region if needed.

rstrouse, 2023-05-26:
Yeah the entire posix database is around 500 entries.  To make sure we get valid log times it seemed pretty expensive to use the whole thing.  Below is a link to one of the myriad of databases for them.  I probably should have started with the Etc/GMT+- and eliminated those first.  I am keying on the code (right hand column) and not the city for that very reason.

You may be surprised at how weird all of this DST stuff is when trying to keep track of the 4th dimension.  I'll take a stab at reworking tags to generalize them.

https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

> Not willing to pay the $230 Somfy want for their bridge for something that contains $20 worth of hardware!

To add to that I don't want any more cloud stuff in my house.  10 years from now folks are going to look around at their home automation battlefield to see many soldiers that worked for a couple of years but couldn't keep up the fight.  Imagine the carnage and blown out rusted relics littering the scene.

rstrouse, 2023-06-16:
The new time zone data will be posted in v1.7.1 in the next couple of days.

rstrouse, 2023-06-17:
v1.7.1 has been released with the timezone changes discussed.

closed 2023-06-17 by rstrouse