rstrouse/ESPSomfy-RTS #50: Shade pairing doesn't work

skupu, 2023-05-16:
Hi @rstrouse,

I have problem with pairing the Somfy Altus RTS shades - I can see the commands in log, so the CC1101 works fine, I can link the remote, but unfortunately the pairing procedure gives no result.

[image]
[image]

Please support.

rstrouse, 2023-05-16:
Change TX to GPIO-13 and RX to GPIO-12 then hit save.  Do not rewire just change the two values in the screen above.  GDO0 must be wired to TX.  You can receive on either pin but can only send on GDO0.

skupu, 2023-05-16:
Thanks. I replaced ESP to wroom (before I used ESPDUINO and none of the pins were working - I tried some of them).

Now the pairing works, but the shades are not moving...

Any idea?

rstrouse, 2023-05-16:
Somfy pairing can be really dumb.  You may be trying to pair and the shade mystically jogs but what it is actually doing is unpairing.  The other thing is that motors only have a certain number of slots so it will jog but not store the address.  Somfy recommends resetting to factory settings.

skupu, 2023-05-16:
I just did with one of the 5 shades I have. Now I only have one telis paired, cannot pair the other that worked (somehow it went unpaired) and I don't know why still the ESPSomfy doesn't work

skupu, 2023-05-16:
I tried with another shade and hapily it started working - both remotes are ok and the RTSomfy works well.
So I will try with the rest and post results. Possibly I will make another factory reset of the not working one tomorrow afternoon.

getselim-stack, 2023-05-26:
> Change TX to GPIO-13 and RX to GPIO-12 then hit save. Do not rewire just change the two values in the screen above. GDO0 must be wired to TX. You can receive on either pin but can only send on GDO0.

I made the same mistake - thanks!

rstrouse, 2023-06-01:
I am closing this as I assume everybody has everything working whether it is related to ESPSomfy RTS or not.  If you need further assistance open another issue.

closed 2023-06-01 by rstrouse