rstrouse/ESPSomfy-RTS #51: Unable to receive data, logs empty

TranceSetter, 2023-05-17:
Hi,

I've been following your project for quite some time and have started a few weeks ago to get the hardware to be able to control my sunshades. I have some difficulty getting this to work. I've retraced my steps from the beginning of the hardware setup at least 20 times now but I just cannot see the issue with it.
Of course I checked the other issue-items and checked many of the provided tips, tricks and solutions:

- My ESP32 is a "DOIT ESP32 DEVKIT V1"
- The transceiver module is a "E07-M1101D-SMA" as per the hardware guide of preferred modules
- Connections made using Dupont-cables, verified by a multimeter on the board pins
- Connections as per suggested way as described on the hardware setup wiki
- Antenna is 433MHz and has a male center pin
- The Somfy remote is a "74300 - TELIS 4 - RTS PURE", no IO version

The firmware is installed without problems. The configuration works exactly as described in the wiki. I can enable the radio and the "RADIO NOT INITIALIZED" message disappears. When I set any of the pins incorrectly it does come back, so it tells me there is communication with the transceiver module(?). But the logs screen does not show any received messages.

Additional things I checked:
- Tested with another ESP32, same result
- Ordered 2nd transceiver module, same result
- Swapped the TX and RX pins in the software, same result
- Swapped the TX and RX pins on the ESP32, same result

I'm not sure what I can do to get this working but hope you still have some thoughts and why it might not work.

[image]
[image]
[image]
[image]
[image]

rstrouse, 2023-05-17:
Very thorough work on the troubleshooting.  The radio not initialized message going away means the SPI communication with the module is working.  Your wiring looks fine.

SCK = yellow = GPIO-18
CSN = orange = GPIO-5
MOSI = green = GPIO-23
MISO = blue = GPIO-19
TX = brown = GPIO-13
RX = purple = GPIO-12

I am not seeing a Network SSID in your screen.  What happened to it?  Is this connected to your wifi network?

TranceSetter, 2023-05-18:
I've updated to v1.6.0 and moved the RX Bandwidth to 250KHz and that worked. Every button press was picked up perfectly and with an average of -35dB. Then I gradually decreased the bandwidth and it kept working. Now the bandwidth is at 97.39KHz and it still works as expected.

After that I added all my shades and can control them perfectly. As an addition I added the Home Assistant integration as well and am now able to automate everything. Great stuff!

The only thing I need to figure out why 1 of my 2 blinds which are next to each other goes faster than the other. They now do not have the same position when moving them to 15% for example. That is however not for this issue, which I will close with a perfect solution.

closed 2023-05-18 by TranceSetter

steveomg, 2023-07-20:
I am having this same issue.  I have verified wiring dozens of times.  I have tried multiple hardware units.  I have tried sliding the Rx bandwidth to 250.  I am unable to get anything to print in the logs.  

My remote is a somfy Telis 4 RTS US Patio.  White with blue edges.  Could the frequency be different?