rstrouse/ESPSomfy-RTS #52: A Valid Shade ID not Supplied

nobbyneb, 2023-05-21:
Strangely, a number of additional shades were created in addition to the original Shade i had setup (Garage at the top). Not sure how this happened. On trying to delete, i get the error 'A Avalid Shade ID not Supplied'. I have tried editing the name and parameters of these but continue to get the same error. Thanks!

[image]
[image]

rstrouse, 2023-05-21:
Go to the updates tab and create a backup.  Then post that here.

nobbyneb, 2023-05-21:
Thank you, attached.

[ESPSomfyRTS 2023-05-21T20_26_22.backup.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/11525364/ESPSomfyRTS.2023-05-21T20_26_22.backup.zip)

rstrouse, 2023-05-21:
I do not understand what went on with that but it looks like it wrote a corrupted configuration.  Anyhow if you only had one shade configured, under the gears option go to updates and restore this file after you extract it.
[ESPSomfyRTS 2023-05-21T20_26_22.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/11525425/ESPSomfyRTS.2023-05-21T20_26_22.zip)

Any insight into what you were doing when the corruption occurred?

nobbyneb, 2023-05-22:
Very kind of you, thank you - the backup worked perfectly.

It was very strange. I was fumbling around with the loose ESP and Radio module (must have been while they were still powered). The corrupted config then appeared when i next logged in, i can only think i shorted or misconnected some pins somehow. Really appreciate the help.

rstrouse, 2023-05-22:
That would do it.  Let me know if you need anything else.

closed 2023-05-22 by rstrouse