rstrouse/ESPSomfy-RTS #53: Link Group to Other Shades

peterchs, 2023-05-24:
I have 3 blinds set up in ESPSomfy controlling solo/individually.  
I have an additional group shade in ESPSomfy to control all 3 blinds together (so they all move perfectly together on 1 transmission).

Its all working fine however if I control with the grouped blind it does not update the other solo shades so the blind location gets out of sync.

What I need to do is be able to link a remote/id between shades in ESP Somfy - i.e. link shade 4 (group one) to all 3 of the other shades so the blind location % is correct.  Is there a way I'm missing to do this?

(Thanks for great project by the way, very well implemented and write up/instructions are awesome)

rstrouse, 2023-05-24:
I have been thinking about ways to group the shades and even kicking around the idea you described above.  However, I think that this needs to be something more in that a group all by itself cannot reliable have a position.  This is because many times the shade lengths and transition times are wildly different.

ESPSomfy RTS is handling the other direction of this currently very well, where the shades are grouped by a remote channel.  This is done by pairing a group of shades to a specific channel on the remote then linking that channel with each shade in ESPSomfy RTS.  Mashing the button on the remote with the grouped channel results in each of the shades maintaining proper position.

Ultimately, this probably needs to be a different shade type that simply has no position of its own and can trigger a remote channel.  Unfortunately, that probably won't work without pairing each of the shades like the remote does because issuing the frame would make the remote go out of sync.  I'll have to think about what that would look like from a UI perspective.

bradphillips921, 2023-06-05:
Just putting another vote in for this.

My office has three shades, I have a channel on the remote for each individually plus 1 channel to operate all 3 at once.  I linked the "group" channel on the remote to each individual shade in ESPSomfy RTS but if I send a command via ESPSomfy RTS to open all the shades using the "group" channel then the position of the individual shades in ESPSomfy RTS does not update.

rstrouse, 2023-06-05:
Yep that is next on the list.  I’ll need to figure out how to make it all work with HA.

011V32, 2023-06-06:
Wouldn't it be possible to make an own shutter with an own remote address and then add that remote address to the other shutters as a linked address?
(Of course, this assumes that the commands sent also carry out an action like when a command is received...)

rstrouse, 2023-06-06:
Yes but you can only do that by first pairing the new shade to all the required motors then linking them in a backup file then uploading it.

rstrouse, 2023-07-03:
This functionality is complete for ESPSomfy RTS but I have not yet published a release for it.  I need to finish off the Home Assistant component to support groups but you can update your firmware and application from the main repository to get the grouping functions.  The groups just won't show up in Home Assistant until I release that component.  The UI had to change so I could support all the extra data but you can find a reference here
https://github.com/rstrouse/ESPSomfy-RTS/wiki/User-Interface-Reference

The process for grouping can be found in this wiki.
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software

rstrouse, 2023-07-08:
v2.0.0 has functionality for groups.  Thanks for the suggestion!

closed 2023-07-08 by rstrouse