rstrouse/ESPSomfy-RTS #54: Add "Awning" type and support for sun/wind sensors

Noltari, 2023-05-28:
First of all many thanks for this wonderful project :D
I was using RFLink and wanted to switch to something better ^^ and since I had a CC1101 lying around...

I have 4 sun awnings on my windows from [Arqualuz](https://arqualuz.com/) (Spanish brand, Somfy RTW compatible) paired with a sun/wind sensor. I'm not sure about its brand, but it looks like a [Soliris RTS](https://www.somfy.es/productos/1818211/soliris-rts) or some pretty similar clone.

The remote can switch the awnings from  wind mode to sun/wind and viceversa:
- Wind mode: the awnings will go up if there's a lot of wind to prevent any kind of damage.
- Sun/Wind mode: Wind mode + the awnings will go down if there's enough sun and will go up if there isn't enough sun.

It would be great to be able to detect the awning modes and to be able to change their mode from ESPSomfy-RTS.

<details>
<summary>For the record, this is what I can see in the logs when going from wind to sun/wind mode:</summary>

```
[{"encKey": 141,
  "address": 9347787,
  "rcode": 355,
  "command": "Unknown(15)",
  "rssi": -44,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [778,1202,3278,50,5162,10926,2490,2552,2462,2551,4813,1295,1272,667,619,673,627,1289,648,650,100,175,1226,1376,1222,1346,596,660,630,660,623,674,1253,1299,1272,666,595,1403,529,723,596,679,605,685,1235,1329,609,671,1272,643,643,644,647,1312,51,78,302,652,50,275,326,576,1356,583,698,596,681,1257,1323,601,686,1255,1300,1286,644,101,152,609,1382,103,153,280,306,556,696,597,709,1207,696,598,1335,1252,1307,631,646,1297,637,657,621,644,659,607,734,549,724,576,686,603,696,596],
  "time": "2023-05-28T20:48:36.480+0200"
},
{"encKey": 141,
  "address": 8299211,
  "rcode": 364,
  "command": "Unknown(15)",
  "rssi": -44,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [2503,2538,2476,2537,2462,2589,2446,2562,2473,2543,2470,2539,2466,2560,4798,1301,1261,673,356,406,607,758,530,1359,572,707,1232,1322,1256,1324,623,658,633,643,622,679,1231,1371,1228,694,596,1306,630,672,623,656,631,654,1275,1283,658,620,1296,666,596,723,100,150,275,326,551,1370,1221,1344,597,673,628,661,1261,1301,633,663,1273,1285,1284,646,599,1395,545,705,605,710,1201,703,605,1307,1271,1312,625,647,1276,649,646,653,629,662,103,155,607,721,547,733,581,698,595,679,606],
  "time": "2023-05-28T20:48:36.893+0200"
},
{"encKey": 141,
  "address": 8299211,
  "rcode": 364,
  "command": "Unknown(15)",
  "rssi": -46,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27488,2496,2526,26,2490,2549,2471,2560,2477,2531,2469,2565,2473,2538,2500,2516,125,178,4815,1325,1257,645,621,688,604,1323,623,662,1281,1271,126,152,1258,1347,602,685,607,672,621,682,1235,1321,1281,648,621,1307,632,646,644,633,657,637,1278,1307,609,698,1249,685,605,671,621,1307,1269,1305,635,647,649,628,1283,1299,649,634,1260,1351,1232,673,620,1310,628,670,622,660,1260,645,647,1287,1301,1282,608,722,1223,693,595,697,600,667,620,670,623,670,605,670,620,659,634],
  "time": "2023-05-28T20:48:36.899+0200"
},
{"encKey": 141,
  "address": 8299211,
  "rcode": 364,
  "command": "Unknown(15)",
  "rssi": -50,
  "bits": 56,
  "valid": false,
  "sync": 6,
  "pulses": [27515,2481,2551,2471,2558,74,10063,2451,2565,2450,2555,2504,2498,4841,1264,1315,626,652,631,656,1269,680,618,1302,1280,1287,1297,630,661,628,670,624,662,1283,1272,1307,619,669,1261,657,643,646,636,660,618,1320,1266,667,631,1284,674,617,658,632,1287,1296,1269,670,630,654,613,1306,1279,660,622,1319,1266,1293,659,632,1299,623,667,619,644,1297,642,642,1279,1311,1272,657,613,1331,629,628,661,635,644,645,660,633,646,643,632,655,637,652],
  "time": "2023-05-28T20:48:37.385+0200"
}]
```
</details>

<details>
<summary>And this is what I can see in the logs when going from sun/wind to wind mode:</summary>

```
[{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -47,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [10453,10890,2512,2523,2488,2520,4822,1288,1297,636,659,623,670,1257,660,646,634,678,1255,671,620,1309,633,641,645,636,633,649,646,653,1281,1296,1275,1293,619,684,607,673,620,680,606,661,1277,654,654,638,651,620,648,641,657,1269,1306,1272,672,640,621,671,1249,1335,621,656,1258,1296,1302,623,670,1281,634,646,644,633,1309,643,618,1336,1252,1306,633,646,1270,667,619,670,626,638,657,644,645,636,664,643,642,631,630],
  "time": "2023-05-28T20:46:33.859+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -52,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27493,2521,2514,2498,2540,2494,2541,2473,2535,2495,2519,2491,2541,2496,2518,4828,1283,1283,648,651,642,643,1297,632,643,646,630,1304,649,642,1285,634,671,621,658,633,660,626,648,1275,1300,1296,1267,644,634,659,649,642,628,678,613,1308,622,655,660,604,670,646,660,607,1298,1297,1294,645,634,633,663,1274,1284,657,643,1270,1319,1272,656,618,1296,655,641,644,650,1281,621,667,1283,1259,1323,632,669,1249,657,633,673,619,654,632,651,643,649,648,641,643,645,649],
  "time": "2023-05-28T20:46:34.195+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -51,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27460,2515,2512,2502,2535,2492,2545,2473,2547,2475,2550,2482,2525,2509,2522,4814,1282,1292,658,634,647,645,1281,633,646,644,634,1309,617,669,1289,634,670,597,684,607,669,648,653,1260,1293,1303,1272,646,636,659,647,644,632,656,639,1279,650,631,679,595,673,626,665,620,1296,1284,1295,651,641,643,646,1276,1300,644,637,1311,1269,1257,698,595,1333,605,669,621,658,1285,643,646,1287,1278,1280,660,644,1293,641,615,697,599,670,634,645,646,659,635,647,621,655,655],
  "time": "2023-05-28T20:46:34.197+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -48,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27473,2490,2533,2496,2523,2525,2518,2490,2539,2472,2543,2503,2523,2482,2522,4842,1288,1274,634,659,635,650,1274,658,651,615,678,1256,673,621,1307,632,670,620,658,633,652,645,626,1303,1293,1275,1292,621,693,605,670,621,683,608,661,1274,658,633,649,644,628,653,655,643,1271,1308,1269,676,620,632,672,1274,1295,620,683,1262,1296,1282,645,643,1285,656,645,644,637,1311,618,644,1318,1246,1320,624,650,1275,668,643,631,655,642,644,627,654,636,657,644,644,661,608],
  "time": "2023-05-28T20:46:34.279+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -50,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27633,2514,2512,2491,2537,2490,2556,2471,2535,2520,2535,2473,2542,2469,2545,4827,1286,1264,676,626,661,633,1297,627,663,630,645,1297,636,654,1270,657,642,649,620,644,693,596,676,1270,1295,1271,1293,645,658,633,649,636,650,649,642,1293,657,604,710,583,673,622,668,620,1298,1281,1296,649,643,629,645,1297,1293,645,617,1310,1272,1288,696,596,1321,609,671,623,661,1285,649,648,1263,1312,1283,633,646,1294,641,640,674,600,668,621,673,623,657,632,672,619,659,633],
  "time": "2023-05-28T20:46:34.502+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -47,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27490,2492,2540,2499,2539,2501,2514,2488,2545,2502,2522,2509,2520,2487,2521,4840,1294,1296,616,668,625,651,1293,641,634,631,693,1245,670,633,1296,649,642,645,647,622,667,645,622,1298,1292,1295,1267,643,696,580,695,620,657,631,661,1276,632,656,651,643,626,650,642,660,1270,1309,1270,650,665,598,697,1249,1321,621,669,1257,1299,1284,645,645,1268,668,627,652,638,1312,625,150,427,626,1355,25,1146,75,1418,596,676,1276,646,646,655,631,637,650,647,648,617,669,620,673,634,130,583],
  "time": "2023-05-28T20:46:34.556+0200"
},
{"encKey": 142,
  "address": 8305963,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -62,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27554,2493,2548,2485,2545,2484,2522,2513,2523,2496,2531,2503,2536,2479,2537,4821,1297,1280,646,645,633,655,1289,645,633,656,638,1280,653,627,1321,619,657,634,668,597,703,602,661,1279,1305,1284,1273,655,651,649,625,650,641,646,672,1250,695,598,677,603,670,618,650,653,1291,1296,1286,615,702,602,682,1229,1331,607,672,1272,1312,1277,658,632,1287,621,681,609,710,1228,1209,50,75,1410,1247,1337,599,674,1251,668,645,631,657,637,652,647,623,658,633,643,648,631,658],
  "time": "2023-05-28T20:46:34.698+0200"
}]
```
</details>

rstrouse, 2023-05-29:
What do you see in the logs when the following happens.
1. You turn the sun and wind off
2. The wind or sun sensor triggers the awning to close
3. The sun sensor triggers the awning to open.

Noltari, 2023-05-29:
> What do you see in the logs when the following happens.
> 
> 1. You turn the sun and wind off

<details>
<summary>This is what I provided on the first post because only the sun mode can be disabled, since the wind mode is always enabled for protection.</summary>

```
[{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -47,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [10453,10890,2512,2523,2488,2520,4822,1288,1297,636,659,623,670,1257,660,646,634,678,1255,671,620,1309,633,641,645,636,633,649,646,653,1281,1296,1275,1293,619,684,607,673,620,680,606,661,1277,654,654,638,651,620,648,641,657,1269,1306,1272,672,640,621,671,1249,1335,621,656,1258,1296,1302,623,670,1281,634,646,644,633,1309,643,618,1336,1252,1306,633,646,1270,667,619,670,626,638,657,644,645,636,664,643,642,631,630],
  "time": "2023-05-28T20:46:33.859+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -52,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27493,2521,2514,2498,2540,2494,2541,2473,2535,2495,2519,2491,2541,2496,2518,4828,1283,1283,648,651,642,643,1297,632,643,646,630,1304,649,642,1285,634,671,621,658,633,660,626,648,1275,1300,1296,1267,644,634,659,649,642,628,678,613,1308,622,655,660,604,670,646,660,607,1298,1297,1294,645,634,633,663,1274,1284,657,643,1270,1319,1272,656,618,1296,655,641,644,650,1281,621,667,1283,1259,1323,632,669,1249,657,633,673,619,654,632,651,643,649,648,641,643,645,649],
  "time": "2023-05-28T20:46:34.195+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -51,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27460,2515,2512,2502,2535,2492,2545,2473,2547,2475,2550,2482,2525,2509,2522,4814,1282,1292,658,634,647,645,1281,633,646,644,634,1309,617,669,1289,634,670,597,684,607,669,648,653,1260,1293,1303,1272,646,636,659,647,644,632,656,639,1279,650,631,679,595,673,626,665,620,1296,1284,1295,651,641,643,646,1276,1300,644,637,1311,1269,1257,698,595,1333,605,669,621,658,1285,643,646,1287,1278,1280,660,644,1293,641,615,697,599,670,634,645,646,659,635,647,621,655,655],
  "time": "2023-05-28T20:46:34.197+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -48,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27473,2490,2533,2496,2523,2525,2518,2490,2539,2472,2543,2503,2523,2482,2522,4842,1288,1274,634,659,635,650,1274,658,651,615,678,1256,673,621,1307,632,670,620,658,633,652,645,626,1303,1293,1275,1292,621,693,605,670,621,683,608,661,1274,658,633,649,644,628,653,655,643,1271,1308,1269,676,620,632,672,1274,1295,620,683,1262,1296,1282,645,643,1285,656,645,644,637,1311,618,644,1318,1246,1320,624,650,1275,668,643,631,655,642,644,627,654,636,657,644,644,661,608],
  "time": "2023-05-28T20:46:34.279+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -50,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [27633,2514,2512,2491,2537,2490,2556,2471,2535,2520,2535,2473,2542,2469,2545,4827,1286,1264,676,626,661,633,1297,627,663,630,645,1297,636,654,1270,657,642,649,620,644,693,596,676,1270,1295,1271,1293,645,658,633,649,636,650,649,642,1293,657,604,710,583,673,622,668,620,1298,1281,1296,649,643,629,645,1297,1293,645,617,1310,1272,1288,696,596,1321,609,671,623,661,1285,649,648,1263,1312,1283,633,646,1294,641,640,674,600,668,621,673,623,657,632,672,619,659,633],
  "time": "2023-05-28T20:46:34.502+0200"
},
{"encKey": 142,
  "address": 8299211,
  "rcode": 363,
  "command": "Unknown(15)",
  "rssi": -47,
  "bits": 56,
  "valid": false,
[... 109 lines cut ...]
  "pulses": [650,16748,9940,2459,2518,2477,2534,2526,2459,2682,25,100,2304,2457,2524,2466,2528,2460,2516,4784,1265,1284,1249,1296,1263,1298,629,635,1291,1274,1284,1285,592,674,1263,655,645,1272,1265,1299,1253,1291,638,646,1280,633,648,1270,1287,1278,1273,1285,620,644,1285,634,623,1294,1266,663,621,643,631,663,630,1265,1280,638,639,1298,625,635,1292,1272,1280,1266,1290,1285,624,643,1273,1312,594,645,1286,640,623,25,1295,637,634,1277,645,636],
  "time": "2023-05-29T10:27:01.888+0200"
},
{"encKey": 169,
  "address": 10569114,
  "rcode": 0,
  "command": "Unknown(14)",
  "rssi": -79,
  "bits": 56,
  "valid": false,
  "sync": 15,
  "pulses": [637,7862,9408,1641,4570,3217,2462,2522,2467,2519,2449,2541,2446,2542,2471,2506,2473,2526,2453,2523,4769,1280,1270,1284,1284,1272,1285,623,642,1278,1286,1272,1278,632,644,1274,659,633,1277,1277,1283,1270,1279,613,665,1271,654,634,1284,1273,1282,1278,1271,636,632,1273,650,636,1290,1270,652,613,660,625,721,50,124,580,1259,1292,644,628,1284,625,645,1276,1289,1271,1278,1265,1296,632,635,1291,1296,613,635,1279,645,633,1282,648,618,1280,661,613],
  "time": "2023-05-29T10:27:01.895+0200"
}]
```
</details>

> 3. The sun sensor triggers the awning to open.

It's rainy these days, so I will provide this as soon as the weather helps :(

BTW, from what I can see in the logs, the sun and wind sensor broadcasts if the awnings should go up or down from time to time, so If I put the awnings in sun/wind mode and then I command down an awning, it will take some time before the awning will go up (this is not instant).
Also I think that we're gonna need something like https://github.com/rstrouse/ESPSomfy-RTS/issues/53 because there are 5 channels in the remote, one per awning and the group channel that controls the 4 awnings at the same time and where the sun/wind sensor is linked.

Thanks!

rstrouse, 2023-05-29:
Interesting.  This looks like the first two sets of commands on the first post are definitely the RTW protocol.  Just to be sure I assume these are all from the remote on two different channel addresses `9347787` and `8299211`... the last repeat on address `8305963` looks like a collision so that isn't a valid frame.

On the emits from the Soliris this is the RTS protocol and it looks like it is on address `10569114`.  However, the rolling code has been co-opted to include data about the current status of the sun and wind.  What is very odd to me here is that it emits a My command which if this were a linked remote would send the awning to the favorite position.  I do not see an instance where any Somfy device could make a determination that the sender is a Soliris.

The third decoded byte is only part of the story because the entire data element in that output is 16 bits not 8.  It appears that the RTS protocol reserves rolling code values <= 2 and  >= 32770 or perhaps ignores the rolling code when the command is the sun flag.  

I have never rolled over any of the codes but if you do the math on this once a remote is paired a value of 2 can still be reached but only on the first button press after pairing.  It looks like the high order bit (sign flag) may be ignored so the rolling code is only uses 15 bits worth of information and not 16 so when it hits 32768 it rolls back to 0.  

So if you were to mask the value, the high order bit could be our indicator.   When it is set the rcode value will be 32770 and 2 when it is not.  There is lots of weirdness with the protocol but every bit in the 56-bit stream is accounted for.  It is noteworthy that the Soliris has an address that is very large where the high order of the 24-bit representation is set.  Accessory devices may simply have this bit set so the command structure doesn't interfere with shades, blinds, awnings, or draperies.

The final tidbit of this which has me curious.  When you press the prog button on the Soliris does it send the command in both protocols?  I can already see it is communicating in RTS but all of the RTW devices I have seen do not recognize the RTS protocol.  Then again if it is communicating in RTS then how does your awning know to move.  I would love to capture this being triggered.

Noltari, 2023-05-29:
> Interesting. This looks like the first two sets of commands on the first post are definitely the RTW protocol. Just to be sure I assume these are all from the remote on two different channel [...]

Unfortunately, pressing the prog button of the Soliris isn't an easy task, due to its location (it's drilled into the building façade close to one of the awnings, and I live on the 3rd floor). However, if I mange to press it I will try to report the data.

Yeah, and in fact my devices don't respond to RTS protocol, only to RTW, but I guess that the Sun Status is the only RTS thing supported...

rstrouse, 2023-05-29:
If you could capture an instance when the Soliris makes the awning move that would be an eye opener.  I presume that the Soliris simply sends an up or down command when it meets the conditions.  The prog command would have been really helpful.  Where is Spiderman when you need him?

rstrouse, 2023-05-31:
Ok so I took the first stab at adding the functionality for setting the sun flag.  It has not been released so pull from the repo.

If my understanding is correct, clicking on the sun icon should enable and disable the sun flag.
[image]

Noltari, 2023-05-31:
> Ok so I took the first stab at adding the functionality for setting the sun flag.  It has not been released so pull from the repo.
> 
> 
> 
> If my understanding is correct, clicking on the sun icon should enable and disable the sun flag.
> 
> [image]
> 
> 

Thanks!
I will test it this afternoon and report back.

Henk72, 2023-05-31:
Thank you for the great work. I will test this also later this week!

rstrouse, 2023-05-31:
> Thank you for the great work. I will test this also later this week!

Do you have a Sun/Wind sensor?  If you do then it would be great if I could get a log of the comms when you press the prog button on the back of the Soliris and all of the demo mode outputs.

Noltari, 2023-05-31:
> Ok so I took the first stab at adding the functionality for setting the sun flag. It has not been released so pull from the repo.
> 
> If my understanding is correct, clicking on the sun icon should enable and disable the sun flag. [image]

Confirmed, the awning jogs when the sun icon is clicked and the awnings go down if there is sun and go up if there's no sun.
Thanks!

BTW, it would be great to be able to enable/disable the sun mode via Home Assistant. Maybe we could add it via switch platform?
And it would also be great to have a binary sensor both in the website and Home Assistant with the value reported by the sun sensor indicating if there's sun or not.

rstrouse, 2023-05-31:
Yes I will be releasing an update to the home assistant component this week.  It will include the new awning cover type as well.  A sensor will be added.  I'll have to investigate if there are already options on the cover domain for the switch.  If not I may just add it as a separate entity.

rstrouse, 2023-05-31:
Btw I am pretty sure the sun indicator is coming through on 0x0E with an extension in the rolling code bytes.  However, I really need to get a handle on that address.  So we need to make sure we capture which address belongs to the Soliris and which ones belong to the awning.  Joining it as a linked remote will give us what we want.

Noltari, 2023-05-31:
> Btw I am pretty sure the sun indicator is coming through on 0x0E with an extension in the rolling code bytes. However, I really need to get a handle on that address. So we need to make sure we [...]

Ok, I will try to do that in the following days.
I'll probably try to find some really large stick and attach something to it...

rstrouse, 2023-06-01:
Alright I am adding the awning to HA but I have found the icons are backwards and research has led me to yet another series of dismissive crap.  This is probably why every integration that tries to use the standard cover entity has to create their own.  Honestly I do not understand the ridiculousness of it.  The percentage of open is fraught with the same silliness.
https://github.com/home-assistant/frontend/issues/5513

Below is what it looks like with the awning retracted.  The little compression icon is to extend it.  The icons work for a gate but not for an awning and apparently there is no changing it.  There are so many talking heads trying to justify this that I feel the need to enter the fray with swords out.  As a dev I can truly say that the logic evades me.  If you have that many people saying you are wrong you are at least mistaken.  If anybody can remove the non from this and make sense out of it I would love to hear it.  At least the arrows are correct for shades.
[image]

Henk72, 2023-06-01:
Rstrouse, I couldn't agree more. The icon and logic indeed makes no sense. 

I also have an awning that extends. I use your software and HA integration and at least as shade, the arrows direction works logically. The status and percentage however are inverted (at least in my opinion). When the awning is retracted ('closed'), the status is open. If the awning is fully extended to create the shade, it is closed. 
Apart from the confusion, other platforms that I use with HA show the closed awning as open and therefore they look active in Apple Home and Google Home.

Still I'm very happy with at least the arrow directions.

Noltari, 2023-06-01:
> 

I totally agree with you and that's why I usually expose awnings as shades...
[image]

Henk72, 2023-06-01:
I tried to get some log files when putting the remote to automatic Sun mode. Maybe it can help because you have now feedback from 2 people. I used your standard firmware no special pull. It seems to recognise the sun button.

**Press Down and stop it by pressing My**
[{"encKey": 165,
  "address": 10147277,
  "rcode": 3932,
  "command": "Down",
  "rssi": -67,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [95980,10443,7350,2463,2575,2466,2571,4803,1319,1262,1329,1256,684,593,1344,1240,1321,629,677,597,674,633,665,1270,678,606,681,619,1322,1251,1346,599,676,598,700,1259,1318,612,686,1242,1325,623,668,1259,1340,597,699,1233,696,603,680,604,1341,1256,663,621,1321,1259,1324,1261,1318,627,679,615,666,618,668,609,689,623,670,607,681,628,665,619,672,1260,673,626,677,614,1316,624,677,1257,670,606,1331,619,668],
  "time": "2023-06-01T10:15:44.733+0200"
},
{"encKey": 166,
  "address": 10147277,
  "rcode": 3933,
  "command": "My",
  "rssi": -66,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [1036,714,6545,18031,161132,162069,478,638,653,1448,39542,103069,101825,51822,10426,7352,2480,2566,2458,2568,4814,1328,1247,1323,1276,667,611,1318,618,672,1269,1325,1262,1342,601,652,1286,1322,608,680,1245,1330,1270,1304,625,666,633,661,1277,670,608,1341,591,693,614,665,619,671,1263,690,619,1318,1270,668,605,678,615,1334,602,683,603,678,624,671,615,688,614,668,1267,1332,1252,1319,1268,1314,1270,1303,623,688,1262,1314,626,676,1245,694,603,1336,618,671,1260],
  "time": "2023-06-01T10:15:47.198+0200"
}]

-------------
**Press and hold Sun button for 5 sec to turn Sun on**
[{"encKey": 165,
  "address": 10147277,
  "rcode": 3932,
  "command": "Down",
  "rssi": -67,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [95980,10443,7350,2463,2575,2466,2571,4803,1319,1262,1329,1256,684,593,1344,1240,1321,629,677,597,674,633,665,1270,678,606,681,619,1322,1251,1346,599,676,598,700,1259,1318,612,686,1242,1325,623,668,1259,1340,597,699,1233,696,603,680,604,1341,1256,663,621,1321,1259,1324,1261,1318,627,679,615,666,618,668,609,689,623,670,607,681,628,665,619,672,1260,673,626,677,614,1316,624,677,1257,670,606,1331,619,668],
  "time": "2023-06-01T10:15:44.733+0200"
},
{"encKey": 166,
  "address": 10147277,
  "rcode": 3933,
  "command": "My",
  "rssi": -66,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [1036,714,6545,18031,161132,162069,478,638,653,1448,39542,103069,101825,51822,10426,7352,2480,2566,2458,2568,4814,1328,1247,1323,1276,667,611,1318,618,672,1269,1325,1262,1342,601,652,1286,1322,608,680,1245,1330,1270,1304,625,666,633,661,1277,670,608,1341,591,693,614,665,619,671,1263,690,619,1318,1270,668,605,678,615,1334,602,683,603,678,624,671,615,688,614,668,1267,1332,1252,1319,1268,1314,1270,1303,623,688,1262,1314,626,676,1245,694,603,1336,618,671,1260],
  "time": "2023-06-01T10:15:47.198+0200"
},
{"encKey": 167,
  "address": 10147277,
  "rcode": 3934,
  "command": "Sun Flag",
  "rssi": -68,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [200179,102433,189842,10426,10700,2467,2579,2462,2573,4811,1308,1248,1338,1246,682,616,1329,603,680,605,680,1269,660,638,1295,635,684,616,668,609,684,1241,1320,1290,645,634,1312,619,673,1266,674,625,1303,1274,671,610,1316,620,673,1266,1325,608,662,1277,693,609,1315,621,668,612,684,595,695,1257,1317,623,652,1282,665,629,658,613,1322,1260,678,619,673,608,1334,622,652,632,681,619,671,612,665,1273,1324,600,672,625,680,1257],
  "time": "2023-06-01T10:16:20.268+0200"
},
{"encKey": 167,
  "address": 10147277,
  "rcode": 3934,
  "command": "Sun Flag",
  "rssi": -68,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [13106,14473,2493,2569,2474,2572,2471,2563,2474,2541,2480,2562,2484,2544,2486,2543,4814,1308,1272,1326,1274,671,612,1314,620,672,610,693,1243,674,634,1317,622,650,634,665,619,668,1262,1314,1284,658,619,1319,614,671,1276,661,641,1293,1284,670,603,1331,616,669,1261,1316,624,680,1260,669,629,1311,618,669,608,687,625,670,1253,1336,596,675,1259,698,602,679,607,1316,1276,664,622,669,630,1310,620,668,612,682,615,669,632,659,1266,1324,606,690,621,667,1262],
[... 200 lines cut ...]
  "pulses": [568,26977,2497,2549,2481,2572,2447,2577,2474,2561,2478,2566,2455,2569,2484,2544,4816,1302,1267,1325,1252,1320,1269,667,605,682,614,696,604,678,615,667,618,668,612,1314,644,670,613,661,622,668,1282,665,617,668,611,684,619,667,629,658,629,665,622,671,609,690,595,1322,1283,1297,636,663,624,670,604,683,605,678,621,673,608,691,1246,676,637,662,622,1318,1258,1320,1260,675,621,678,607,1316,1275,664,623,671,604,681,630,1327,609,659,1269,1320,610,690,595,672,1280,1291],
  "time": "2023-06-01T10:22:43.384+0200"
},
{"encKey": 168,
  "address": 10147277,
  "rcode": 3935,
  "command": "Flag",
  "rssi": -66,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [621,25976,500,623,2486,2564,2461,2567,2488,2554,2462,2575,2463,2569,2474,2549,2476,2582,4777,1319,1279,1316,1266,1295,1283,674,600,681,605,691,619,671,609,690,595,693,609,1312,619,669,609,688,622,665,1257,691,619,668,608,684,616,671,602,682,629,676,596,699,606,666,618,1321,1256,1319,633,658,620,668,611,688,625,670,602,681,629,651,1272,686,602,673,625,1304,1271,1322,1265,671,610,666,621,1318,1285,645,627,680,605,24,690,619,1297,627,677,1273,1314,620,672,609,690,1246,1322],
  "time": "2023-06-01T10:22:43.510+0200"
},
{"encKey": 168,
  "address": 10147277,
  "rcode": 3935,
  "command": "Flag",
  "rssi": -72,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [590,26990,2484,2561,2464,2583,2466,2575,2463,2575,2453,2569,2468,2580,2450,2567,4803,1322,1267,1319,1241,1345,1243,669,611,691,623,671,603,681,605,693,617,671,610,1315,620,674,609,689,596,693,1259,665,619,694,610,664,624,670,606,682,605,677,621,674,608,691,619,1293,1269,1343,615,665,621,670,609,690,596,695,608,685,600,674,1252,688,625,671,605,1333,1246,1328,1248,670,634,665,624,1321,1261,675,621,677,607,666,619,1319,629,676,1246,1338,594,689,609,687,1245,1322],
  "time": "2023-06-01T10:22:43.650+0200"
}]

Noltari, 2023-06-01:
@rstrouse if I disable the sun mode with my physical remote and then I enable it from the ESPSomfy website, the awnings start reacting to the sun status messages sent by the Sun/Wind sensor and so far I haven't seen a Down command from the sun/wind sensor, just the sun status:

<details>
<summary>Frame logs:</summary>

```
[{"encKey": 169,
  "address": 10569114,
  "rcode": 2,
  "command": "Unknown(14)",
  "rssi": -79,
  "bits": 56,
  "valid": false,
  "sync": 15,
  "pulses": [620,617,1304,1255,647,647,1280,618,643,1275,656,623,1302,632,636,606,23710,2979,2464,2527,2459,2546,2454,2521,2471,2541,2449,2535,2466,2512,2467,2541,4778,1255,1263,1309,1267,1284,1269,636,635,1287,1298,1256,1275,647,651,1263,637,646,647,636,666,616,1264,1291,1283,641,623,1290,633,639,645,630,635,658,1289,1252,1285,639,672,1263,634,631,1299,1265,635,639,644,647,631,637,1261,1291,657,628,1266,653,631,1306,1251,1287,1267,1304,1256,648,640,1270,1292,647,625,1277,644,640,1281,666,620,1269,634,640],
  "time": "2023-06-01T10:32:30.390+0200"
},
{"encKey": 169,
  "address": 10569114,
  "rcode": 2,
  "command": "Unknown(14)",
  "rssi": -82,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [648,17131,803,8753,2432,2551,2453,2547,2450,2502,2482,2537,2465,2487,2486,2535,2458,2520,4778,1254,1282,1305,1278,1279,1274,639,637,1277,1270,1295,1275,627,655,1280,647,623,654,617,651,642,1274,1288,1275,635,635,1293,621,654,630,634,657,646,1272,1262,1288,626,654,1291,620,650,1282,1271,643,650,634,641,634,647,1296,1257,628,642,1274,658,652,1255,1281,1289,1270,1291,1254,646,651,1288,1270,631,632,1289,643,629,1281,649,642,1280,637,640],
  "time": "2023-06-01T10:32:30.401+0200"
},
{"encKey": 169,
  "address": 10569114,
  "rcode": 2,
  "command": "Unknown(14)",
  "rssi": -81,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [612,11999,5648,778,4755,3518,2452,2521,2494,2514,2443,2542,2465,2532,2439,2544,2498,2497,2446,2535,4748,1303,1252,1294,1277,1291,1266,647,620,1286,1283,1303,1237,654,648,1294,614,643,646,655,609,632,1306,1257,1286,637,673,1261,636,631,626,647,676,630,1258,1280,1306,611,636,1277,654,629,1268,1301,663,621,623,649,627,662,1261,1287,627,638,1286,643,626,1288,1288,1269,1287,1267,1273,636,658,1267,1272,663,609,1295,644,632,1283,648,620,1303,609,650],
  "time": "2023-06-01T10:32:30.742+0200"
},
{"encKey": 169,
  "address": 10569114,
  "rcode": 2,
  "command": "Unknown(14)",
  "rssi": -81,
  "bits": 56,
  "valid": false,
  "sync": 15,
  "pulses": [621,16058,822,559,6560,2702,2450,2544,2445,2539,2460,2536,2436,2537,2457,2551,2425,2543,2456,2536,4773,1277,1251,1281,1293,1271,1283,642,620,1279,1294,1296,1262,651,620,1306,610,639,648,646,627,664,1261,1280,1282,628,623,1304,632,637,651,649,621,657,1283,1267,1276,647,638,1280,632,633,1267,1298,655,597,669,628,630,657,1285,1267,646,631,1277,645,630,1292,1343,1192,1297,1288,1264,624,648,1288,1260,671,626,1281,630,662,1251,662,614,1289,646,631],
  "time": "2023-06-01T10:32:30.742+0200"
},
{"encKey": 169,
  "address": 10569114,
  "rcode": 2,
  "command": "Unknown(14)",
  "rssi": -82,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [634,16520,768,66371,7301,95075,102437,89011,705,12682,17263,450,50811,2427,2550,2449,2538,4743,1315,1251,1277,1288,1290,1256,660,623,1278,1286,1264,1275,660,634,1297,621,628,653,638,662,624,1273,1287,1276,636,632,1286,621,672,605,662,763,531,1247,1288,1279,647,654,1258,626,651,1303,1261,646,646,603,661,637,652,1249,1308,627,646,1275,632,632,1293,1253,1303,1271,1288,1276,652,629,1285,1249,656,634,1294,623,651,1255,679,624,1277,632,661],
  "time": "2023-06-01T10:32:31.357+0200"
},
{"encKey": 169,
  "address": 10569114,
  "rcode": 2,
  "command": "Unknown(14)",
  "rssi": -83,
  "bits": 56,
  "valid": false,
  "sync": 5,
  "pulses": [629,16065,831,9845,2456,2548,2424,2568,2426,2545,2445,2541,2457,101,327,376,550,1982,2460,2548,2435,2542,4758,1287,1254,1287,1274,1306,1244,674,606,1308,1271,1277,1281,649,624,1284,619,672,626,663,611,657,1249,1312,1264,648,623,1268,632,675,648,632,630,658,1255,1295,1257,652,622,1287,634,660,1264,1301,606,636,645,675,628,662,1232,1298,657,632,1257,647,627,1290,1270,1303,1244,1319,1258,648,645,1279,1287,618,647,1280,629,647,1274,660,612,1292,624,657],
[... 145 lines cut ...]
DEC 169 230   0   2 161  69 154   0   0   0 
INVALID FRAME KEY:169 ADDR:10569114 CMD:Unknown(14) RCODE:2
    KEY  1   2   3   4   5   6  
--------------------------------
ENC 169  79  79  77 236 169  51   0   0   0 
DEC 169 230   0   2 161  69 154   0   0   0 
INVALID FRAME KEY:169 ADDR:10569114 CMD:Unknown(14) RCODE:2
    KEY  1   2   3   4   5   6  
--------------------------------
ENC 169  79  79  77 236 169  51   0   0   0 
DEC 169 230   0   2 161  69 154   0   0   0 
INVALID FRAME KEY:169 ADDR:10569114 CMD:Unknown(14) RCODE:2
    KEY  1   2   3   4   5   6  
--------------------------------
ENC 169  79  79  77 236 169  51   0   0   0 
DEC 169 230   0   2 161  69 154   0   0   0 
INVALID FRAME KEY:169 ADDR:10569114 CMD:Unknown(14) RCODE:2
    KEY  1   2   3   4   5   6  
--------------------------------
ENC 169  79  79  77 236 169  51   0   0   0 
DEC 169 230   0   2 161  69 154   0   0   0 
```
</details>

Therefore, apparently linking the sun/wind sensor with ESPSomfy-RTS remote isn't needed, since my awnings seem to link directly with the sun/wind sensor.

Noltari, 2023-06-01:
This is weird, I've seen the Sun/Wind sensor emit this codes along with the other Unknown(14):

<details>
<summary>With sun:</summary>

```
[{"encKey": 169,
  "address": 10569114,
  "rcode": 32770,
  "command": "Sun Flag",
  "rssi": -78,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [619,2988,101,13728,645,4883,4493,2463,2530,2486,2526,2463,2525,2482,2510,2474,2531,2474,2531,2446,2547,4757,1269,1324,1263,1275,1265,1285,643,648,1288,1265,701,154,307,433,582,1290,620,653,1265,641,642,676,607,659,626,1322,1239,1309,621,626,1291,651,620,649,637,640,650,1270,1292,1304,599,649,1269,677,594,1297,1294,623,650,651,637,640,639,1274,1265,657,614,1317,635,643,1252,1299,1266,1299,1257,1288,644,647,1265,1307,621,657,1243,679,621,1302,619,652,1268,652,637],
  "time": "2023-06-01T11:18:15.602+0200"
},
{"encKey": 150,
  "address": 10569114,
  "rcode": 2,
  "command": "Up",
  "rssi": -78,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [658,15760,815,2504,48,7628,2467,2507,2488,2521,2456,2544,2452,2531,2475,2535,2451,2542,2455,2532,4795,1256,1348,26,229,333,461,753,1290,1275,1286,640,647,1277,1264,1276,1304,639,639,1278,626,636,661,639,646,625,1302,1267,1273,640,658,1267,649,612,683,612,674,599,1298,1276,1295,610,667,1270,655,612,1285,1287,633,637,644,645,653,638,1288,1252,659,632,1293,620,648,1290,1295,1268,1271,1269,1285,646,644,1276,1278,649,652,1265,650,644,1275,630,649,1273,635,660],
  "time": "2023-06-01T11:18:20.829+0200"
}]
```
</details>

<details>
<summary>Frame logs:</summary>

```
[{"encKey": 169,
  "address": 10569114,
  "rcode": 32768,
  "command": "Sun Flag",
  "rssi": -81,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [647,542,466,11553,3454,809,935,4248,552,4139,2465,2531,2450,2555,2448,2534,2472,2525,2443,2535,2469,2521,2459,2520,4778,1274,1277,1282,1274,1279,1292,629,636,1290,1321,409,486,736,1289,621,674,1269,653,614,1292,1272,1287,1258,1289,629,671,1260,668,603,1298,1261,1290,1279,1283,614,649,1272,657,637,1294,1257,658,616,639,645,651,638,1286,1255,655,634,1289,619,656,1262,1315,1254,1295,1245,1308,627,645,1273,1283,644,620,1286,640,642,1300,587,683,1255,649,611],
  "time": "2023-06-01T15:39:29.078+0200"
},
{"encKey": 166,
  "address": 10569114,
  "rcode": 32768,
  "command": "Up+Down",
  "rssi": -80,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [613,12078,4788,1025,3964,544,4285,2470,2524,2469,2525,2460,2530,2465,2533,2454,2545,2472,2517,2442,2516,4799,1263,1272,1286,1274,514,45,773,1283,625,648,1270,1314,505,460,271,1574,616,674,1261,643,641,1270,1268,1307,1259,1284,633,674,1257,658,609,1291,1267,1283,1281,1295,612,664,1257,670,609,1287,1279,674,611,635,628,662,642,1276,1270,647,621,1304,615,642,1263,1335,1251,1281,1262,1297,634,633,1286,1293,634,634,1271,651,633,1287,619,643,1290,631,619],
  "time": "2023-06-01T15:39:29.494+0200"
}]
```
</details>

rstrouse, 2023-06-01:
> I tried to get some log files when putting the remote to automatic Sun mode. Maybe it can help because you have now feedback from 2 people. I used your standard firmware no special pull. It seems to recognise the sun button.
> 
> **Press Down and stop it by pressing My** 

Great this looks like your awning is a Pure RTS not RTW.  Now I can make sure we normalize this.

> **Press and hold Sun button for 5 sec to turn Sun on**

Wait are 5 seconds required to turn the sun on?
 
> **Press Sun button to turn Sun off**

Are 5 seconds also required to turn the sun off?

rstrouse, 2023-06-01:
@Noltari so address 10569114 is the sun/wind sensor device and not the remote.  The motors only have ears and no mouth.  It is odd that it is issuing an up command when there is sun and an up+down command when there is no sun.  I would have expected this to be reversed and without a button combo.

While the address of the sensor is what is sending this command the sensor has been paired with the motor and should respond.  up+down is typically only sent when setting the endpoint adjustments.

Noltari, 2023-06-01:
> Great this looks like your awning is a Pure RTS not RTW.  Now I can make sure we normalize this.
>
>
>
>
>
>
> Wait are 5 seconds required to turn the sun on?

Yes, this happens on my remote too.

> 
> 
> 
> Are 5 seconds also required to turn the sun off?

And also this.

It's basically the same button and it will light 2 LEDs indicating sun on or the other 2 LEDs indicating sun off after those 5s and just then is the SunFlag or Flag command issued.

rstrouse, 2023-06-01:
Oh so it doesn't issue 5 seconds of sunflag/flag pulses.  It simply debounces for 5 seconds on the remote before issuing the short commands.  

The reason for the confusion is that Somfy has a couple of different modes for each button.  There is a short press, long press, and really long press.  The really long press is related to opening the shade memory and setting the my position.  The long press is used for separate tilt motors.  If you long press it will move the tilt and short press moves the lift.

Noltari, 2023-06-01:
BTW @rstrouse it's funny that the remote doesn't allow to individually change the sun/flag on a single awning, since this can only be done on the fifth channel (the one where you're supposed to link all the awnings for the group commands).
However, with your implementation I can successfully change the sun/flag on every single awning :)

rstrouse, 2023-06-01:
> BTW @rstrouse it's funny that the remote doesn't allow to individually change the sun/flag on a single awning, since this can only be done on the fifth channel (the one where you're supposed to link [...]

That is funny.   Given the sparse nature of their remotes, the engineer had to be a one-button mouse fanboy.  When I get around to creating groups we should be able to replicate that booger no matter how gross it is.

Noltari, 2023-06-01:
Ok, so I replaced the antenna with the one I had on the RFLink device and adjusted both the RX bandwidth and the Frequency Deviation to 150 kHz...
Yes, I know those values deviate a lot from the ones you posted on the wiki, but it turns out I couldn't capture the sun/wind sensor packets with the values from the wiki...
And I'm still capturing packets from 10569114, so I guess that one is from my neighbor :)

And here are the results:
<details>
<summary>No wind + No sun</summary>

```
*** PENDING TO BE CAPTURED ***
But it's going to be rcode=0
```
</details>

<details>
<summary>Wind + No sun</summary>

```
[{"encKey": 169,
  "address": 12602266,
  "rcode": 1,
  "command": "Unknown(14)",
  "rssi": -54,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [101509,102398,15289,87145,102430,5710,96783,102249,102396,3305,50,101,153,92773,6329,7940,88120,6319,102405,26648,2515,2513,2471,2513,4807,1251,1294,1267,1299,1259,1312,615,646,1263,1302,1254,1313,618,669,609,658,1281,1293,1263,1290,1259,1312,644,629,611,661,1278,1292,1260,1315,1256,1290,624,656,637,641,1274,1299,615,657,1270,1305,614,665,620,657,610,661,1275,1301,610,657,1270,653,616,1309,618,648,1264,679,618,645,614,685,615,1301,1257,1309,1246,1333,1245,1297,1262],
  "time": "2023-06-01T18:55:01.408+0200"
},
{"encKey": 169,
  "address": 12602266,
  "rcode": 1,
  "command": "Unknown(14)",
  "rssi": -54,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [640,16633,828,752,4473,3969,2510,2521,2489,2496,2478,2521,2496,2496,2498,2516,2474,2511,2500,2509,4820,1220,1295,1269,1298,1258,1309,619,674,1240,1302,1255,1312,619,668,614,659,1270,1293,1263,1290,1276,1286,617,674,610,660,1258,1318,1242,1304,1277,1289,619,670,611,661,1266,1295,614,662,1254,1318,617,657,616,672,624,652,1262,1291,631,662,1269,645,624,1306,625,645,1274,662,604,660,625,654,610,1317,1274,1285,1272,1297,1267,1320,1240],
  "time": "2023-06-01T18:55:01.468+0200"
},
{"encKey": 169,
  "address": 12602266,
  "rcode": 1,
  "command": "Unknown(14)",
  "rssi": -55,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [643,16345,669,769,598,2370,128,280,462,5643,2487,2528,2473,2532,2477,2498,2479,2522,2491,2496,2504,2495,2490,2502,4803,1255,1319,1243,1305,1249,1314,619,671,1261,1284,1270,1313,596,669,628,658,1270,1298,1259,1283,1275,1310,622,642,631,660,1268,1296,1261,1283,1272,1287,622,695,607,660,1242,1296,633,662,1266,1296,611,661,628,672,619,652,1264,1290,631,660,1259,669,602,1308,627,668,1255,662,602,670,621,653,615,1310,1269,1286,1274,1300,1257,1296,1263],
  "time": "2023-06-01T18:55:01.713+0200"
},
{"encKey": 169,
  "address": 12602266,
  "rcode": 1,
  "command": "Unknown(14)",
  "rssi": -55,
  "bits": 56,
  "valid": false,
  "sync": 14,
  "pulses": [631,17471,1265,726,614,4967,1607,2520,2519,2472,2511,2492,2510,2488,2507,2492,2507,2489,2511,2475,2525,4797,1243,1319,1242,1302,1277,1289,618,696,1242,1308,1247,1313,595,668,657,636,1240,1317,1282,1289,1253,1287,618,670,611,685,1241,1317,1264,1301,1252,1290,645,645,636,640,1269,1297,610,663,1266,1296,637,661,616,653,616,674,1268,1278,625,655,1265,672,618,1307,616,650,1270,660,607,679,617,646,601,1340,1244,1307,1248,1322,1244,1297,1261],
  "time": "2023-06-01T18:55:01.751+0200"
},
{"encKey": 169,
  "address": 12602266,
  "rcode": 1,
  "command": "Unknown(14)",
  "rssi": -55,
  "bits": 56,
  "valid": false,
  "sync": 14,
[... 752 lines cut ...]
ENC 169  71  71  68 132 207  85   0   0   0 
DEC 169 238   0   3 192  75 154   0   0   0 
INVALID FRAME KEY:169 ADDR:12602266 CMD:Unknown(14) RCODE:3
    KEY  1   2   3   4   5   6  
--------------------------------
ENC 169  71  71  68 132 207  85   0   0   0 
DEC 169 238   0   3 192  75 154   0   0   0 
INVALID FRAME KEY:169 ADDR:12602266 CMD:Unknown(14) RCODE:3
    KEY  1   2   3   4   5   6  
--------------------------------
ENC 169  71  71  68 132 207  85   0   0   0 
DEC 169 238   0   3 192  75 154   0   0   0 
INVALID FRAME KEY:169 ADDR:12602266 CMD:Unknown(14) RCODE:3
    KEY  1   2   3   4   5   6  
--------------------------------
ENC 169  71  71  68 132 207  85   0   0   0 
DEC 169 238   0   3 192  75 154   0   0   0 
```
</details>

So I guess it's obvious that the rolling code is used for:
- BIT(0) = Wind
- BIT(1) = Sun

BTW, I may not be able to reach my sun/wind sensor for pressing Prog button, but at least I could use a dust blower to simulate the wind xD

Noltari, 2023-06-01:
> 10569114

And this explains the fact that I need different RX bandwith and Frequency Deviation parameters:

The Sun/Wind sensor is transmitting at 433.38 MHz:
[image]

The Arqualuz remote is transmitting at 433.39 MHz:
[image]

And the sun/wind sensor from my neighbor (10569114) is transmitting closer to 433.42 MHz:
[image]

rstrouse, 2023-06-01:
And finally somebody who is a radiohead!  I am simply a bit flipper who focuses on that side of things.  Honestly, I have had a few users who needed to tweak the RX Bandwidth to hear their remotes so I believe this to be differences related to the manufacturing of the device.  Not quite a full channel but enough that the bandwidth and deviation can be tuned to capture it.

I may kick around the ability to change the base frequency but unless folks are at your level with this it may be a bridge too far.  Perhaps I can create a center frequency and allow the user to move it within the channel.

Noltari, 2023-06-01:
> And finally somebody who is a radiohead!  I am simply a bit flipper who focuses on that side of things.  Honestly, I have had a few users who needed to tweak the RX Bandwidth to hear their remotes [...]

I think this is the best solution. Being able to change the base frequency would be great and I vote for that.
Also, I only used a rtlsdr to see what was going on so please don't take me for someone who knows a lot of that 😅

> Perhaps I can create a center frequency and allow the user to move it within the channel.

rstrouse, 2023-06-01:
If you update with the repo I added the ability to tweak the base frequency.

rstrouse, 2023-06-02:
Ok so I have pushed the code to capture the sensor flags and made indicators for sunny and windy.  Below is what it looks like when the windy thingy kicks in.
[image]

And this is what the sunny looks like.
[image]

To get this to work you will need to link the sun/wind sensor as if it is a remote.  To do this simply click the link remote button then press the prog button or trigger the wind or sun sensor.

Noltari, 2023-06-02:
> If you update with the repo I added the ability to tweak the base frequency.

Thanks, I'm now using this config and so far so good ^^
[image]

> Ok so I have pushed the code to capture the sensor flags and made indicators for sunny and windy. Below is what it looks like when the windy thingy kicks in. [image]
>
> And this is what the sunny [...]

Wow! Thanks!

I will test this today with one minor modification: https://github.com/rstrouse/ESPSomfy-RTS/pull/62
The rolling code can be 0 for `Sensor` so we need to remove the `rollingCode > 0` for that specific case.

I will report any other findings as I test it :)

Noltari, 2023-06-02:
So this is working great:
[image]

The only thing I'm missing is that we should probably add some more logic so that:
- somfy_flags_t::Windy received, the awnings should update their position to Up (as if an Up command had been issued). This doesn't depend on the SunFlag at all.
- SunFlag Enabled + somfy_flags_t::Sunny set, the awnings should update their position to Down as if a Down command had been issued.
- SunFlag Enabled + somfy_flags_t::Sunny NOT set, the awnings should update their position to Up as if an Up had been issued.

Henk72, 2023-06-02:
Great work guys!!!

Since the HA awning/shade logic seems inverted, would it be an idea to make a workaround: a user parameter to invert all logic in the integration?

rstrouse, 2023-06-02:
@Noltari just a little bit of clarification on the rules before I change the awning targets.  Btw thanks for testing the frame decoding. I don't have anything generating that frame right now.
> * somfy_flags_t::Windy received, the awnings should update their position to Up (as if an Up command had been issued). This doesn't depend on the SunFlag at all.

1. If there is a windy frame issued can this be overridden by pressing the down button on the remote or is everything disabled?

> * SunFlag Enabled + somfy_flags_t::Sunny set, the awnings should update their position to Down as if a Down command had been issued.

1. I believe this is a My command if it has been set and a down if not.  Is this how it actually works?  
2. If you have extended the awing already to say 10% and the sun comes out does it still push them to the position discussed in   the previous question?

> * SunFlag Enabled + somfy_flags_t::Sunny NOT set, the awnings should update their position to Up as if an Up had been issued.

1. If I have manually set the position of the awning will it still retract on its own if Ra decides to take a nap?

@Henk72  thanks for the vote of confidence.
> Since the HA awning/shade logic seems inverted, would it be an idea to make a workaround: a user parameter to invert all logic in the integration?

Unfortunately, the original dev of the cover domain for HA was clearly dealing with something that was normally closed.  The entire logic stream works great for a rolling gate where you would think of it in percentage of open.  You might even make a wobbly case for a garage door, window, or even a drapery.  

Where this all starts to fall apart is when you try to apply this same logic to a shade, blind, or awning.  Logically, the zero position is up or retracted for such a device.  To get around this weirdness we simply accept that the position is expressed in percentage of open.  We can do this with shades because the icons still make sense an up arrow moves the cover up and a down arrow moves it down.

Now when we start talking about an awning things get a bit weird.  The reason is that it doesn't quite fit the notion of open or closed or up and down.  You can either extend or retract it to throw a verb or two out there.  Unfortunately, that is a foreign concept to the cover domain in HA.   

As a result, what has been done with many of cover integrations is to perform gymnastics in the code to create some semblance of the desired results.  It's not ideal but it is workable and I suspect that it has cemented the demented in that changing it now would break the tweaks.  

In fact it would still be workable if the icons were in the correct positions.  But that is not what we are dealing with and I have no faith that I can convince a change to make it all right.  So the integration will simply engage the extend verb when the left icon is pressed and the retract verb on the right.  While this may be a direct assault on muscle memory, at least the icons will be correct.

Noltari, 2023-06-02:
> @Noltari just a little bit of clarification on the rules before I change the awning targets. Btw thanks for testing the frame decoding. I don't have anything generating that frame right now.
>
>
> 1. If [...]

It will ignore the Down commands for some seconds, but after those seconds you will be able to do that.

> > * SunFlag Enabled + somfy_flags_t::Sunny set, the awnings should update their position to Down as if a Down command had been issued.
> 
> 1. I believe this is a My command if it has been set and a down if not.  Is this how it actually works?

Actually I haven't set the My position on my awnings and I'm not sure that this is supported on my Arqualuz motors, since my awnings just stop when I press the My button if they are moving and they do nothing if they aren't moving.
What's a My command supposed to do? Set it to a memorized position or something like that?

> 2. If you have extended the awing already to say 10% and the sun comes out does it still push them to the position discussed in   the previous question?

Yes, that awning will go all the way down.

> > * SunFlag Enabled + somfy_flags_t::Sunny NOT set, the awnings should update their position to Up as if an Up had been issued.
> 
> 1. If I have manually set the position of the awning will it still retract on its own if Ra decides to take a nap?

Yes, it will as long as the SunFlag is enabled for that awning.

rstrouse, 2023-06-02:
I didn't include all the levels so we don't look like we are conversing in python.

> It will ignore the Down commands for some seconds, but after those seconds you will be able to do that.

Doesn't it send this repeatedly?  If you do that won't it just retract it again or is it maintaining an internal state that it was moved manually.

> Actually I haven't set the My position on my awnings and I'm not sure that this is supported on my Arqualuz motors, since my awnings just stop when I press the My button if they are moving and they do nothing if they aren't moving.
What's a My command supposed to do? Set it to a memorized position or something like that?

Yes it moves to a memorized position.  So for instance if it is sunny it opens half-way.  The instructions state it moves to the my position.

> Yes, it will as long as the SunFlag is enabled for that awning.

Interesting.  I guess you are in control of the sun flag.

Henk72, 2023-06-02:
With my Solaris RTS I can use the My button to stop the extending and open the awning for example 50%. If the Sun automation is on, it often moves to full extension later automatically. This is because in sun mode it overrules your manual settings with the remote. 

On the Sun/Wind Solaris you can also 'program' the requested position. If you would set it to 50% and the awning is closed (not extended) and sun comes up beyond the threshold, it will be openend to 50%.

rstrouse, 2023-06-02:
I would almost put money on your not setting a my position.  If you extend your awning to say 50% then hold down the my button until it jogs it will have then set a my position.  From here on out if you press the my button when the awning is at rest it will go to that position.  If you want to clear it press the my button to allow it to go to the stored position then after it has stopped press and hold the my button again and the awning will once again jog.  At that point the my position will be cleared.

Henk72, 2023-06-02:
Ah, I only pressed it once. It acts as a stop. I get it now, thanks

rstrouse, 2023-06-04:
@Noltari I made a couple of changes related to the My position settings when it is set.  When this is not set then it is assumed to extent to 100%.

@Henk72 do you have access to your Soliris?  It would be great to capture the frames that are sent when the sensor is placed in demo mode then when the sensor is removed from demo mode.  Apparently this should job the awning when it happens.

I also managed to work my way through tweaking the functionality of the awning for HA.  This does some gymnastics with the services so that up is down and down is up.  But in this case it is a bit more insidious than that in that the icons are also reversed.  However, I think I struck a pretty good balance with the HA component which I will release soon.  I am testing it against an old firmware to make sure that it doesn't break anything.

rstrouse, 2023-06-04:
The companion component for Home Assistant has be added as a pre-release.  If you have updated to 1.7.0 then you can run this component and it will add the switches and sensors for awnings.

Henk72, 2023-06-05:
I will try to access the Solaris tomorrow and get the logs.

Noltari, 2023-06-05:
> I have put the awning in demo mode (wind to 100%) and removed it. This is the log:
> 
> [{"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -73, "bits": 56, "valid": true, "sync": 4, "pulses": [204698,13977,2446,2539,2471,2538,4751,1293,1271,1286,1276,649,637,638,655,622,643,654,636,641,624,1296,1263,666,620,645,656,634,632,1277,651,634,1285,1271,1283,643,647,644,629,635,641,1295,644,628,1283,1279,1276,661,625,646,648,1283,1266,1296,615,653,640,647,1260,664,616,1300,1292,620,650,634,639,657,620,1293,1262,647,644,656,614,1294,645,651,629,635,630,670,622,654,1263,1300,632,630,634,657,1270,1283], "time": "2023-06-05T14:55:47.171+0200" }, {"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -73, "bits": 56, "valid": true, "sync": 14, "pulses": [614,4916,21896,2457,2552,2462,2530,2465,2532,2465,2535,2457,2544,2454,2538,2447,2538,4773,1298,1254,1294,1272,633,661,632,640,646,629,663,633,650,618,1329,1228,668,627,637,635,655,644,1295,615,638,1316,1257,1276,645,634,638,640,644,649,1283,613,666,1255,1317,1267,633,635,657,645,1261,1289,1278,624,661,636,643,1272,648,635,1293,1270,656,635,640,648,648,605,1285,1280,644,637,634,661,1275,652,633,640,641,623,694,608,637,1266,1288,656,635,628,649,1271,1287], "time": "2023-06-05T14:55:47.242+0200" }, {"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -73, "bits": 56, "valid": true, "sync": 14, "pulses": [614,6785,562,2382,123,223,349,17088,2444,2568,2457,2522,2472,2534,2446,2541,2447,2564,2445,2549,2449,2544,4776,1287,1261,1294,1270,654,638,632,640,643,629,637,667,627,621,1298,1287,620,645,656,640,640,622,1296,635,639,1287,1270,1288,642,631,661,634,624,641,1297,639,646,1271,1292,1272,629,635,667,624,1294,1285,1263,652,633,640,640,1298,626,636,1296,1271,631,637,641,641,648,645,1281,1268,644,651,635,640,1267,646,650,661,631,626,649,619,657,1287,1272,630,663,644,614,1297,1309], "time": "2023-06-05T14:55:47.384+0200" }, {"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -72, "bits": 56, "valid": true, "sync": 14, "pulses": [575,6152,460,604,6930,12585,2474,2537,2447,2556,2450,2556,2449,2550,2445,2552,2447,2525,2467,2555,4752,1296,1270,1295,1263,651,631,639,641,642,645,647,634,639,629,1275,1301,640,624,668,629,634,636,1283,644,635,1317,1250,1274,639,640,661,625,636,660,1265,645,629,1314,1272,1259,658,642,640,630,1286,1266,1301,642,623,641,649,1287,627,643,1299,1259,668,627,629,635,655,640,1296,1264,644,645,633,636,1271,644,648,639,643,639,643,646,634,1286,1270,657,640,634,625,1293,1289], "time": "2023-06-05T14:55:47.517+0200" }, {"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -72, "bits": 56, "valid": true, "sync": 14, "pulses": [625,4935,569,499,1994,125,222,376,18689,2481,2568,2449,2517,2479,2532,2468,2503,2470,2555,2449,2530,2478,2546,4759,1290,1270,1285,1264,650,635,658,615,671,631,633,635,663,617,1301,1268,643,624,660,637,641,625,1297,635,643,1283,1300,1262,650,618,651,632,639,653,1271,632,657,1268,1293,1287,645,620,658,639,1267,1296,1260,637,666,629,634,1286,641,649,1281,1266,649,636,639,651,643,619,1312,1264,643,627,662,616,1290,644,630,660,642,626,647,623,660,1258,1294,632,662,632,649,1268,1289], "time": "2023-06-05T14:55:47.673+0200" }, {"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -73, "bits": 56, "valid": true, "sync": 4, "pulses": [5645,76342,102553,109275,95346,104365,120450,82389,102898,104192,111888,47080,2472,2544,2442,2548,4769,1274,1276,1295,1270,658,617,666,625,643,648,610,659,641,646,1273,1286,650,643,630,638,641,640,1276,651,634,1294,1274,1284,642,649,619,651,634,639,1281,646,627,1286,1292,1280,641,641,645,625,1283,1281,1294,636,633,645,643,1307,603,645,1270,1286,648,669,602,662,641,621,1297,1261,653,641,646,634,1291,644,623,657,641,624,648,642,623,1309,1265,630,657,639,646,1272,1291], "time": "2023-06-05T14:55:49.707+0200" }, {"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -73, "bits": 56, "valid": true, "sync": 14, "pulses": [617,7044,479,1633,1515,16166,2480,2525,2467,2543,2451,2551,2450,2542,2467,2535,2444,2552,2461,2524,4777,1281,1283,1271,1289,649,616,650,635,639,655,644,621,656,642,1287,1279,636,641,638,647,634,640,1287,616,651,1287,1271,1278,640,653,643,623,658,635,1293,618,676,1262,1295,1270,640,632,650,643,1277,1266,1294,639,634,646,644,1279,642,625,1294,1286,640,618,644,633,663,633,1276,1281,639,638,646,643,1282,654,625,645,628,660,633,641,639,1282,1278,696,580,639,652,1267,1282], "time": "2023-06-05T14:55:49.882+0200" }, {"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -73, "bits": 56, "valid": true, "sync": 15, "pulses": [623,7111,16819,2790,2463,2537,2482,2541,2449,2541,2448,2559,2443,2538,2473,2527,2468,2532,4778,1275,1275,1260,1291,657,635,631,642,663,634,637,638,645,626,1282,1280,644,634,664,635,627,642,1298,638,640,1271,1281,1267,658,636,638,646,643,628,1290,648,642,1255,1287,1295,634,640,641,649,1271,1287,1279,650,635,635,640,1276,649,633,1293,1269,660,612,660,624,645,652,1261,1303,645,635,638,641,1279,624,656,639,636,643,643,632,637,1286,1274,658,639,637,649,1270,1287], "time": "2023-06-05T14:55:50.032+0200" }, {"encKey": 160, "address": 9277338, "rcode": 6, "command": "Sensor", "rssi": -72, "bits": 56, "valid": true, "sync": 14, "pulses": [623,6873,520,506,18869,2465,2545,2453,2541,2448,2539,2474,2537,2448,2547,2469,2525,2470,2544,4751,1293,1268,1287,1274,647,635,635,639,640,652,634,640,640,648,1288,1266,643,650, [... cut ...] 1264,647,643,1282,1281,644,628,636,661,641,622,1278,1294,627,643,648,634,1292,644,625,629,638,658,625,644,654,1287,1266,648,629,639,647,1268,1288], "time": "2023-06-05T14:56:33.005+0200" }]

@Henk72 did you start the logs with the demo mode enabled?
Judging from the logs it seems that you started logging with the demo enabled (BIT(2)) and at some point between `2023-06-05T14:55:50.127+0200` and `2023-06-05T14:55:50.263+0200` you removed the demo mode, so:
BIT(0) = Wind
BIT(1) = Sun
BIT(2) = Demo

Henk72, 2023-06-05:
Indeed, there was not data before. I set the Solaris to demo mode and after that back. Good to see it is working.

I tested with your latest firmware and the awning looks nice and also the sun button seems to work.

Henk72, 2023-06-05:
I tried the latest pre-release integration. The awning icons are logical. It looks like extend or retract. The status is unfortunately showing 'open' where the awning is fully retracted (in). I would expect it the other way around, but this is probably the HA cover issue you mentioned.

rstrouse, 2023-06-05:
> I tried the latest pre-release integration. The awning icons are logical. It looks like extend or retract. The status is unfortunately showing 'open' where the awning is fully retracted (in). I would expect it the other way around, but this is probably the HA cover issue you mentioned.

Yep that is the frustrating goofiness in the cover domain.  If you flip it around it will disable the wrong button sadly.  The problem appears to manifest itself as if the original code was for a gate.  If you look at it as a gate it all makes sense.  Retracted is open and extended is closed.  It also works for a door and a window, just not for what I would consider a window cover that isn't a drapery or an awning.

Henk72, 2023-06-05:
The nastiness is that I have automations to close the awning at sunset or if it is raining according to the local weather radar. 

I like this automation to only fire if the awning is not fully retracted. The current status for fully retracted until but not including fully extended is 'open'. This makes it difficult to detect if the awning is (a bit) extended. Using a state != closed is also no solution. Any ideas? Or is it possible to make a toggle button for the user to invert the logic? 
I prefer to have the automations run correctly.

rstrouse, 2023-06-05:
I'll take another perusal through their code but it looks like they determine open/close based upon the percentage.  But I may be able to override the service.  The eye opener was to see all the hacks in other integrations.  A flag will not do it because simply inverting it leaves the wrong button disabled unless you only range the between 1-99% (yes that is a common hack).  Then it is never open or closed.

Henk72, 2023-06-05:
Sounds good! I hope you find a way. There might be an automation workaround to nog check state, but the position for the HA automations.

I just tested the automatic sun button. After a reboot of HA, the system started with the sun automation set to on (but the remote showed it was off). Maybe I did something wrong, or is it default set to on?

What is the sun sensor showing? Is it possible to see if there is sun, even though the system is not set to automatic?

rstrouse, 2023-06-05:
Yes it will show as a little sun in the upper left of the awning icon in the ESPSomfy RTS UI and a sensor should have been created in Home Assistant for both sun and wind for each of your awnings.

Henk72, 2023-06-06:
I tested a bit, the sun is up now and I can see it in the log. In HA however, the sun sensor is not up.

[{"encKey": 162,
  "address": 10147277,
  "rcode": 3977,
  "command": "Down",
  "rssi": -65,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [204912,102314,102508,102324,204844,204791,204822,102449,102356,102424,204785,102468,206720,52479,10435,7327,2461,2572,2480,2567,4795,1320,1240,1341,1254,664,622,668,632,1311,1269,1302,622,669,607,709,1252,1318,1240,1345,1232,1347,609,664,624,670,1259,689,596,1344,1232,1341,1238,1342,603,680,1257,1323,614,686,1247,672,610,1340,596,672,609,685,619,668,1262,1333,603,679,1267,668,611,686,600,1323,1264,665,619,669,612,1339,595,695,609,664,623,693,607,685,1250,1319,610,686,594,696,1232],
  "time": "2023-06-06T16:43:42.156+0200"
},
{"encKey": 163,
  "address": 10147277,
  "rcode": 3978,
  "command": "My",
  "rssi": -64,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [2224,102,227,355,10009,10415,7339,2474,2548,2471,2562,4802,1320,1253,1341,1233,696,603,679,606,1340,608,658,628,674,1250,1339,593,695,609,664,623,693,1235,688,594,1344,1242,1341,607,666,1246,702,606,1308,626,679,1258,670,605,1332,619,668,611,687,1250,696,587,1339,618,669,1262,1317,1254,661,620,691,610,1309,617,694,1236,1330,605,682,628,664,1271,1310,619,692,1236,1340,1254,1314,619,672,614,690,1247,1324,600,673],
  "time": "2023-06-06T16:43:55.974+0200"
},
{"encKey": 164,
  "address": 10147277,
  "rcode": 3979,
  "command": "Sun Flag",
  "rssi": -62,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10420,10711,2469,2540,2497,2568,4786,1321,1241,1343,1240,693,610,1314,1246,683,628,665,620,671,611,1337,596,674,1258,1315,1263,1342,1254,664,621,1320,610,665,622,670,1258,1317,1272,1311,1269,1331,592,691,1259,666,620,668,611,1339,1244,680,607,1317,1250,1340,1246,1334,601,685,608,684,616,670,607,680,605,690,620,671,610,664,622,669,1259,692,593,694,608,1313,619,669,1259,679,619,1323,616,669],
  "time": "2023-06-06T16:45:28.628+0200"
},
{"encKey": 164,
  "address": 10147277,
  "rcode": 3979,
  "command": "Sun Flag",
  "rssi": -63,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [604,690,619,670,36057,6516,2434,1889,2360,2569,2474,2569,2458,2568,2458,2567,2485,2545,2464,2586,4790,1328,1241,1328,1249,693,609,1315,1268,660,629,675,598,674,614,1341,598,675,1265,1327,1260,1314,1252,685,599,1348,586,689,622,672,1255,1316,1270,1338,1244,1302,623,669,1259,685,619,669,607,1314,1269,679,606,1316,1278,1313,1246,1334,617,670,609,689,599,671,625,681,602,679,620,670,609,691,593,694,1262,687,594,670,608,1341,593,696,1261,675,597,1349,592,668],
  "time": "2023-06-06T16:45:28.908+0200"
},
{"encKey": 164,
  "address": 10147277,
  "rcode": 3979,
  "command": "Sun Flag",
  "rssi": -63,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [606,26952,2489,2582,2452,2592,2455,2567,2455,2561,2483,2543,2486,2569,2461,2559,4791,1328,1243,1326,1248,697,608,1338,1247,660,25,627,674,599,700,589,1354,582,684,1255,1321,1267,1321,1262,664,618,1349,602,675,596,676,1257,1340,1235,1343,1256,1314,622,669,1261,663,625,680,603,1331,1255,683,618,1321,1258,1320,1236,1341,601,681,602,679,621,671,608,691,619,668,610,687,599,698,601,679,1259,667,609,679,629,1302,631,687,1253,668,611,1334,595,693],
  "time": "2023-06-06T16:45:29.053+0200"
},
{"encKey": 164,
  "address": 10147277,
  "rcode": 3979,
  "command": "Sun Flag",
  "rssi": -61,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [584,27053,2504,2552,2461,2579,2463,2577,2461,2572,2466,2546,2472,2586,2451,2564,4785,1321,1280,1318,1260,669,605,1331,1247,673,609,686,618,668,613,1339,594,693,1264,1315,1258,1315,1245,684,628,1317,605,680,604,683,1269,1310,1271,1328,1247,1322,598,697,1255,659,623,692,607,1310,1266,678,607,1317,1275,1314,1246,1333,621,667,608,689,600,677,624,677,609,681,617,670,611,683,616,669,1258,683,591,696,611,1309,620,703,1231,673,625,1306,621,671],
  "time": "2023-06-06T16:45:29.186+0200"
},
{"encKey": 164,
  "address": 10147277,
  "rcode": 3979,
  "command": "Sun Flag",
  "rssi": -61,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [606,26846,2603,2577,2476,2563,2453,2587,2456,2569,2466,2561,2458,2587,2463,2569,4791,1327,1246,1317,1266,696,582,1317,1272,685,600,675,622,677,607,1340,603,677,1260,1319,1264,1304,1255,692,594,1346,606,665,620,669,1260,1342,1253,1313,1248,1334,619,668,1260,665,620,694,610,1315,1248,680,604,26,1331,1258,1342,1248,1310,618,692,586,688,623,670,606,679,603,705,594,697,609,664,620,669,1259,693,592,669,636,1311,619,671,1260,675,597,1327,614,693],
  "time": "2023-06-06T16:45:29.329+0200"
},
{"encKey": 164,
  "address": 10147277,
  "rcode": 3979,
  "command": "Sun Flag",
  "rssi": -62,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [584,26966,2470,2570,2467,2575,2474,2557,2451,2569,2480,2566,2459,2566,2462,2587,4791,1305,1271,1296,1278,669,609,1339,1243,683,600,679,621,698,605,1317,598,704,1230,1321,1259,1354,1233,689,596,1345,606,659,615,694,1263,1315,1253,1317,1270,1312,619,668,1262,665,618,693,590,1337,1247,680,605,1341,1256,1311,1271,1311,618,670,609,686,601,695,602,684,602,702,598,697,581,691,620,667,1261,664,619,670,608,1340,618,671,1265,662,623,1320,597,698],
  "time": "2023-06-06T16:45:29.468+0200"
},
{"encKey": 165,
  "address": 10147277,
  "rcode": 3980,
  "command": "My",
  "rssi": -64,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [3940,98463,102370,38550,10413,7325,2479,2573,2470,2560,4807,1316,1260,1319,1262,676,596,1324,1273,1321,614,669,1256,1342,594,673,635,664,1246,1324,1267,1325,1262,1320,615,669,1258,1316,1270,1311,1266,679,603,1317,625,681,605,691,1245,680,605,1330,604,684,1253,1320,1259,671,626,682,602,1332,604,681,1256,1319,631,666,620,667,1261,1341,603,679,1259,1321,1261,1300,633,657,616,696,1252,1317,618,672],
  "time": "2023-06-06T16:45:39.627+0200"
}]

Henk72, 2023-06-06:
It seems that if the Solaris extends automatically, that the HA Somy is not updated. Please let me know if you need a specific scenario to test.

Noltari, 2023-06-06:
> I tested a bit, the sun is up now and I can see it in the log. In HA however, the sun sensor is not up.
>
> [{"encKey": 162, "address": 10147277, "rcode": 3977, "command": "Down", "rssi": -65, "bits": [...]

But those logs are from your remote, right?
The address you posted for the sensor was `9277338` and these commands are from `10147277`, so if your sensor is detecting and transmitting a Sensor packet you should see somethings like:
```
{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -72,
"bits": 56,
"valid": true,
"sync": 14,
"pulses": [607,5538,2174,25,11505,4237,3356,2450,2559,2435,2543,2451,2561,2450,2533,2444,2557,2446,2531,2468,2529,4805,1254,1270,1314,1244,654,634,639,655,641,620,651,634,641,641,1281,1276,660,638,641,628,1289,643,641,630,636,1287,1271,1280,641,650,638,631,1293,674,596,650,636,1287,1272,1287,638,623,668,627,1284,1268,1276,641,651,641,645,1263,650,644,1300,1269,620,646,657,634,634,648,1295,1292,610,646,644,634,1286,615,644,658,641,630,639,644,653,1262,1274,650,661,615,639,1294,1274],
"time": "2023-06-05T14:56:32.603+0200"
}
```

Henk72, 2023-06-06:
I see these coming in, I guess they are the sun detection flag. The sun sensor from the integration is however not set to 'on'.

[{"encKey": 160,
  "address": 9277338,
  "rcode": 2,
  "command": "Sensor",
  "rssi": -73,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [2464,2525,2463,2544,4764,1300,1262,1287,1284,654,619,645,655,638,630,646,620,670,634,1291,1270,628,636,665,626,1298,634,639,627,647,1272,1289,1296,627,637,658,642,1275,637,655,616,646,1277,1291,1269,657,632,640,646,1295,1263,1300,627,659,614,640,1296,634,635,1283,1277,659,635,649,643,627,636,1290,1296,633,635,633,648,1293,610,658,642,643,629,662,615,679,1245,1281,653,644,621,658,1282,1274],
  "time": "2023-06-06T17:27:35.730+0200"
},
{"encKey": 160,
  "address": 9277338,
  "rcode": 2,
  "command": "Sensor",
  "rssi": -72,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [629,5914,7650,13131,2464,2554,2452,2560,2447,2551,2446,2538,2471,2526,2470,2527,2468,2527,4797,1276,1284,1285,1266,635,660,651,618,648,631,659,640,645,632,1282,1269,644,635,640,641,1297,632,635,659,642,1273,1280,1280,643,630,662,614,1292,619,676,636,636,1274,1277,1285,643,652,634,636,1293,1271,1282,640,643,628,663,1283,617,653,1290,1276,647,636,638,640,650,618,1302,1284,643,628,635,639,1292,645,627,636,659,641,639,629,663,1256,1291,635,660,615,668,1253,1309],
  "time": "2023-06-06T17:27:35.864+0200"
}

Noltari, 2023-06-06:
> 

TBH I haven't had time to update and test the HA integration and I had already exposed the sun/wind sensor with MQTT, so I will test it and report back.

rstrouse, 2023-06-06:
Guys, this does not need MQTT to process those sensors.  It is emitted on the socket.

Noltari, 2023-06-06:
> Guys, this does not need MQTT to process those sensors. It is emitted on the socket.

Yes, I know, but since I already had them in MQTT I had no urge to update the HA integration.

rstrouse, 2023-06-06:
Pull v1.1.1 of the HASS integration.  I think I have the flags working properly now.

Noltari, 2023-06-06:
> Pull v1.1.1 of the HASS integration. I think I have the flags working properly now.

I've installed v1.1.1 and submitted a PR, but unfortunately there's no more sun today here, so I'll report back tomorrow.

[20 more replies]

closed 2023-06-18 by rstrouse