rstrouse/ESPSomfy-RTS #60: Can command blinds or read remote signal but not both at the same time

Gorouk, 2023-05-31:
i'm using a WROOM32 and a CC1101

Hi @rstrouse ,

I'm facing a weird issue, when I initially tried to bind the blinds I have, it was not working, so after a quick search I manage to see that inverting RX and TX should solve my problem, and it did, I could control the blinds and pair the remote.

But now, with inverted TX and RX, i can control the blinds, but the remote signal can be seen, I've nothing in the logs.
But when i change RX and TX back to the initial pins, the remote commands can be intercepted, but the blinds can't be controlled anymore...
I tryed with another ESP32, with the same result from the beginning this time. I don't have another CC1101 to see if its coming from there.

Do you have any idea of what could be the problem ?

Best regards,
Gorouk

rstrouse, 2023-05-31:
Try this.  Change both of the pins in the configuration to be the same pin when you are successfully able to send commands.  Perhaps there is an issue on the GDO2 for the transceiver.  This will send and receive on the same pin.

Gorouk, 2023-06-01:
Wow, nice thinking, I did that and it works !
Thank you very much for your quick answer and solution !

closed 2023-06-01 by Gorouk