rstrouse/ESPSomfy-RTS #64: Dude!   Have you seen this?

FlaMike, 2023-06-07:
Y'all are getting famous world-wide!  Too bad I can't understand a word he says.  I did just turn on translate to English subtitles & will watch it.

https://www.youtube.com/watch?v=dtvTQM6dqcQ

rstrouse, 2023-06-07:
I listened to it in German.  I was waiting to hear the word dummkopf but it never came up.  If you are reading this @smart-live.net, thanks for the compliments!  If the video wasn't flattering I'll just go on believing it was.

FlaMike, 2023-06-07:
Whatever translation errors or omissions there may have been I still think
you're a rock star!  I don't think you should ever close this "issue!". 😅

caibo84, 2023-06-13:
Since I am from Germany, I can tell you that he did not say Dummkopf.  It was more in the direction that you are a genius. 

From my side thanks for this great solution. One year ago i made a controll based on the Arduino SOmfy libraries and integrated the blinds in IOBroker. I tried o also detect the level with the some scripts but it didn't work out well. I also had the problem when somebody using normal remote the level doesn't fit anymore.  you're solution with detecting the remote signals is really cool

rstrouse, 2023-06-13:
Thanks for the confirmation but I don't know about the genius thing I do plenty of stupid things.  That being said, the receive portion of this almost had me buying equipment to listen for OOK pulses.  Who does that kind of thing?  In the end though, I am glad others have found utility in it.  

Really I wrote this because I hated that I didn't have control over my roller shades and dug into it deeper because there were screws that needed turning on it.  I almost didn't make the repo public.

pfak, 2023-06-14:
Just got Somfy awning, and I see 4 days ago you added support. Thanks a lot, this project is great!

rstrouse, 2023-06-14:
> Just got Somfy awning, and I see 4 days ago you added support. Thanks a lot, this project is great!

Aren't you glad you squinted for 4 more days?  Thanks for the kudos!

FlaMike, 2023-06-15:
FYI, Mr. Kieser (pfak above) is my discord buddy who was interested in your project when I mentioned it a couple of months ago.  So happy everything worked out, including "just in time" awning features going GA!  :-)

nardev, 2023-11-24:
Why the h you don't have some kind of a donate button.

You just saved me bunch of money and time.

rstrouse, 2023-11-25:
@nardev just glad you found some use for it!

piv3rt, 2024-03-11:
A big thank you to @rstrouse and all the other contributors to this great project! Impressive work! 👏
This works flawlessly in my home with HomeAssistant and inspired me to publish a [YouTube video](https://www.youtube.com/watch?v=1C-lgoN1hmk) too.

rstrouse, 2024-03-11:
Great work on the video and great explanations of the operation.   I will look at adding a section to the wiki or perhaps setting up the discussions section so others can showcase their work.  Bravo!

Nickduino, 2024-04-01:
> I almost didn't make the repo public.

That would have been a shame. Any other gem you're sitting on and have not made public?

kieranc, 2024-11-02:
> Why the h you don't have some kind of a donate button.
> 
> You just saved me bunch of money and time.

I'd also like to be able to buy you a beer @rstrouse this is a fantastic bit of software, well documented, easy to use and massively cheaper than the 200€ box somfy want to sell me to do the same job. Thank you!

kieranc, 2024-11-02:
No, you're right, it doesn't do that, but in defence of the Somfy box, I feel like this project doesn't even try to monetize my personal info? Missed opportunities....

Nickduino, 2024-11-02:
I wonder if the Tahoma does that. Like, do you see ads for umbrellas if your Somfy rain detector is active more often than the average customer?

dietrmat, 2025-02-11:
Thanks a lot for this awesome project @rstrouse 🤟 
I bought all components, connected all together and flashed an ESP32 for the first time.
Apart from that it is working perfectly with my shutters, this was so much fun. 😍 
I really like your documentation, it's detailed and fun.

memorex23, 2026-10-16:
Thank you very much and greetings from germany!