rstrouse/ESPSomfy-RTS #66: Dies not Work if MQTT Broker is offline/crashed

d3bu9, 2023-06-10:
Wenn my MQTT Broker is not reschsble also the App is not working on Port 80. It says Radio not connected.

rstrouse, 2023-06-10:
The Radio not Initialized message is not related to MQTT rather this occurs when either the radio is not enabled in the configuration or the wiring to the SPI interface to the radio is not correct which causes the initialization to fail with the driver.

I will set up a broker then disconnect it to see if this makes any difference but I suspect the two are not related.

rstrouse, 2023-06-10:
I have tried to replicate your scenario and cannot seem to get it to fail.  However, I did clean up some of the code to reset the MQTT connection when there is a change to the configuration.  Please update your device to v1.7.0.

d3bu9, 2023-06-13:
Just tried it again. As soon as I disconnect my Raspi with the MQTT Broker from the LAN and try to reload the ESPSomfy RTS it shows this:
[image]

As soon, as I reattached the LAN to the Raspi, ESPSomy RTS worked again. With disabled MQTT Client this does not happen. I'll do the Update to v1.7.0. an repport again.

d3bu9, 2023-06-13:
Hmm, the bad news: i just updatet to v1.7.0. And it's the same (eventhough I had to refresh the browser window three times).

Also, I was not able to update the Firmware, I got the folowing error. So I just could update the Application:
[image]

[image]

rstrouse, 2023-06-13:
Hmmm.  That message appears when there are no more available sockets for the webpage.  This can happen if you have too many clients accessing the device at the same time.  Do you have multiple browser windows open to the ESPSomfy web page?

rstrouse, 2023-07-07:
I am closing this for now.  Let me know if you have any further issues.

closed 2023-07-07 by rstrouse