rstrouse/ESPSomfy-RTS #67: no pairing or logs. what do i wrong?

remb0, 2023-06-10:
I bought a esp32 and the receiver.
I want to pair my telis remote but doens't received something.

Updated to 1.7.0

my layout is:
[image]
[image]

I enless checked but can't see what I do wrong.

rstrouse, 2023-06-10:
Did the Radio Not Initialized message go away?  Check the carrier board to ensure TX and RX map to GPIO12 and 13 respectively.  Typically these are mapped to 1 and 3.  Make sure GDO0 is connected to the pin you identify as TX in the setup and GDO2 is connected to the pin you identify as RX in the setup.  Alternatively, you can use only the GDO0 pin but assign both RX and TX to this pin.

rstrouse, 2023-06-10:
I am not familiar with that carrier board but almost always GPIO 12 and 13 are on the other side of the board.

remb0, 2023-06-11:
thank you for your quick response!
I bought this board: https://nl.aliexpress.com/item/1005004268911484.html?spm=a2g0o.cart.0.0.477461d7xFTlgm&mp=1&gatewayAdapt=glo2nld
the pin layout is: 
[image]

12 and 13 are indeed on other side. but it has also separated rx tx pins. (gpio 1 and 3)
I have changed that in your interface and I got an rssi and see logs :)

I don see a not initialized message before although the wires were not correct.

I can continue!  thank you very very much.  hope this message can help others as well

rstrouse, 2023-06-11:
Keep in mind gpio 1 and 3 are attached to the UART on the chip.  The TX pin will attempt to send output on boot through a debugger and pin 3 must remain high at boot or it will go into debug mode.  It should be fine unless you notice weird happenings when the board is booting.  I try to shy away from any pin that has special purpose during boot.

GelsemaP, 2023-06-12:
Hi, I have a similar issue. I cannot pair. Nothing happens on both my blinds and awning. At first I thought the memory of the blinds was full so i did a factory reset. To no avail.
I have also the ESP32-Wroom32 bought from azdelivery (https://www.azdelivery.de/nl/products/esp32-developmentboard).
I have used the default, suggested wiring from you and I have the radio working. I am getting nice reception, I see it in the logs and everything seems to work until I try to pair. I get the blinds in pairing mode (push prog on remote, wait for jog, then push on prog on the espsomfy page) but nothing happens. Neither on the blinds or on the awning. The esp is within 2.5 meters from the blind and 2 meters from the remote. Whenever I try to pair (i suppose transmit) a red light flickers on the esp.

see screenshots

[image]

[image]

rstrouse, 2023-06-12:
Which transceiver do you have is it an E07-M1101D-SMA?  If it is trace pin 3 (GDO0) on the transceiver and given your setup above it should connect up to D12 or GPIO12.  If if connects to D13 or GPIO13 then swap the RX and TX settings.

rstrouse, 2023-06-12:
Also if it is an E07-M1101D-SMA your max TX power is 10dbm.

rstrouse, 2023-06-13:
Not weird at all that you could receive but not transmit.  That is because the TI chip can receive on either GDO0 or GDO2 but only transmit on GDO0.  You are not the first to confuse this with serial comms though where, it is actually TXD and RXD which in many instances the D is silent and how the pins are actually read is determined by the protocol.  

I am trying to formulate the magic text that will allow folks to understand these differences.  It is funny that it seems folks with a greater exposure to these type of electronics are more likely to think of it as a serial transmission whereas the peeps with little exposure just match the pin numbers and set the configuration.   What makes this more confusing is that you can even use a single pin for both TX and RX so long as it is connected to GDO0.  

I could use GDO0/2 but that is even more confusing when you read the transceiver docs.

rstrouse, 2023-06-13:
Glad you got it working.  I am going to close this for now.

closed 2023-06-13 by rstrouse