rstrouse/ESPSomfy-RTS #68: Corrupted Configuration

isngofoz, 2023-06-12:
Not sure what happened here. 
I have 3 shutters in total, just a few days ago, one of the shutters were not visible on HA so I opened up the configuration and saw this. 
Coincidentally, I just installed the new HA core update and also saw the new ESP-Somfy HA integration available so I figured I might have missed some important firmware and application updates. 
Luckily for the backups, I restored from a backup and proceed to update the firmware to the latest 1.7.0 and also the HA integration.

Two days later, the same thing happened again.

Any thoughts?
[image]
[image]
[image]

rstrouse, 2023-06-13:
Hmmm... create a backup and post it here.  This looks like corruption on the flash but it could be something that I am just not seeing.  How are you powering the ESP32?

rstrouse, 2023-06-13:
Also, did you attempt to downgrade the firmware?

isngofoz, 2023-06-13:
Backup attached. No, I did not downgrade the firmware. The last firmware I installed before 1.7.0 was 1.6.1 

Something weird I thought I saw after I upgraded to v1.7.0 and on the repeat of this issue was that the firmware was showing as 1.6.0 but after I rebooted the ESP, it became 1.7.0

[ESPSomfyRTS 2023-06-13T21_50_43.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/11735594/ESPSomfyRTS.2023-06-13T21_50_43.zip)

rstrouse, 2023-06-13:
Which ESP32 device do you have and which pins are you using for the radio.  You can just shoot a screenshot of the transceiver configuration.  I ask because it sounds like the firmware update process is not reading the flash back until a reset.  This can be caused because one of the bootstrap pins are being used for another purpose.

isngofoz, 2023-06-15:
I think it is a standard ESP32 but probably with ch9102x variant.

[image]
[image]
[image]

rstrouse, 2023-06-15:
I do not see any weird use of pins.  I do not understand why the littlefs file system is getting corrupted.  I wonder if it might just be a bad chip.  It looks like it loaded the storage out of alignment but your backup file looked great which is what it loads on boot.  If it does this again please report what version number is showing in firmware screen.  Could it be just reverting to an earlier firmware???

isngofoz, 2023-06-16:
Yes, I will monitor.
It has not happened again (I guess this might jinx it).

rstrouse, 2023-07-07:
I am closing this for now.  I don't know what the issue might have been but I assume you are up and running.

isngofoz, 2023-07-08:
Hi, actually it occurred again a few days ago shortly (1-2 days maybe) after I upgraded to the new version v1.7.2. I have just been too busy to post. Unfortunately, I cannot recall if I powered cycle the chip or if I just rebooted it after the firmware and app update.

I have confirmed that I was not dreaming before. This is the snapshot of the firmware when the configuration went crazy. 
[image]

This is the snapshot after a power cycle.
[image]

rstrouse, 2023-07-08:
That is weird.  It is acting like the reboot after the firmware update is not actually happening.  This means that it is not catching the configuration updates when you update the application.  There are some reports on the interweb about ESP32 reverting the firmware after update.  Weirdly reverting it after restart.  I am researching but it seems to be on random devices with no reference to specific firmware.

isngofoz, 2023-07-09:
I just updated to v2.0.0 and this time I power cycled the chip. I will monitor to see if it happens again.

rstrouse, 2023-07-13:
Closing this for now.  If it happens again let me know.  I have seen lots of reports online where some of the carrier boards to not properly set the boot voltage internally so it never really commits the firmware.  I would power cycling after installation with your ESP32.

closed 2023-07-13 by rstrouse