rstrouse/ESPSomfy-RTS #69: Add activity flag and inverted flag

caibo84, 2023-06-13:
I use your remote together with IOBroker via MQTT. Also it's coupled with Google Home

First issue is that for the VIs in IOBroker via Vis it's good to have a activity flag so that the stop button is only pressable when the activity flag is high. So activity flag needs to be true when blind is moving or tilting.

Second thing is that the level in Google Home and also in IOBroker is inverted. So 0% is fully closed and 100% is fully open. It would help if the value for position could be inverted via the configuration.

rstrouse, 2023-06-13:
The stop button should also be pressable when the shade is at rest as this is actually a capability of the RTS protocol.  The reason is because that button is also used to send the movement to the my favorite position.  

However, a flag already exists in the MQTT emit for movement.  This is direction and it will be 0 if the shade is at rest, -1 if it is moving upward and 1 if it is moving downward.

I have very strong opinions on the inverted flag and if it were simply about the emits I would make a quick change to get it all to work.  However, that is not the case.  This has more to do with the way one thinks about a shade or blind vs a gate, window or door.

If I come up with a solution that doesn't result in the wrong icons being shown or the inability to operate the motor at the 0 or 100% range I'll make a change to get it to switch the values.

larsimmisch, 2023-06-15:
First of all: this project is great and worked out of the box. Nice! 

Also: excellent documentation, thanks!

I have a similar problem like caibo84: my awnings show the wrong position in the ESPSomfy-RTS web interface. So when the awning is fully extended, it looks retracted in the UI.

The down button on the original remote will retract, BTW.

rstrouse, 2023-06-15:
Hmmm... the Somfy docs say this is incorrect and it will not interface with a sensor correctly.  To change this you need to perform the function below.  There is another way to do this by holding down the my button for 15 seconds but I do not know if this is only when setting the limits.

https://www.youtube.com/watch?v=xM_rq0WfrWA

Please try this first as this is the recommended installation.  However, if up is down and down is up and it is not just a matter of a single remote or display in percentage of open then an invert checkbox may be able to work by translating the commands while decoding.

larsimmisch, 2023-06-15:
Okay! I could reset the direction on one of my awnings. But I can't control it with ESPSomfy-RTS any more. I tried unpairing ESPSomfy-RTS and re-pairing, but the buttons on the web interface don't do anything.

I can still see the commands from my manual remote in the transceiver logs. If the key has anything to do with the rolling code, then these are out of sync now. Should I maybe set the rolling code?

rstrouse, 2023-06-15:
It may just indicate that it is paired when it is unpaired and vice versa.  There is no real indication from the motor when it pairs or unpairs successfully.  Are you getting the awning to jog?  Open the pairing/unpairing screen with the awning in pairing mode and press the prog button.  If it jogs try to move the awning with ESPSomfy RTS.  If it does then it is paired.  Do not tell it that it is unpaired.

larsimmisch, 2023-06-16:
Thank you for your quick reply. A variation of this worked fine (I had to press the unpair button after the prog button. Just pressing prog and not unpair after opening the memory through the remote didn't 🤷‍♂️ ).

rstrouse, 2023-06-16:
@caibo84 I have begun decoupling the emits from the commands to see if I can simply twist the output to spit out a percentage of open.  I have added an flip commands flag but that does not change the disposition of the output percentage from percentage of closed to percentage of open.  I am working on a flip position that will operate and emit the percentages as a percentage of open.

rstrouse, 2023-06-17:
v1.7.1 has been released and the Invert Position option has been added.  An option for Invert Commands has also been added if you simply cannot reverse the direction of your motor for the proper commands.

closed 2023-06-17 by rstrouse

Anashost, 2023-06-17:
@rstrouse thanks for the (invert commands) option. It make more sense for my brain this way 😅 as im using balcony sun shades. So closed means shade is retracted all the way, and 100% open means shade is open all the way.

But it would be better if the buttons are not reversed, so:
Down arrow = open - extend the shade
Up arrow = close - retract the shade

I had only to check (invert commands) option to show right percent, closed and open values in home assistant.

Checking the second option (invert Position) makes things worng in Home assistant to the point i don't understand what's going on.

So For now i will stick with the (invert commands) option as it works.

rstrouse, 2023-06-20:
If you set these as an awning in ESPSomfy RTS with the most recent HA integration you will find that the buttons will be correct in HA as well as the position.